tretyakovsa/Sonoff_WiFi_switch #2: 2 мелкие орфографические прпвки

dinovasya, 2016-12-17:
Воскресенья","Понидельник

Замените на Воскресенье","Понедельник

Спасибо

renat2985, 2016-12-17:
Спасибо, исправил. Буду рад если найдете еще какие-то недочеты.

dinovasya, 2016-12-17:
Да, в твиттер отписал. Короче сразу при первом запуске sonoff. На первой
странице setup.htm ввести свою сетку вай-фай невозможно. Тип поля
неправильный, не текстовый, а выпадающий список. А в нём пусто.  Прилагаю
скрин.

dinovasya, 2016-12-17:
https://postimg.org/image/5gf9apb0p/
https://postimg.org/image/rlk73oisx/

renat2985, 2016-12-17:
Хм. Должно быть так:
[image]

То есть в выпадающем списке вы должны видеть все доступные Wi-Fi сети. Я предполагаю вам нужно обновить файловую систему и сделать рефрешь на странице setup.htm, т.к. в последних версиях кешируются js файлы.

Если не получится, свяжитесь со мной в skype: renat2985 
Буду на связе через 2 часа.

dinovasya, 2016-12-17:
я каждый раз файловую перешиваю.. но вот кеширование возможно. Хром собака
тот еще- куки удалять не помогает- в нем только сброс до заводских настроек
помогает кеш снести.
Ну вот этот выпадающий список пуст.

dinovasya, 2016-12-17:
да, дело было в хроме. сбросил настройки до заводских- сраболтало. в хроме теперь открылось нормально..

dinovasya, 2016-12-17:
еще одна орфографическая правка:
Здесь указается время в секундах как долго будет работать реле.
Здесь указЫВается время в секундах как долго будет работать реле.

А лучше кратко написать: Продолжительность работы таймера.

Кстати данный блок на всех языках выглядит на Русском ))

renat2985, 2016-12-17:
Ошибку исправил. Спасибо.

jkjk007, 2017-11-13:
Гостиная пишется с одним "Н".

renat2985, 2017-11-13:
@jkjk007 я не русский мне можно и с двумя :) это местонахождение, там вы можете написать все что угодно.