tretyakovsa/Sonoff_WiFi_switch #4: Как смотрите на возможность голосового управления

dinovasya, 2016-12-17:
Например на странице управления Кнопку Микрофон добавить, а зе ней спрятано несколько сотен строк кода)) .
и кнопка "записать комманду" - которая отправит комманду на гугл voice, а вернёт результат, который в базу данных запишется. И кнопка присвоить комманду на действие.

dinovasya, 2016-12-17:
Можно будит сказать "Ок google, включи". - По моему- это самое идеальное решение.
А то,что я писал относится к тому, если мы имеем, допустим, 10 розеток.
Тогда, не легче нажимать кнопку , а лучше сказать, "Ок google, включи чайник"или "Ок google, включи свет в зале". "Ок google, выключи телик у детей". 
Воот. чтобы присвоить комманду к управлению какой-либо конкретной розеткой.

А можно какую-нибудь уже альфа-версию выпустить уже с одной кнопочкой голосовой? Уж очень хочется ну если не опробовать, то хотя бы увидеть эту кнопочку в веб-интерфейсе..  ))

dinovasya, 2016-12-17:
"Ок google, жалюзи закрыть". включить синий цвет подсветки

tretyakovsa, 2017-03-25:
Никаких кнопок не нужно. Скачайте Дусю создайте простенький скрипт и пользуйтесь.