tretyakovsa/Sonoff_WiFi_switch #5: кнопка ВКЛЮЧИТЬ/ВЫКЛЮЧИТЬ

dinovasya, 2016-12-17:
Заметил тонкость.
Вы недавно поправили эту кнопку чтобы она от статуса реле отображала правильный текст на кнопке.
Это отлично.
Но, другое дело с Таймерами.
При включении по таймеру эта кнопка остается с прежним тстатусом- не меняется.
Так-что при включении по таймеру- таймер должен также менят статус данной кнопки ))

renat2985, 2016-12-17:
Не только по таймеру, если вы с кнопочки на самом sonoff выключили или включили, статус тоже не поменяется. Как вариант можно спрашивать каждые 5 секунд статус. Я подумаю об этом, спасибо.

dinovasya, 2016-12-17:
незнаю- хорошо ли это постоянный опрос проводить в смысле кодинга.. может всё-таки лучше посылать статус? но вам лучше знать..

alxnovo, 2017-09-20:
Похожая проблема или фича когда ESP'шка управляется с разных устройств. Жена у меня понапереключала через, а у меня на ноуте при этом изменения не отображались, пока обновить страницу не нажал

renat2985, 2017-09-20:
@alxnovo статусы устройств пока не обновляются сами. Вам нужно сделать refresh или нажать на кнопочку (см. рис.) и только тогда вы увидите их реальный статус на данное время. В будущем мы тоже с этим что-то придумаем. Сейчас работаем над сценариями.
[image]

alxnovo, 2017-09-27:
Пока в page.htm вставил костыль `<meta http-equiv="Refresh" content="15"/>` чтобы страница управления автоматически обновлялась и статус кнопки, соответственно, тоже. 
Как обновлять только элемент я ещё не знаю, поэтому пока только страницу

renat2985, 2017-09-28:
@alxnovo не стоит так делать. Лучше нажмите Ctr+M и добавьте "reload":"3000", как на фото ниже
[image]
Таким образом у вам будет обновляться не вся страница, а только json файлы с информацией.
Почитайте документацию: https://github.com/tretyakovsa/Sonoff_WiFi_switch/wiki/%D0%92%D0%BE%D0%B7%D0%BC%D0%BE%D0%B6%D0%BD%D0%BE%D1%81%D1%82%D0%B8-page.htm%3F*

alxnovo, 2017-09-28:
@renat2985 Вау, есть такая фича как рефреш не всей  страницы :) Согласен, обновлять json правильнее с точки зрения ресурсов! 

Понадеялся что можно так только кнопку обновлять - не сработало, только json. Визуально смотрится хуже чем когда обновлялась вся страница, из-за таблицы с таймером элементы интерфейса прыгают, т.к. перезаполняются данные в таблице таймеров.

Вот видео в котором видно прыгающий интерфейс, рефреш специально сделал 1 сек для наглядности https://youtu.be/gmmitn18bqk

Если таблицу с таймерами убрать ниже видимой части экрана, то такой прыгатьни не будет, только фокус с кнопки не снимается и она цветом как при нажатии :(

Всё-таки в моём случае нужно видеть таймеры и нужно обновление кнопки (а ещё лучше по одному из алгоритму которому [я писал в другой ветке](https://github.com/tretyakovsa/Sonoff_WiFi_switch/issues/20#issuecomment-332695478)

За документацию спасибо, это повторение одного из видео, но читать и использовать документацию как справочник удобнее.

Может есть ещё документация как можно обновить только кнопку
- или как для кнопки сделать свой JSON и обновлять его из программы. Я бы на время движения жалюзи заблокировал кнопку таким JSON 
```
{
   "type":"button",
   "title":"в процессе...",
   "action":"",
   "class":"btn btn-block btn-lg btn-default"
  }
``` 
А то когда из другой комнаты закрываешь ролеты не понимаешь команда дошла или нет до них и нажимаешь повторно (жена не один раз нажимает) в итоге хотел закрыть, а итог открытые ролеты. Когда разберусь как избавиться от [delay](https://github.com/tretyakovsa/jaluzi/issues/3#issuecomment-332698220) то хочу сделать чтобы при движении была кнопка "СТОП"
```
{
   "type":"button",
   "title":"СТОП",
   "action":"jalousiestop",
   "class":"btn btn-block btn-lg btn-danger"
  }
```
- или как прописать в коде чтобы после нажатия кнопки (жалюзи начали двигаться) интерфейс начинал обновляться с кнопкой и после того как жалюзи остановились (определять по какому-нибудь флагу, а флаг меняться может по разным событиям...по времени, оборотам и т.д.) перестать обновлять. Лучше, конечно обновлять только кнопку, а не весь интерфейс или данные (JSON)

renat2985, 2017-09-28:
@alxnovo старайтесь писать меньше текста, мне лень все читать, и разбивайте что ли на темы, тогда та тема которая мне понравиться, я прочту и отвечу вам. :) Ваше сообщение прочел, подумаю, приму к сведенью. Спасибо.

alxnovo, 2017-09-28:
@renat2985 Я столько никогда и не писал :) Стараюсь подробно написать, т.к. понимаю что пишу на делитанском языке, будь я программистом, то текста в разы было меньше.
Рекомендации принял.

QLAWQ, 2018-03-28:
Я так понимаю проблема с состоянием кнопки ещё актуальна? Или есть решение без перезагрузки блоков?

renat2985, 2018-03-28:
@QLAWQ , ну как вам сказать. Сейчас когда вы открываете главную или какую либо другую страницу вы получаете актуальные данные, то есть видите relay/rgb включено или выключено. Если же вы не будите закрывать страницу, и какой-то статус изменится вы этого не увидите. Поэтому не рекомендуется держать страницу всегда открытой, да и это лишняя нагрузка на ESP.
Советую сделать ярлык на рабоче столе или закладку в браузере. Тогда у вас не будет проблем с актуальными данными. То есть. Открыли веб интерфейс, нажав на ярлык или закладку. Посмотрели состояние компонентов, сделали нужные действия и закрыли страницу.
На мой взгляд проблема решена.
  
[image]

QLAWQ, 2018-03-28:
Так то оно так. Но адаптацию чтоб отправлять по сериал данные в arduino mega очень сложно... В коде чёрт ногу сломит :(

renat2985, 2018-03-28:
@QLAWQ в каких целях используете Mega? Проект прекрасно работает и без нее. Я вообще считаю стоит избавляется от этого динозавра.
Как вариант можете парсить http://esp/config.live.json там находятся данные состояния реле и других модулей.

QLAWQ, 2018-03-28:
**renat2985** Для расширения портов. Т.к. автоматизирована вся квартира, тёплые полы, освещение, полив растений, жалюзи, увлажнители, вытяжка. Датчики газа, пламени, протечки, герконы на дверях. Делаю всё в меге flprog и отправляю\принимаю в есп.

QLAWQ, 2018-03-28:
`  HTTP.on("/UVSon", HTTP_GET, []() {
            UVS = 1; //включить увлажнитель
            jsonWrite(configJson, "stateUVS", int(UVS));            
    HTTP.send(200, "text/html");
  });
  HTTP.on("/UVSoff", HTTP_GET, []() {
            UVS = 0; //отключить увлажнитель
          jsonWrite(configJson, "stateUVS", int(UVS)); 
    HTTP.send(200, "text/html");
  });`
Обновляется статус при перезагрузки страницы.... :(