tretyakovsa/Sonoff_WiFi_switch #8: Добавить возможность переназначить пины на вход или выход

2007-i, 2017-03-24:
Здравствуйте.
Спасибо за сборку!
Можно ли добавить переназначение пинов (физических выводов) на другие прямо в веб интерфейсе, например выход реле с пина 12 переназначить на 2-ой пин (конечно же еще нужно делать проверку на совпадение что бы не было на одном пине несколько разных выходов или входов).

tretyakovsa, 2017-03-25:
Да такое есть в планах, но пока это только хотелка. В первую очередь такой подход сдерживается тем , что прошивка создавалась под конкретное устройство.

2007-i, 2017-03-25:
Но можно же создать универсальную прошивку в которой можно назначать пины модуля на вход или выход (можно ещё добавить pwm). Тогда её можно будет прошить в разные модули для разного назначения и они все будут отображаться в общем интерфейсе. Тогда её можно заливать и в Sonoff и в например беспроводной датчик температуры и модуль управления нагрузкой и т.д. А что бы не загружать "лишнего" можно перед прошивкой добавлять или убирать нужные функции в виде скетчей *.ino?

tretyakovsa, 2017-03-25:
Тогда это будет уже не прошивка, а скетч. И если добавлять разные модули и делать то руками, что мешает задать пины в ручную.

2007-i, 2017-03-25:
Я просто имел ввиду настройка портов прямо в веб интерфейсе
[image]
Как то вот так например как у Макса.
А руками конечно тоже можно редактировать, но для это постоянно нужно перезаливать прошивку в модуль. А по поводу сборки разных прошивок из скетчей это уже моя личная "хотелка" можете не обращать на это внимание если считаете не интересным. Просто думал что то наподобие этого wifi-iot.com/p/esp8266/ имею ввиду выбор нужных опций у будущей прошивки, но опять же у этого ресурса есть платные функции а у вас это всё на добровольной основе (кто хочет поддержать деньгами проект). Хотя я думаю у него сборка происходит на сервере из выбранных модулей что можно делать и самому в нашем случае.

renat2985, 2017-03-28:
Начали делать новую страницу modules.htm, сегодня ее выгрузил на github. В будущем на ней можно будит выбрать какие модули вы хотите использовать. И думаю сделаем там возможность переназначения портов. На нее будит автоматическое перенаправление при первом старте ESP
[image]

2007-i, 2017-03-28:
Вот об этой возможности я и говорил, думаю многим понравится!
Спасибо!

ilya89vet, 2017-07-23:
Доброго времечка вам!
Возможность переназначения пинов уже доступна или еще в стадии разработки?

renat2985, 2017-07-23:
@ilya89vet не так давно стало возможно включать\выключать модули, то есть можно включить или выключить MQTT, DDNS, поменять pin для relay и многое другое. 
Подключая первый раз устройство к своему роутеру, вам в выпадающем меню нужно выбрать сборку RGB, RELAY или д.р.
[image]

Далее ищите выбранную вами сборку [в config.modules.json файле](https://github.com/tretyakovsa/Sonoff_WiFi_switch/blob/master/data/config.modules.json) через файловый менеджер, и можете корректировать его, добавлять или удалять модули или менять pin-ы. Более удобного веб интерфейса пока нет.
[image]

ilya89vet, 2017-07-24:
А как сохранить изменения в config.modules.json ?
И еще
Я использую Arduino реле которое вкл отрицательным потенциалом. Как-то можно инвертировать выход?

renat2985, 2017-07-24:
@ilya89vet, пожалуйста создавайте новые темы, не пишите все вопросы в одну.