tretyakovsa/Sonoff_WiFi_switch #9: Режим диммера (шим регулятор)

2007-i, 2017-03-24:
Еще вопрос, можно ли добавить режим диммера (pwm), удобно было бы управлять например светодиодной лентой подключенной через мосфет (возможно и управление лампой или др. нагрузкой 220 v. если подключить через симистор) меняя яркость в зависимости например от указанного времени (или от фотоэлемента если он подключен).
Если нужно могу тестировать пока есть еще свободные esp8266.

tretyakovsa, 2017-03-25:
Фактически мы начали писать эту прошивку для Sonoff с целью, чтоб ее можно было залить в готовое устройство. Для добавления функций способных работать на других устройствах желательно понимать: 
способ подключения (например управлять мосфетом, совсем не то, что управлять симистором ) ,
интерфейс управления (зависит от конкретного устройства)
Предлагайте варианты подключения и интерфейса.
Мы все больше и больше приближаемся к модульной системе прошивки с целью добавлять туда любые функции, но это уже больше, чем sonoff.

2007-i, 2017-03-25:
Да, согласен, модульная система это уже не sonoff, но с другой стороны это получится универсальная прошивка для разных устройств разного назначения.
Кстати у sonoff есть устройство с диммером для управления светодиодной летны: https://www.itead.cc/sonoff-led.html
В интерфейсе можно добавить ползунок с регулятором яркости рядом например с кнопкой включения. А вариант подключения например тот же что и у модуля выше по ссылке. В конце концов если нужно будет только включить или выключить можно поставить просто реле после транзистора (или напрямую реле например omron) и пользоваться просто кнопкой вкл и выкл.