tretyakovsa/Sonoff_WiFi_switch #12: Управление через mqtt

2007-i, 2017-03-29:
Скажите пожалуйста по какому get запросу можно управлять с помощью этой прошивки?
То что показывает в истории URL: http://192.168.10.142/sonoff не совсем подходит для управления через mqtt т.к. запрос одинаковый, если бы было например URL: http://192.168.10.142/sonoff-on например для включения и URL: http://192.168.10.142/sonoff-off для отключения реле. Может есть такое а я это пропустил?

PS. Кстати обновление скетча mqtt от Mar 24, 2017 решило проблему с отправкой данных, раньше не получалось что то передать, а теперь сразу все отправляет. Спасибо за вашу работу!

renat2985, 2017-03-30:
Создаем профиль в cloudmqtt.com, выбираем Free план.
Заходим в http://локальный_ip_устройсва/setup.htm указываем там в MQTT сервер, логин, пароль, порт полученные от cloudmqtt.com.

Далее, переходим в раздел цифра 1,
Цифра 2, видим статус вашего реле в реальном времени, 1 - включено 2 - выключено 

Управлять можно следующим образом:
Цифра 3, пишем topic.
Цифра 4 пишем 1 если хотим включить или 0 если хотим выключить. И жмем Send.
[image]

2007-i, 2017-03-30:
Как с MQTT управлять я в курсе, спасибо. Просто я раньше управлял через Мажордомо и делал это через скрипт:

```
if ((getGlobal('ESP8266-01.sw')) =="1")
{
 getURL('http://192.168.10.142/gpio?st=1&pin=0', 0);
} else if ((getGlobal('ESP8266-01.sw')) =="0") 
{ 
 getURL('http://192.168.10.142/gpio?st=0&pin=0', 0);
}
```

**ESP8266-01** - это я менял на **RELE_1** а вот с get запросом выходит накладка, при указании http://192.168.10.142/sonoff в обоих случаях, у меня переключатель не видит в каком состоянии находится модуль, хотя данные на MQTT приходят.
Вот так у меня сейчас:

```
if ((getGlobal('RELE_1.sw')) =="1")
{
 getURL('http://192.168.10.142/sonoff', 0);
} else if ((getGlobal('RELE_1.sw')) =="0") 
{ 
 getURL('http://192.168.10.142/sonoff', 0);
}
```
Вот что имел ввиду.

Sergey-SRG, 2017-03-30:
Для управления этим реле в мажордомо, нужно всего привязать Связанный объект и его свойства!
При этом в случае изменения свойства мажордомо будет пупликовать в топик.
И не нужно не каких дополнительных гет запросов городить(

Sergey-SRG, 2017-03-30:
[image]

Sergey-SRG, 2017-03-30:
Хотелось бы видеть при управлении по mqtt разделение топиков 
топик status (для обратной связи, бывают ситуации что реле не в сети и мы не когда не узнаем было переключение или нет :(  )
и топик switch (для переключения)

AVL4GITHUB, 2017-04-03:
По управлению через МКТТ мои замечания.
Бардак следующий.
1 -включает реле.
0 - выключает реле.
Вроде всё ОК, но:
ЛЮБОЕ другое число (не 0 и не 1)  изменяет состояние реле на ПРОТИВОПОЛОЖНОЕ.
ЛЮБАЯ текстовя строка выключает реле.
Таким образом легко поймать непонятное и непредсказуемое состояние реле.
Можно ЭТО исправить?
Спасибо!

AVL4GITHUB, 2017-04-09:
Как успехи?
Все с нетерпением ждут продолжения.

Sergey-SRG, 2017-04-11:
Ребята пилят вещь которая старается работать без сервера управления.

Если нужен чистый mqtt , то лучше попробовать это [https://github.com/arendst/Sonoff-Tasmota](https://github.com/arendst/Sonoff-Tasmota)

Berkseo, 2017-05-25:
я решил вот так:
 if (String(pub.topic()) == prefix + "/"+chipID + "/RELE_1_not/control") // проверяем из нужного ли нам топика пришли данные
    //if (String(pub.topic()) == prefix + "/"+chipID + "/RELE_1_not/status") // проверяем из нужного ли нам топика пришли данные
   {
    int stled = payload.toInt(); // преобразуем полученные данные в тип integer
    Serial.println(stled);
    if(stled==1)   {
      command = "relayOn";
      relayOn();
    }
    if(stled==0)   {
      command = "relayOff";
      relayOff();
    }
     Serial.println(command);

  }

теперь если пишем в топик /control - 1(включает)...

symparua, 2017-06-11:
Sergey-SRG, а можете показать как у Вас настроены Связанный объект и его свойства в Мажордомо?

egven, 2017-06-25:
Здравствуйте!
Подскажите, пожалуйста, с чем может быть связана проблема с подпиской не следующие топики

[image]

tretyakovsa, 2017-06-27:
А что вы пытаетесь сделать? Откуда подписываетесь? Работает ли при этом IoT manager?

egven, 2017-06-27:
я пытаюсь поуправлять реле через соединение esp8266-cloudmqtt-mqtt mobile client

скачал предпоследнюю прошивку, не настраивал mqtt, сразу обновился по воздуху, настроил mqtt, и щас вижу по логу с сервера, что не конектится устройство, хотя прошивка из ветки RGB вроде конектилась, но топипи не публиковались

через ком порт кроме фразы MQTT ниче нет на тему соединения с сервером, настроки печатает правильные, как вообще можно проверить работу IoT managerа

мои настройки тут

[image]

tretyakovsa, 2017-06-28:
Поставить IoT manager. Подключится к серверу.

egven, 2017-06-28:
Если речь идет о мобильно клиенте, то тут он не при чем, пока нет конекта между esp8266 и cloudmqtt, для проверки я еще приводил скрин выше, где вебклиент, подписанный не топики, не получает ничего

tretyakovsa, 2017-06-28:
Я специально проверил. Последняя версия работает и все публикуется. Вы настраиваете SSL Port или Port? Нужно использовать Port без SSL.

egven, 2017-06-29:
я обычный порт ставил, без ссл, а через ваб клиент на cloudmqtt, по скрину если смотреть, я корректно все делаю? может из-за того что я по воздуху обновился, надо перепрошиться

i81, 2017-07-13:
Товарищи, а можно подробнее, как настроить MQTT. Сделал как написано выше, но во вкладе WebsocketUI у меня пусто :(
[image]
[image]
[image]

egven, 2017-07-13:
i81, собственно у меня пока такая же проблема, на одной из последних прошивок в логе видно, что could't connect to mqtt server, а в чем трабла не понятно, вроде до сервиса privet.lv он достукивается, ok приходит, а до mqtt никак.

i81, 2017-07-13:
Ладно хотя бы я не один такой...
А как ещё можно управлять с Андроида, кроме http запросов?

renat2985, 2017-07-13:
C SQL разберемся в ближайшие дни.
@i81 Я пользуюсь [HTTP Request Shortcuts](https://play.google.com/store/apps/details?id=ch.rmy.android.http_shortcuts) программка умеет делать ярлыки на рабочий стол с нужными вам GET запросами. Таким образом нажав на ярлык с рабочего стола можно управлять устройствами. Так же там есть возможность управления голосом. Но пока это работает достаточно плохо.

symparua, 2017-07-13:
Пользуюсь MQTT Dash для включения/выключения реле с Андроида. Все работает...

renat2985, 2017-07-13:
@i81, постучитесь ко мне в Skype: renat2985, попробуем вместе настроить.

i81, 2017-07-13:
Ренат,  всё оставил в офисе, уже дома... Завтра уезжаю на месяц в отпуск... большое Вам спасибо заранее, что не оставляете в беде меня! Если это будет возможно - я получу к Вам в середине августа...

alxnovo, 2017-09-28:
А чем, кроме IOT Manager можно управлять по MQTT под IOS?
IOT Manager исчез из AppStore, а Андроида у меня нет :(

Или как можно управлять такими топиками, что в том же CloudMQTT нужно написать чтобы открыть/закрыть жалюзи?

[image]

alexandersciv, 2017-09-30:
Решил вопрос как правильно определиться с топиком, что в нём нужно прописать.
Настроил CloudMQTT. Прописал данные в прошивке модуля. Установил программу IoT Manager, прописал в ней данные из CloudMQTT. После подключения IoT Manager к CloudMQTT появился мой модуль. Включил его в  IoT Manager и в CloudMQTT Console появился топик
[image]
В **Send message** -  Topic  прописываю значение 
[image]
в Message пишу 1 или 0. Всё работает.

AVL4GITHUB, 2017-09-30:
Через сеть 3G сеть  IoT Manager  у Вас работает? У меня только через ВайФай...

alexandersciv, 2017-09-30:
**AVL4GITHUB** подключаюсь по ВайФай, пока решаю вопрос стабильного подключения к MQTT.
В основах CloudMQTT разобрался. Теперь подключаюсь к локально к Mosquitto.

alexandersciv, 2017-10-16:
**@renat2985**  = Я пользуюсь **HTTP Request Shortcuts** программка умеет делать ярлыки на рабочий стол=
Ни как не могу разобраться, как настроить программу. Подскажите где почитать?

renat2985, 2017-10-16:
@alexandersciv прекратите пользоваться дурацкими выделениями и =, для этого есть специальные кнопки в верху
[image]

О HTTP Request Shortcuts, ок если будет время завтра сделаю видео обзор.