tretyakovsa/Sonoff_WiFi_switch #13: Не допустимая ситуация после подачи питания на ЕСП.  IOT5 самопроизвольно переключается.

AVL4GITHUB, 2017-04-03:
Всем привет.
Такая вот проблема.
 На ИОТ5 висит у меня твердотельное реле, которое управляет светодиодной 220В лампочкой, мощностью 5 Ватт.
После подачи питания (ресета) лампочка, если была предварительно выключенной - включится секунд на 5.
Если была предварительно включённой - то включится секунд на 5, выключится секунд на 5 и уже включится на постоянно.
Поскольку я собираюсь управлять девайсом двумя гаражными воротами (въездными и гаражными то такое поведение выходов ЕСП меня совсем не устраивает, ибо, если ЕСП почему-то перегрузится (пропадало питание, к примеру), то ворота самопроизвольно ОТКРОЮТСЯ. А поскольку гаражные ворота соеденины  с  домом, то открывается доступ к всему дому.
Можно ЭТО исправить?
Спасибо!

tretyakovsa, 2017-04-03:
Это вы на какой плате делаете? И к какому GPIO подключено реле?

AVL4GITHUB, 2017-04-03:
Плата СОНОФФ.
Моё дополнительное реле подключено к ГПИО5.
В скетче я поменял 12 и 5 местами.
Когда в скетче ничего не меняю, тотакого люка не наблюдается.
Разве-что, есть зависимость состояния реле от  значения пина в МКТТ.
Если, к примеру, обесточить СОНОФФ, после чего, по МКТТ дать команду на изменение состояния пина, то после подачи питания - состояние пина изменится.

AVL4GITHUB, 2017-04-05:
Свежие замечания.
Полностью  авторская прошивка, ничего не меняю.
Следующий глюк.
СОНОФФ.
Реле выключено.
Передёргиваю питание и через пару секунд реле ВКЛЮЧАЕТСЯ.
По МКТТ показывает, что реле ВЫКЛЮЧЕНО.
То-есть не порсто втихаря включается, но ещё и врёт по МКТТ.