tretyakovsa/Sonoff_WiFi_switch #14: Нужны термодатчики 18B20.  Когда будут?

AVL4GITHUB, 2017-04-03:


AVL4GITHUB, 2017-04-04:
Когда?

7FKS, 2017-04-15:
Поддерживаю. 
Использую именно эти датчики, т.к. они недорогие и с широким диапазоном изменения температуры. Ну и используется только один pin на несколько датчиков.
Именно поэтому вопрос актуален.

tretyakovsa, 2017-04-19:
В каком виде хотите иметь данные с термодатчиков и сколько их вам требуется.

7FKS, 2017-04-19:
@tretyakovsa Немного не понял вопрос "в каком виде?" Возможно имеете ввиду это:
по шине I2C сканируются датчики, определяется их количество и выводятся порядковые номера в виде столбца с параметрами температуры:
Датчик 1:    +10С
Датчик 2:    -5С
Датчик 3:    +12С
...
с возможностью изменять имена датчиков.

Мне пока требуется максимум 2 датчика на одном модуле ESP: 1 датчик в доме рядом с окном, другой за окном на улице. 
Область применения, где на минимальном расстоянии может быть расположено несколько датчиков, в различных системах ограничивается только фантазией, например: ПК (если кому-то жизненно необходимо знать температуру процессора, моста, винчестера и видеокарт), авто (температуру двигателя, радиатора, на улице и в салоне), самогонный аппарат (я, правда, не знаю его строения :) ...

AVL4GITHUB, 2017-05-08:
+2

symparua, 2017-06-06:
В прошивке easy esp есть поддержка этого датчика. Может оттуда вытянуть код?

symparua, 2017-06-06:
Тоже плюсую

symparua, 2017-07-11:
Вроде работает. График рисует с шагом 1 градус. [image]
Хорошо бы иметь еще постоянное отображение цифрового значения текущей температуры с точностью до десятых...
Спасибо!

renat2985, 2017-07-12:
@symparua, мысль хорошая, обдумаю.

AVL4GITHUB, 2017-09-22:
Температуру отображает только за последнюю минуту. 
Нет возможности видеть температуру за последний час или день или неделю, месяц?

renat2985, 2017-09-22:
Пока нет такой возможности. Как вариант вы можете сокрасить частоту обновления с 1 раза в 3 секунды, например на 1 раз в 30 секунд.
И увеличить число точек на графике, таким образом вы сможете отследить данные за более длинное время. Но с страницы уходить и обновлять ее нельзя.
[image]

О других возможностях в графиках вы можете узнать из этого видео:
https://www.youtube.com/watch?v=ooD_2luC4W4&t=19s&index=17&list=PL6NJTNxbvy-IPTDQk8XjTV41oRrFafrRi

AVL4GITHUB, 2017-09-25:
Спасибо.
Если в прошивке оставить только один язык, то не освободится ли место для накопления данных о температуре, к примеру.

renat2985, 2017-09-25:
@AVL4GITHUB  Освободится, но это не существенно. Каждый lang файл занимает всего около 2Кб. Вся файловая система сейчас весит около 150Кб. bin файлы которые мы создаем скомпилированы для ESP с 1Мб, flash памятью. Там выделяется 256Кб на файловую систему, таким образом у нас свободно еще около 100кб.

Если скомпилировать для ESP с 4Мб flash памяти. Можно выделить под файловую систему уже 3Мб. Тогда без проблем там можно хранить информацию со всех ее датчиков за пол года.

Но мы пока не планировали подобную функцию. :) Сейчас работам над сценариями.

AVL4GITHUB, 2017-09-25:
Спасибо.
У меня впаяно  4 Метра флеши. Говорю, на всякий случай, может чего протестировать надо на такой памяти.

johnes33, 2017-09-29:
Рекомендую Вам, сразу же сделать возможность задавать имя датчика по его ID, иначе народ будет путаться в цифрах, да и ds18b20 висящие на одной шине иногда любят не все инициализироваться (т.е. иногда пропадают изза помех)  и по этому номера дачиков будут меняться, что не есть гуд.

И еще: для хранения данных большой временной глубины (день, неделя, месяц, год) - то можно использовать  кольцевые базы данных ([RRD](https://ru.wikipedia.org/wiki/%D0%A6%D0%B8%D0%BA%D0%BB%D0%B8%D1%87%D0%B5%D1%81%D0%BA%D0%B0%D1%8F_%D0%B1%D0%B0%D0%B7%D0%B0_%D0%B4%D0%B0%D0%BD%D0%BD%D1%8B%D1%85)) а-ля как это сделано в mrtg