tretyakovsa/Sonoff_WiFi_switch #15: Пароль на доступ

semen7907, 2017-04-09:
Хотелось бы, чтоб была возможность установить пароль на доступ к устройству(как это реализовано в др. прошивках) и к запросу на вкл/выкл реле (192.168.**.**/sonoff)

tretyakovsa, 2017-04-09:
Зачем это нужно? Хотелось бы понять причину что это даст?

semen7907, 2017-04-10:
Для безопасности) Просто имея доступ к локальной сети можно беспрепятственно включать и выключать свет..

renat2985, 2017-04-10:
Зачем в локальной сети спрашивать пароль? В локальную сеть ведь не может попасть любой желающий, там только те кому вы дали пароль от WiFi, или те кто непосредственно к роутеру подключили кабель. Не вижу в этом смысла.

tretyakovsa, 2017-04-10:
Так может стоит подумать о безопасности сети в первую очередь. Ведь если сеть будет доступна, то своровать любые данные и получить доступ к любым устройствам в сети дело времени будет там защита или нет уже не важно. Даже если сделать пароли на страницы это ничего не даст. Кодировать все запросы передаваемые по сети накладно и бестолково. Проще защищать сеть нормальным образом.