tretyakovsa/Sonoff_WiFi_switch #16: Падает web сервер

semen7907, 2017-04-13:
Заметил такую проблему, иногда становится недоступен web сервер и get запрос, при этом выключатель работает с кнопки. проблема случается через какое то время, например вечером было все хорошо, на утро web сервер недоступен.  Наблюдал это на двух sonoff 

И так же заметил, когда стал недоступен web сервер, лампочка на sonoff заморгала (видимо перезагрузился) но web сервер так и не поднялся.. 

И хотел узнать, есть ли сейчас подобная проблема у других, или может у меня что то с настройками роутера?

renat2985, 2017-04-13:
Обратите внимание, когда web сервер недоступен. В вашем списке WiFi сетей, появляется sonoff с которым проблемы? 
Мне почему-то кажется проблема из за плохого WiFi сигнала, точнее из за плохой антены в модуле sonoff. У меня были похожие проблемы, после перенес sonoff в комнату где стоит роутер и все стало работать хорошо. Попробуйте.
Пробовал эту прошивку ставить на ESP8266-01 работает во всех комнатах, без всяких проблем.

semen7907, 2017-04-14:
sonoff находится прямо у роутера, расстояние меньше метра, сегодня утром так же не мог зайти на web, ip даже не пинговался, затем зашел в роутер, в список wi-fi подключений и там был mac-адрес sonoff, нажал разъединить соединение, sonoff переподключился и снова стал доступен.. Так что это похоже происходит из за роутера? 

И есть небольшое пожелание, чтобы вы в дальнейшем добавили в таймеры функцию перезагрузки уст-ва