tretyakovsa/Sonoff_WiFi_switch #17: Таймер запросом

semen7907, 2017-04-24:
Подскажите, можно ли как нибудь создать таймер запросом? Например get или post

renat2985, 2017-04-24:
Практически все настройки в проекте настраиваются через GET.
Таймеры настраиваются следующим образом: 
http://IP/timerSave?module=relay&trigger=on&id=9450&day=All&time=10:10:10

**module=relay** - модуль на который хотите установить таймер, в будущем будит и rgb сейчас доступно только relay
**trigger=on** - on - включить, off - выключить, not - переключить
**id=9450** - идентификатор таймера, должен быть всегда разным
**day=All** - включать каждый день. Можно заменять на Mon или Tue или Wed и т.д.
**time=10:10:10** - время

semen7907, 2017-04-25:
Спасибо большое

i81, 2017-07-13:
А можно добавить таймер обратного отчета? Т.е .выключить через ЧЧ:ММ:СС

renat2985, 2017-07-13:
@i81, да мы думаем об этом. Скорее всего в ближайшие недели появится эта функция.

i81, 2017-07-13:
Ренат, как я могу поддержать проект Ваш? Есть кошелёк Я.Денег?

i81, 2017-08-10:
Ещё вопрос по таймерам: нужно сделать включение в 06:40:00 и выключение в 07:15:00 в будни. "Будней" в web интерфейсе конфигурирования размеров нет, добавляю по дням... и толь меня глючит, толь есть ограничение на кол-во таймеров... получилось максимум вроде 7 добавить, но потом они исчезают и остаются три... :(

renat2985, 2017-08-11:
@i81 Хм. Как таковых ограничение на количество таймеров нет. У меня жалюзи работает следующим образом, каждый будний день открывается в 7:15, по выходным в 9:00, закрывается в 20:20. Итого 8 таймеров. Все работает. Попробуйте скачать новую версию. И сделать похожим способом как у меня. Если не получится опишите проблему подробней.
[image]

Кошелек, YandexMoney: 410011740665305
Спасибо.

i81, 2017-08-11:
Собственно и описывать нечего...
Добавляю из мобильного Хрома, при добавлении меняю только день недели, всё остальное (время и состояние релле остаётся) рандомно (на 3, 4 иногда 5) раз пропадает предыдущий ввод при добавлении нового.  Один раз получилось добавить все (5 включений и 5 отклбчений) по-моему потом страница обновилась и осталось только 3 притом не последовательных (понедельник, вторник, пятница) включения. Добавлял последовательно...
П.с. при первом включении после отпуска был вверху воспалительный знак с сообщением об обновлении модулей - обновил,  больше предложений обновиться нет...

renat2985, 2017-08-11:
Хм, возможно проблема в браузере. Или же если у вас несколько устройств, после добавления иногда в списке (2) они меняется местами. Будьте внимательны, таким образом вы добавите таймер на другое устройство.

[image]

Зайдите в конфигурацию и посмотрите за какое число у вас прошивка.
[image]

i81, 2017-08-11:
Нет у меня одно устройство. Сейчас ютуб обрабатывает видео, записанное мною с экрана - там вообще фонтастика... 
П.с. передёрнул питание SonOff может поможет...

i81, 2017-08-11:
https://www.youtube.com/watch?v=FBqFy0uYumg&feature=youtu.be вот видео - прямо чудо

i81, 2017-08-11:
О! Спасибо за идею с браузером. Попробовал в этом новом Microsoft EDGE - там добавились с первого раза нормально...

tretyakovsa, 2017-08-11:
Спасибо за видео. Еще бы понять почему это происходит у вас и не происходит у нас. Мы только в Хроме обычно все проверяем.

i81, 2017-08-11:
Может Вам http порт покинуть до устройства? Оно ещё на тумбочке тестируется...

tretyakovsa, 2017-08-11:
Было бы здорово. Если скайп есть стучитесь. vtplus

i81, 2017-08-11:
Что-то с пробросом портов не то... интерфейс открывается, но почему-то половину не прогружается...

tretyakovsa, 2017-08-11:
telegram +79027688318

i81, 2017-08-11:
Ура!!!
Только вот как по мне еще бы просто кнопку включить на заданное время т.е. без явного указания времени старта (слева на скринах у Вас)

tretyakovsa, 2017-08-12:
Как это оформить? Кнопку включение на время? Есть мысли?

i81, 2017-08-12:
В разлеле таймеров, подросдел таймер обратного отсчета - просто поле ввода времени и правее от него кнопка включения. Получается задал время и включил. Что бы люди не путалсись можно время вводить не по формату ЧЧ:ММ:СС, а просто в минутах (186 минут)

i81, 2017-08-12:
Или написать текстом "Выключить через [ЧЧ:ММ:СС] {кнопка Старт}"