tretyakovsa/Sonoff_WiFi_switch #18: Обновлять устройство стало проще!

renat2985, 2017-04-24:
Теперь когда появляется новая прошивка на странице появляется красный треугольник, нажав на который устройство само скачает новые bin файлы и обновит себя, включая и файловую систему.
[image]

Так же на странице setup.htm вы можете самостоятельно установить нужную вам прошивку и файловую систему. Устройство все скачает и установит само вам нужно только выбрать нужную вам прошивку или файловую систему.
[image]

renat2985, 2017-06-02:
Странно, попробовал прошить по вашей ссылки все работает.
Рекомендуется обновлять систему нажимая на красный треугольник, тогда обновляется сразу и файловая система (таймеры не стараются) и прошивка на актуальную версию. Обновление с страницы setup можно делать, но это больше для разработчиков, не стоит этим пользоваться.

Делать обновление нажимая на красный треугольник с главной страницы пробовали?

2007-i, 2017-06-02:
Не появляется красный треугольник. Поэтому думаю нужно наверное обновить бинарниками через флешер.

renat2985, 2017-06-02:
Свяжитесь со мной в Skype: renat2985

2007-i, 2017-06-02:
Большое спасибо Ренату за оказанную помощь!

renat2985, 2017-06-02:
Итог. Если после прошивки какие-то страницы не получается открыть, появляется надпись "File not found"
Качаем [этот bin файл](https://github.com/tretyakovsa/Sonoff_Loader/raw/master/Sonoff_Loader.ino.generic.bin) заходим на страницу: http://IP_устройсва/update и обновляем esp скачанным ранее bin файлом.

acrixl, 2017-06-07:
а как отключить эту фичу?

renat2985, 2017-06-08:
@acrixl зачем?

acrixl, 2017-06-08:
не люблю обновления просто

renat2985, 2017-06-09:
Зря, тем самым вы многого себя лишаете. Новые прошивки работают быстрее старых. Не так давно стало возможным редактировать многие страницы, можно самостоятельно удалять и добавлять элементы. Вчера добавили новый дизайн, теперь их 3. Скоро появятся сценарии. Сегодня или завтра выложу еще обновления, после чего в default дизайне появятся вот такие графики
[image]

Обновления это хорошо и их стоит делать.

acrixl, 2017-06-09:
это конечно всё замечательно, но стабильность важнее любых обновлений

tretyakovsa, 2017-06-09:
Тогда, просто не обращайте внимание на значок.

KlaydT, 2017-06-16:
День добрый.
обещал отписаться о обновлении через "треугольник". Обновился. Лог1. видно какие версии заливались. Конфиг родной затерся. Модуль поднял свою сеть.  бекап закачать - не получилось, поскольку сеть то своя собственная. Перегрузился. Лог2. тот же самое. Модуль поднимает сеть 192.168.4.0/24. По адресу 192.168.4.1 - FileNotFound. по адресу http://192.168.4.1/update можно залить бинарник. Сгенерировал свой со старого скетча рабочего. Залить через /update не смог. Страница не ответила. В «мониторе порта» мигает «15 minutes». Восстановил старый скетч через Arduino IDE. 1.8.1 Скетч и файлы.
[log2.txt](https://github.com/tretyakovsa/Sonoff_WiFi_switch/files/1080513/log2.txt)
[log1.txt](https://github.com/tretyakovsa/Sonoff_WiFi_switch/files/1080512/log1.txt)

renat2985, 2017-06-17:
@KlaydT, спасибо за информацию.

i81, 2017-09-13:
Добрый вечер. 
Друзья,  обновился сегодня через восклицательный знак, и после этого перестал грузиться интерфейс. 
[image]

renat2985, 2017-09-13:
Плохо встала файловая система.
Попробуйте обновится через Sonoff_Loader https://github.com/tretyakovsa/Sonoff_Loader
С страницы ip/update
Смотрите видео: https://youtu.be/D0Fu8etyVg0?list=PL6NJTNxbvy-IPTDQk8XjTV41oRrFafrRi&t=322

Далее снова нажимаем на красный треугольник на главной странице и обновляемся до последней версии.

i81, 2017-09-13:
Помогло. Спасибо!