tretyakovsa/Sonoff_WiFi_switch #20: Кнопка - включить\выключить

Berkseo, 2017-05-25:
на странице /relay.htm не обновляется кнопка. если на кнопке написано - включить, кликнув на кнопку, включение произойдет, но на кнопке останется надпись - включить

Berkseo, 2017-05-26:
и эта же детская болячка во всех остальных устройствах (rgb, жалюзи)

renat2985, 2017-05-26:
Мы сейчас полностью переписываем код. Скоро все заработает.

Berkseo, 2017-05-26:
ясно, самое главное вы в курсе. кстати по поводу mqtt, думаю удобно бы было через страницу настроек прописывать свои пути топиков, а не задавать их в коде. конечно их легко поправить, но многие же будут пользоватся вашими бинарниками.
---
Отправлено с iPhone

renat2985, 2017-05-26:
Скоро через веб можно будет редактировать полностью все на страницах. Добавлять, удалять, редактировать элементы. Думаю и с mqtt что то придумаем. В начале недели я постараюсь сделать обзорное видео.

2007-i, 2017-05-27:
> Скоро через веб можно будет редактировать полностью все на страницах. Добавлять, удалять, редактировать элементы. Думаю и с mqtt что то придумаем.

Это было бы то что нужно! Ваша прошивка станет более универсальна в применении.

tretyakovsa, 2017-05-28:
К этому и стремимся :)

renat2985, 2017-05-29:
Выложил обещанное видео: https://www.youtube.com/watch?v=71DuzSbWQLI&index=1&list=PL6NJTNxbvy-IPTDQk8XjTV41oRrFafrRi
Все это планируется запустить на новом асинхронном веб сервере для esp8266, он значительно быстрее того что мы сейчас используем. Когда в нем разберемся, все страницы переведем на такое веб редактирование.

alxnovo, 2017-09-19:
> и эта же детская болячка во всех остальных устройствах (rgb, жалюзи)
Такая же ситуация (говорю про жалюзи) на главной странице и на странице управления кнопка либо всегда "открыть" или "закрыть" при этом нажимая на кнопку переключение происходит
P.S. ESP Witty Cloud подключены к GPIO 12 и 13 модуля из 2-х реле (бочонки через оптопары) которые включаются по логическому нулю.

renat2985, 2017-09-20:
@alxnovo дайте модель роль ставней и схему подключения мотора. А как вы останавливать его собираетесь, если у вас нет счетчика оборотов?

alxnovo, 2017-09-20:
@renat2985 извини, неверно выразился. При нажатии кнопки реле перещёлкиваются, а стастус кнопки не изменяется - она либо всегда "открыть", либо "закрыть". Если находишься не у объекта которым управляешь, то не понимаешь в каком состоянии он сейчас находится, да и просто неудобно.

Подключаю [реле](http://avrobot.ru/images/product_images/popup_images/4014_0.jpg) к стандартным пинам указанные в прошивке (12 и 13). Пин к счётку оборотов никуда не подключаю, кнопку внешнего управление тоже не подключена (если правильно понял её к GPIO0 нужно подключить). При нажатии кнопки в интерфейсе реле переключаются, то первое реле включится и выключается второе или наоборот (кнопка статус при этом не меняет, как писал выше), но потом включенное реле не выключается т.к. счётчик оборотов не даёт команды "jalousiestop".
Не хватает в настройках отключение реле не только по счётчику оборотов, но и по таймеру (у меня рольставни открываются и закрываются секунд за 20, другие аналогичные устройства могут это делать дольше, зависит от габаритов).

Останавливается двигатель за счёт концевиков в момент когда ролета доходит до крайних положений. Но всё равно небезопасно оставлять китайские реле (бочонки) под напряженим, поэтому нужно полное отключение по таймауту после начала движения. 
На примере: 
 - нажал на кнопку "открыть" -> включилось реле подключенное к обмотке двигателе "вверх" - включился таймер -> ролета поднялась и концевик отключил питание обмотки на двигателе -> таймер досчитал до установленного времени и отключил реле -> кнопка поменяла статус на "закрыть"
 - нажал на кнопку "закрыть" -> включилось реле подключенное к обмотке двигателе "вниз" -> включился таймер -> ролета опустилась и концевик отключил питание обмотки на двигателе -> таймер досчитал до установленного времени и отключил реле -> кнопка поменяла статус на "закрыть"

Ещё, на всякий случай хорошо бы сделать задержку между выключением одной реле и включением другого, чтобы случайно фаза не пошла на обе обмотки двигателя. Вдруг одно реле ещё не разъединится, а другое уже включится...не знаю с какой скоростью китайские реле меняют свои положения.

Двигатели на ролетах ALUTECH серии AM, вот на них [руководство](http://www.alutech-group.com/upload/iblock/c39/vnutrival-nye-elektroprivody-serii-am0-am1-am1-pp-am2-am2-pp.pdf), по такому же принципу работают двигатели других производителей и используются также в гаражных воротах и шлагбаумах, да и те же элктрошторы и электрокарнизы работают с концевиками [пример](http://xn--80aoccfkkev.xn--p1ai/) только я не понял в этом примере как они определяют где находятся шторы во время движения - либо по оборотам, либо примерно по времени, либо ещё как-то.

alxnovo, 2017-09-20:
Вот [снимок экрана](https://drive.google.com/open?id=0B3EVzTkoj_atcFBkVHZETndOYVU) показывающий что как не нажимай кнопку "закрыть" она не меняется хотя реле перещёлкиваются верно

Кстати, в новой сегодняшней прошивке обновлённой через интерфейс прошивки, через появившийся красный треугольник у названия устройства перестал работать таймер

renat2985, 2017-09-20:
@alxnovo кнопка не меняется потому что вы не установили и не настроили датчик оборотов, кнопка поменяет свой статус только после того как получит нужное количество оборотов.
По поводу таймеров, спасибо, посмотрю.

alxnovo, 2017-09-20:
@renat2985 Установил в конфигурации GPIO4 как счётчик оборотов, у [Witty Cloud](http://www.schatenseite.de/wp-content/uploads/2016/04/160422_witty_cloud_01-1.jpg) на этом пине установлена кнопка и этой кнопкой "эмулирую" обороты, релюшка выключается, но кнопка статус не меняет.
С таймерами я не знаю что случилось. Добавляешь первый - добавился, второй на основании первого (когда сразу следующий вводишь, то данные от предыдущего в полях ввода остаются) с небольшими изменениями минут, то он в списке не появляется, затем вводишь точно такой же, появляются все три. Удаляешь первый таймер, а в списке остаётся удаляемый первый и один из последующих дублёров. И каждый раз алгоритм появления-удаления на странице разный. Штук пять ввёл таймеров (тестировал работу таймера, хотел уже на рольставни ставить, чтобы утром всё-таки проснуться под солнышко, а не просыпаться в середине дня думая что ещё ночь) и при удалении, когда осталось 2 таймера ESP'шка не выдержала и зависла, через 5 минут пришлось перезапустить её, желание отвиснуть у неё не появилось. Кстати, аптайм есть в интерфейсе?
Другим способом обновится смогу только завтра, отпишусь что да как, если будет актуально.

На странице таймеров часы не показывают "0" если в секции цифра меньше 10-ти. Вот [скрин](https://drive.google.com/open?id=0B3EVzTkoj_atVWotLXgxSDhvNFE)

renat2985, 2017-09-20:
По поводу оборотов, скорее всего вы забыли указать количество оборотов:
[image]

По поводу таймеров, такое бывает когда плохо встала файловая система. Мы сейчас стараемся от этого избавится. Посмотрите это виде, здесь я рассказываю о способах обновления: https://www.youtube.com/watch?v=D0Fu8etyVg0&index=18&list=PL6NJTNxbvy-IPTDQk8XjTV41oRrFafrRi

Так же посмотрите другие видео ролики, uptime здесь: /page.htm?starting

alxnovo, 2017-09-21:
@renat2985 Обороты я ставил "2" чтобы подключенную кнопку на пин "оборотов" много не нажимать. Нажимаю открыть - реле включается - я 2 раза нажимаю на физическую кнопку "обороты" - реле выключается - кнопка в интерфейсе статус не поменяла.
По поводу таймера понял, перезалью вечером прошивку и проверю
В голову не приходило что в Starting log можно смотреть аптайм :) привык что под аптаймом подразумевается сколько прошло времени после загрузки.
Видео я практически все смотрел, только в них разрозненная информация и не всегда относится к актуальной прошивке, всю информацию из них запомнить сложно если всё что происходит в видео не повторить на практике...прочувствовать на своей шкуре.

renat2985, 2017-09-21:
@alxnovo давайте сегодня вечером пообщаемся в skype, покажите свой рабочий стол и кнопку. Очень странно что она у вас не меняется. Мой skype: renat2985

alxnovo, 2017-09-22:
Отчитываюсь. Залил прошивку и EEPROM через NODEMCU, всё сбросилось. 
`неудобно что в режиме точка доступа в Serial не пишется IP по которому можно сделать первоначальную настройку и так же в Serial'е нет IP когда его делаешь статическим`
На странице таймеров "пляски" с вводом и удалением таймеров пропали, но таймеры не работают - в назначенное время жалюзи не "открываются" и не "закрываются"
Часы отображаются в без "0" если число меньше 10 на страницах ?setup и ?timer, НО только если если секунд меньше 10, если секунд больше 10 то и часы и минуты отображаются верно...с нулями, если же подождать пока секунды обновляться, то и минуты показываются некорректно, перезагрузка страницы исправляет этот недостаток. На примере...открываем одну из страниц с часами видим 23:06:37 ждём когда секунды обнулятся и видим 23:7:2 обновляем страницу - получаем 23:07:5 т.е. минуты после обновления пришли в норму, а секунды остались без нуля перед цифрой

alxnovo, 2017-09-22:
Видео с кнопкой жалюзи https://youtu.be/RmjDrkuHoKU
Видео с косяком часов https://youtu.be/OiodbIm5iS4

renat2985, 2017-09-22:
По поводу времени которое тикает на странице, Ок исправлю. Это не ошибка, когда вы открываете страницу JavaScript подхватывает время и уже сам отсчитывает его, не обращаясь к ESP, поэтому есть там нолики или их нет, в общем-то можно и не обращать внимание.

По поводу красного треугольника, обновитесь если он есть. На главную страницу проекта откуда вы скорее всего и скачали bin файлы, мы выкладываем только важные обновлениями. То есть, если были какие-то мелкие фиксы, мы не обновляем файлы на главной страницы, но у вас появится треугольник с предложением скачать более новую версию с папки /build/

По поводу кнопки Закрыть\Открыть, она меняется только после обновления страницы или после нажатия на нее. Если вы щелкаете обороты она не поменяет свой статус без рефреша страницы. Я подумаю в чем может быть дело. Спасибо за информацию.

alxnovo, 2017-09-28:
@renat2985 

> В режиме точки доступа вы в браузере можете написать любой домен например http://11.ru и попадете на устройство, если не получилось пишите http://192.168.4.1 это его статический IP.

Всё равно удобнее когда есть подсказка в Serial. И не только когда делаешь первую настройку, но и когда указал статический адрес. Ладно, когда с одним модулем работаешь помнишь его адрес, а когда несколько? Или у меня часто получалось что не сохранялся статический адрес и приходилось каждый раз лезть в роутер смотреть DHCP

> По поводу кнопки Закрыть\Открыть, она меняется только после обновления страницы или после нажатия на нее. Если вы щелкаете обороты она не поменяет свой статус без рефреша страницы. 

Повторюсь...в page.htm вставил костыль `<meta http-equiv="Refresh" content="15"/>` чтобы страница управления автоматически обновлялась и статус кнопки, соответственно, тоже.
Как обновлять только элемент я ещё не знаю, поэтому пока только страницу

Как я вижу алгоритм обновления статуса: при нажатии кнопки "открыть" или "закрыть" кнопка блочилась (не могла нажаться, иначе если связь с модулем плохая - из-за чего жалюзи не сразу начинают двигаться, возникает желание эту кнопку ещё понажимать чтобы, наконец-то, жалюзи начали двигаться и когда куча команд дойдут до жалюзи, они подрыгаются в разные стороны и начнут двигаться в том направление в котором кнопка была нажата последний раз), например, "Loading..." и кнопка начинает обновляться каждую 1-2 секунды пока жалюзи двигаются (пока не достигло кол-во оборотов или время на движение не истекло), потом меняется статус кнопки и обновление убивается.
Ещё лучше если будет не "Loading..", а "СТОП" который можно нажать и остановить движение. Опять же при нажатии "СТОП" статус кнопки меняется на "Открыть"/"Закрыть" и при дальнейшем нажатии движение продолжилось в том направлении в котором было прервано.
И третий вариант :) Если нажата кнопка "СТОП" то появляются две дополнительные кнопки "Открыть"/"Закрыть" и тогда можно будет выбирать направление после остановки.

> По поводу таймеров, такое бывает когда плохо встала файловая система. Мы сейчас стараемся от этого избавится. 

Оказывается таймеры работают, это я невнимательно смотрел видео где говорится что для жалюзей в таймерах пока работает только действие "Перекл." а я пытался создать таймер на "Открыть", "Закрыть".