tretyakovsa/Sonoff_WiFi_switch #21: повторное подключение к wifi

Berkseo, 2017-05-25:
Думаю очень полезной функцией будет, если устройство при разъединение с вайфай и переключения в режим AP, через какое то время пробовало опять подключится к домашнему wifi. возможная ситуация: работает 5 реле, 5 выключателей света, чего нибудь еще, отключается роутер, все устройства переходят в режим AP? что бы их вернуть в домашнюю сеть, их придется в ручную перезапускать.

Berkseo, 2017-05-25:
///в коде что то есть, но не рабоче совсем

tretyakovsa, 2017-05-27:
Да я начинал делать, но решение кривое. Со временем поправим.

i81, 2017-08-10:
Не нашёл более похожей темы к моему вопросу - пишу тут. Строка ввода "Имя устройства" в секции настроек "Подключение к Wi-Fi роутеру" - это неправильный перевод пароля сети? 
Или как вообще подключить модуль к сети? - Имя выбирается из списка,  а где вводить пароль? Есть ли ограничение на пароль сети?

i81, 2017-08-10:
Похоже я прав и эта строка ввода как раз таки для пароля беспроводной сети, вот только беда - мой длинный пароль не принимается - строка выделяется красным цветом и не сохраняется... 
Как можно ввести длинный пароль от беспроводной сети?

tretyakovsa, 2017-08-10:
Если роутер будет выключен, и затем включится. А модуль до этого был подключен, связь востановится. Другое дело когда выключили свет и модули не успевают подключится к роутеру и переходят в режим AP.
Чтоб этого не случилось нужно расширить время соединения в соответствии с временем загрузки роутера.
Для этого нужно задать    "type": "wifi 20 13" где первая цифра это количество попыток подключения, вторая ножка индикации подключения.

i81, 2017-08-10:
Уважаемый, а мне ответите про пароль от сети wi-fi?

renat2985, 2017-08-10:
@i81 нажимаем Ctrl+M ищем такие строчки и цифру 20 меняем на нужную вам, это длинная пароля. Не забудьте нажать на "Save"
[image]

i81, 2017-08-10:
Вроде получилось. Спасибо большое. Только почему-то при подключенной к сети не получалось сдалать, отключил на роутере гостевую сеть, перезапустил sonoff, подключился к его сети и из под неё, опять исправил значение 20 на 60 (у меня длина 51) и вроде все заработало...

renat2985, 2017-08-11:
Хм. Странно. Да и поле обновления прошивки все ваши изменения затрутся на дефолтные. Вам придется их заново внести. Вы можете нажать на облако 
[image] 
и сохранить их себе на компьютер. Потом снова заменить после обновления.

i81, 2017-08-11:
Так она же у меня и так крайняя...

i81, 2017-08-11:
Скажите пожалуйста, стоит ли ожидать в ближайшем будущем появления таймера обратного отсчета?

semen7907, 2017-09-12:
Подскажите, есть ли запрос для переподключения к точке wifi ?

renat2985, 2017-09-12:
@semen7907, вы имеите ввиду GET запрос? На переподключения нет. Как вариант можно использовать рестарт GET: /restart?device=ok
Но зачем вам это? ESP отваливается?

semen7907, 2017-09-12:
Да, у меня 2 модуля sonoff и они часто отваливаются помогает только переподключение к wifi я думаю, если получится, добавить этот запрос в ddns чтобы он переподключал модуль каждые 15 минут.

renat2985, 2017-09-12:
С sonoff у меня к сожалению такие же проблемы были. А вот с esp8266 12 и 07 таких проблем не замечал.
Да и за какое число вы используете прошивку? Попробуйте обновится на последнею, мы в ней сделали некоторые изменения в wifi. Возможно у вас будет стабильнее работать sonoff. Попробуйте, и пожалуйста отпишитесь.

semen7907, 2017-09-12:
Хорошо, обновлюсь, проверю, но все же думаю для sonoff было бы не плохо добавить пункт "переподключение к wifi" чтобы например каждые 2-4 часа модуль переподключался к wifi

renat2985, 2017-09-13:
Сегодня выложил новые бин файлы, посмотрите видео и попробуйте https://youtu.be/rgqQlBs-kYc
Отпишитесь помогло ли.

semen7907, 2017-09-13:
@renat2985 
Хм, обновился но почему то у меня не появилось это поле, и стало 2 кнопки "сохранить"
[url=https://imgbb.com/][img]https://image.ibb.co/e1FyBv/22.png[/img][/url]
[url=https://ibb.co/gymida][img]https://preview.ibb.co/eQA9Ja/image.png[/img][/url]

renat2985, 2017-09-13:
Плохо встала файловая система.
Попробуйте обновится через Sonoff_Loader https://github.com/tretyakovsa/Sonoff_Loader
С страницы ip/update
Смотрите видео: https://youtu.be/D0Fu8etyVg0?list=PL6NJTNxbvy-IPTDQk8XjTV41oRrFafrRi&t=322

semen7907, 2017-09-13:
Спасибо, уже получилось, обновил прошивку еще раз через настройки разработчика

renat2985, 2017-09-13:
Ставьте тогда наверное сразу на 2минуты коннекта. Расскажите потом помогло ли.

semen7907, 2017-09-13:
Странно, 2 раза перепрошил реле, и когда нажимаю кнопку "перезагрузить устройство" то реле начинает включаться и выключаться (как часы тик-так, тик-так), и при перезагрузке все тоже самое, это что то с ФС?

renat2985, 2017-09-13:
@semen7907 значит вы не внимательно смотрели первое видео которое я вам сегодня давал.

semen7907, 2017-09-14:
@renat2985 Я уже пробовал всяко прошивать, раз 10, и через loader и 2 bin файла, такая проблема только на прошивке от 13.09, потому что я ставил от 11.09 там такой проблемы нет.

проблема похоже такая: при перезагрузке реле просто начинает щёлкать, вкл/выкл и так постоянно 
возможно что это случается после изменения настроек времени подключения к wifi сети

renat2985, 2017-09-14:
В видео я говорил о "BLINK LED GPIO", говорил о том если у вас там будит указан GPIO12 то при подключение к WiFI будет щелкать реле. Скорее всего у вас там именно GPIO12 и стоит. Смените его на GPIO13, тогда у вас будет мигать LED лампочка на sonoff.
[image]

semen7907, 2017-09-14:
все же какая то проблема с подключением к wifi есть, если я отключаю модуль от питания , а затем снова его включаю, то в 80% он не подключится, то есть просто моргает лампочка в течении 2х минут, а потом просто горит, но если я перезагружу роутер и в этот же момент модуль, то он подключается без проблем..

renat2985, 2017-09-14:
С миганием разобрались? :) С каким сигналом esp видит ваш роутер? Может проблема в вашем роутере? Когда я подключаю более 10 esp устройств. У меня роутер начинает неадекватно себя вести. Периодически на телефоне и планшете пропадает wifi на несколько минут. Начинает медленней работать интернет. Возможно у вас старенький роутер и ему тяжело раздавать wifi большому количеству девайсов. Попробуйте отключить WiFi в телефонах, планшетах.

semen7907, 2017-09-14:
ну роутер нормальный, к нему только телефон подключен, модуль от него в 30 см., сигнал 90-100%

semen7907, 2017-09-23:
Стоит прошивка от 11 числа, похоже что на sonoff работает стабильно, 4 суток модуль работал без проблем и был доступен, так же подключал второй модуль с прошивкой Tasmota на этой прошивке модуль держится часов 5-8 и потом становится недоступен

sultan5448, 2017-12-29:
переподключение устройства из режима точки доступа на подключение к  роутеру нужно. через промежутки времени. приходится в ручную дергать.