tretyakovsa/Sonoff_WiFi_switch #24: Удаленное управление.

SurovyjElektromonter, 2017-07-12:
Вчера наткнулся на прошивку. Считаю ее лучшей из всего что я нашел тут. В коде мало разбираюсь. Единственное что понял- так это то, что без навыков программирования дистанционно управлять простому смертному скорее всего не получится. Зная немножко пхп, мускул и стили сегодня решил упростить управление устройствами через vk api. Что мы имеем? На данный момент готов макет приложения и база данных. Чего не хватает? Не знаю как установить в прошивке интервал, чтобы например каждые 5000мс устройство тыкало на ссылку https://мойсайт.ру/передаватьвэтотфайл.php?esp_id=номересп&pass=парольустановленныйвпрошивке&stat=состояние устройства..... и соответственно распознавало data, которое отсылает этот файл с vps. Например если в приложении нажали relaynot, оно пишется в базу, при запросе команда отправляется из базы в устройство через data и убирается из базы. 
Вот ссылка на то что получилось https://vk.com/app5508608
Хотелось бы узнать где лучше пофиксить в прошивке (подробно и детально), чтобы привязка устройств к социальным сетям и приложениям заработала. Спасибо.
[image]

renat2985, 2017-07-13:
Какая-то бессмысленная у вас идея. Вы можете поместить свою ссылку в DDNS на странице "Конфигурации", таким образом каждые 15 минут на вашу ссылку будет идти GET запрос.

closed 2017-07-13 by renat2985

SurovyjElektromonter, 2017-07-18:
Но мне необходимо чтобы гет-запрос шел с интервалом в 10 секунд. Это реально сделать в настройках? И какие параметры передаются в гет-запросе? Распознает ли родная прошивка data, которые будет возвращать страница?

renat2985, 2017-07-18:
Для необходимой вам функции нужно корректировать код. Я считаю это бесполезной затеей. Если сможете переубедить, возможно нужная вам функция появиться.