tretyakovsa/Sonoff_WiFi_switch #25: Возможность задать состояние реле при подачи питания на устройство.

i81, 2017-07-13:
Друзья, а можно как-то изменить состояние реле при включении самого устройства SonOff в сеть питания? 
Сейчас - оно разомкнуто, а мне бы хотелось, что бы оно было замкнуто.

renat2985, 2017-07-13:
Пока нельзя. В будущем думаю что ни будь придумаем.

i81, 2017-07-13:
очень жаль - очень востребованная для меня функция...

AleksandrGromov, 2017-10-13:
Функция появилась в последних прошивках.
В настройках модуля можно настроить  статус реле при включении

Paly4, 2017-10-13:
Статус реле меняется только в вэб , а по факту не включается
пробовал на rgb

renat2985, 2017-10-13:
@Paly4, спасибо. Разберемся. Но после рестарта у вас не включится выбранный вами ранее режим, цвет, скорость. Т.к. эти настройки не сохраняются. Включится дефолтный желтый цвет.

Paly4, 2017-10-13:
да ну хоть так , пусть будет дефолтный
еще подскажите , куда подключается кнопка , если собирать на esp8266, в проекте , по схеме не понял

renat2985, 2017-10-13:
@Paly4, кнопка подключается всегда к GPIO 0, поменять это нельзя. Кнопку можно настроить на включение\выключение rgb, реле, жалюзи
[image]

[image]

shady2k, 2017-11-02:
@AleksandrGromov. вы написали, что появилась возможность через веб-интерфейс установить статус реле при включении. Подскажите, пожалуйста, где это в настройках? Все облазил, нет у меня такого пункта. Прошивка от 16.10.2017.