tretyakovsa/Sonoff_WiFi_switch #26: Новые функции

renat2985, 2017-07-16:
# Время от времени буду выкладывать сюда фотографии с новыми возможностями.

Теперь на страницы можно добавлять любые сайты.
[image]

renat2985, 2017-07-31:
Теперь на страницах page.htm можно редактировать отображение графиков и скорость обновления.
Документацию и примеры можно посмотреть здесь: https://gionkunz.github.io/chartist-js/examples.html

[image]

renat2985, 2017-08-22:
Добавил новые [видео ролики](https://www.youtube.com/playlist?list=PL6NJTNxbvy-IPTDQk8XjTV41oRrFafrRi)

vobis77, 2017-10-17:
А возможно ли сделать инвертацию выхода на реле для использования вот таких SSR relay?

vobis77, 2017-10-17:
[image]

renat2985, 2017-10-17:
@vobis77 создавайте новые темы, этот issues совсем о другом.