tretyakovsa/Sonoff_WiFi_switch #27: Параллельное управление ИК пультом

ilya89vet, 2017-07-23:
Дорогие разработчики сделайте пожалуйста управление с помощью ИК пульта! Очень необходимо иметь резервный вариант управления.
Вот пример как это реализовано https://youtu.be/XG_H1fZJbMY?t=419 (на видео не последняя версия)

renat2985, 2017-07-23:
Сделаем. Обязательно сделаем, но пока не знаю как скоро. Возможно если будет больше пожертвований это будет быстрее. ;)

tretyakovsa, 2017-08-14:
Можете подробно описать, что вы имеете в виду.

ilya89vet, 2017-08-14:
Включение или отключение реле с помощью инфракрасного пульта, причём чтобы
пульт можно былозапрограммировать любой в веб-интерфейсе.
Всё это неплохо реализовано в проекте Алексея Морозова ссылку на который я
предоставил в сообщении.

Ветров Илья

grossbel, 2017-08-31:
Друзья, для дома лучше RF пульт, не надо целиться в выключатель или еще какой либо приемник, очень грамотно сделано в сенсорных выключателях SESOO и им подобным, можно запрограммировать все раздельно.
Один пакет (кнопка) от пульта только включает 
Второй пакет только выключает
Третий пакет переключает состояние (то есть одной кнопкой ты можешь и включить и выключить)

При этом допустим на включение можно записать несколько пакетов от пультов (больше 8 не было возможности попробовать), RF пульт можно использовать на базе ev1527, на алике их завались на любой вкус и цвет, протокол открыт и описание найти не проблема.

tretyakovsa, 2017-09-01:
Для дома лучше голосовое управление. Хотя RF приемник уже встроен, как только менеджер скриптов будет сделан можно будет использовать. Тот же менеджер скриптов задерживает выход IR приемника и передатчика.

grossbel, 2017-09-02:
А теперь представим картину, супруга спит, и ты со своим басом: Дом, спальня, нижний свет выключить. В ответ: команда не распознана, попробуйте еще раз. Любимая проснется и расскажет столько много и про дом и про тебя, еще и маму твою вспомнит. Пульт проще, взял в руки волшебную палку с кнопками, нажал кнопку выключить все и перевел систему в режим сна,

renat2985, 2017-09-02:
Какая у вас жена суровая. :) Все же управление голосом я тоже считаю удобнее. У меня есть пульт который включает led подсветку, стараюсь использовать ее в качестве ночного света. Но пульт не всегда в том месте где вам он нужен. Вот легли вы на диван, а пульт на столе. Мне лень вставать и идти к столу, или же вы за компьютером, нужно включить led, а пульт то на диване. А иногда у пульта "ножки вырастают", то жена его переложит, то сынишка запрячет. Это неудобно. Куда проще сказать "Ок google, включи свет" и не заморачивать себе голову. А жену, бросать вам надо. :)

tretyakovsa, 2017-09-02:
:) Будет вам управление и ИК пультом и 433мГ. А пока пользуйтесь IoT-Manager.

zon2004, 2018-12-12:
Подскажите пожалуйста как подключить Ir приемник,
Подключил выход к gpio0, включил галку в конфигурации, в сценарии добавил код кнопки (надеюсь правильный), но ничего не происходит

renat2985, 2018-12-12:
@zon2004 скриншоты покажите.

tretyakovsa, 2018-12-13:
@zon2004 Кнопку отключили на GPIO0? После подключения смотрите приходит ли код в /config.live.json
Вот пример ответа {"wifi":3,"dbm":-52,"voice":null,"time":"11:00:10","relay1":0,"relay2":0,"relay3":0,"weekday":"Thu","A0":200,"irReceived":"20df8e71"}
где irReceived":"20df8e71" код поступивший с пульта. При коротком нажатии.
И какой у вас пульт?

ilya89vet, 2018-12-13:
К нулевому пину лучше не подключать, так как он служит для перевода esp в
режим прошивки! Лучше всего повесить на 2 пин, при нажатии на пульт
светодиод будет моргать.

PS.  А вообще я эту прошивку не использую т.к. она очень сырая и проект
заброшен авторами! Если нужна простота и стабильная работа, то я советую
прошивку Алексея Морозова, а если нужна навороченность то советую проект
Sonoff-Tasmota, я сижу сейчас на ней - функционал просто бешенный!

zon2004, 2018-12-13:
@tretyakovsa Спасибо, вижу что коды получает, но каждый раз разный на одной и той же кнопке,
пульт от tv lg

zon2004, 2018-12-15:
ничего не подскажете? . может приемник как то хитро подключается, или нужен др пульт?

tretyakovsa, 2018-12-15:
@zon2004 
Проверьте частоту приемника ИК они бывают на разные частоты и нужно чтоб соответствовало. На пульте буковки какие написаны или телевизора марку напишите.