tretyakovsa/Sonoff_WiFi_switch #28: Откат на оригинальную прошивку SonOff

i81, 2017-08-10:
Друзья, а кто может подсказать как откатиться на стоковую прошивку SonOff? Очень не хватает таймера обратного отсчёта, а он в оригинале есть...

tretyakovsa, 2017-08-10:
Взять стоковую прошивку и залить :) или дождаться таймера обратного отчета у нас. Если есть идеи как его удобнее оформить в интерфейсе пишите.

i81, 2017-08-10:
Так,  а чего фантазировать? - начните с http запроса, а интерфейс уже и позднее можно придумать...

i81, 2017-08-10:
А где "взять" стоковую прошивку?

renat2985, 2017-08-10:
@i81 насколько я знаю ее не существует в свободном доступе. :)

i81, 2017-08-10:
От это сюрприз:( игла вся надежда на Вас, что Вы сделаете таймер обратного отсчета...

alexandersciv, 2017-10-16:
**@renat2985**  Получилось реализовать таймер обратного отчёта?

renat2985, 2017-10-16:
@alexandersciv, нет. От таймеров в ближайшем будущем вообще избавимся. Время включения\выключения можно будет настроить в сценариях. О обратном отсчете пока следует забыть.

tretyakovsa, 2017-10-16:
Почему забыт?. Как раз перенос таймеров, через алгоритм сценариев создаст возможность сделать обратный отчет для любых устройств.

renat2985, 2017-10-16:
@tretyakovsa А как ты это предлагаешь сделать? Выставляем время включения, а каким образом и где указываем продолжительность работы? 
[image]

tretyakovsa, 2017-10-16:
Таймеры будут на отдельной страницы как и сейчас. Конечно будут отличия в интерфейсе от текущего варианта. Но вот использовать мы будем один алгоритм разбора сценариев и таймеров и там же будет функционал взаимодействия модулей. Он будет скрыт от пользователя, но все эти функции будут в  одном блоке.

renat2985, 2017-10-16:
@tretyakovsa может все таки все стоит разместить на 1 странице? Мне сложно с списком сценариев работать, т.к. он не в json формате, а тут еще и разить его нужно убит на то что показывать а что не показывать. Это проблема.

i81, 2017-10-17:
Так вроде же обсуждали ранее... Ренат, зачем Вы хотите делать время включения и таймер обратного отчёта в одном смысловом поле? "Выставляем время включения, а каким образом и где указываем продолжительность работы?" - это не верный вопрос!
Таймер обратного отсчёта - это не продолжительность работы!!! Таймер обратного отсчёта - это время ЧЕРЕЗ КОТОРОЕ заданное релле перейдет в заданное состояние (переключится, включится или включится)

Paly4, 2017-10-29:
а на какой Sonoff нужна оригинальная прошивка ?

alexandersciv, 2017-10-30:
Есть решение вопроса как слить стоковую прошивку. Есть база стоковых прошивок. Есть возможность залить стоковую прошивку.