tretyakovsa/Sonoff_WiFi_switch #29: Отключение по таймеру) Таймер в сценариях(?)

dilpish, 2017-08-23:
т.е. есть необходимость чтобы устройство проработало... например 15 секунд, и перешло в предыдущее состояние)))

т.е. 

1) Включиться по команде (в идеале /relayon15s например)
2) Проработать 15 секунд
3) Выключится (с запросом или отправкой состояния... но это хотелки)

renat2985, 2017-08-24:
Уже начали делать эту функцию, она уже есть в таймерах, но пока не работает:
[image]

Запустить ее так же можно будит get запросом: http://ваш_ип/timerSave?module=rgb&trigger=not&id=644&day=All&time=07:00:00&work=00:40:00

id в этой строчке обязательно должен быть всегда разным.

dilpish, 2017-08-26:
Это здорово, это очень нужные функции. 

А можно-ли сделать "режим кнопки" - т.е. по кнопке, или по дополнительному датчику в виде кнопки что-то работало только пока "нажато" (хотелка: пока нажато+таймер)

Я понимаю, что всё это можно реализовать из вне, но тут такое применение, что он периодически будет офлайн сам по себе.

i81, 2017-09-14:
Так работает в итоге сейчас таймер обратного отсчёта или нет? Если "да" то как его правильно настроить? Добавил в таймерах строку - не выключается реле.
[image]
[image]

renat2985, 2017-09-14:
Функция "Продолж. работы" еще не работает.
И вы не правильно ее используете. Это не таймер обратного отсчета, это таймер продолжительности работы. То есть в первой ячейки вы указывается во сколько реле включится. А во второй, сколько но должно проработать.

i81, 2017-09-14:
Т.е., в итоге: простого таймера обратного отсчета нет пока вообще никак; функция "продолжительности работы" в интерфейсе реализована, но не работает. Так?

renat2985, 2017-09-14:
@i81 верно.

i81, 2017-09-14:
А какие-то сроки по реализации обычного таймера имеются?

renat2985, 2017-09-14:
Ничего не могу сказать. Пока мы заняты другими вещами. Что мы сделали за это время вы можете посмотреть в видео роликах под названиями "Что нового.", https://www.youtube.com/playlist?list=PL6NJTNxbvy-IPTDQk8XjTV41oRrFafrRi