tretyakovsa/Sonoff_WiFi_switch #30: Какие датчики подходят, схема их подключения

Neko1800, 2017-08-31:
Здраствуйте подскажите какие датчики (температура, освещенность, движения) подходят для этого проекта?
Подходят ли эти датчики (возможно есть и другие варианты)?
Датчик движения HC-SR505
Датчик освещенность GL5528
Датчик температуры 18B20
Как правильно их подключать к ESP8266 (например ESP-07)?

renat2985, 2017-08-31:
Вы посмотрели это видео? https://www.youtube.com/watch?v=TNiup8WdwBk
HC-SR505 - да
GL5528 + 1kOm - Да.
18B20 - Да, если нет, то скоро будит.

Подключение HC-SR505, VCC на VCC, GND на GND, OUT на нужный вам пин, смотрите видео.
GL5528 - 1 ножку подключаем к ADC пину, оттуда же идет резистор на 1кОм к GND, вторую ножку подключаем к VCC.

18B20 схема ниже, подключаем на нужный вам пин, смотрите видео.
[image]

Neko1800, 2017-08-31:
Спасибо. Видео я смотрел, как менять пины я понял.

renat2985, 2017-08-31:
Так же можно управлять relay через uln2003a
[image]

Neko1800, 2017-08-31:
Если нужно подключить одно реле то можно обойтись и транзистором вместо uln2003a или нет?

Neko1800, 2017-08-31:
Можно управлять напрямую вот таким готовым модулем реле? [https://ru.aliexpress.com/item/One-1-Channel-5V-Relay-Module-Board-Shield-with-Optocoupler-Support-High-and-Low-Level-Trigger/32719742919.html](url)
https://ae01.alicdn.com/kf/HTB1tkfDSFXXXXbmXpXXq6xXFXXXn/1-H-L-5.jpg

renat2985, 2017-09-01:
Можно обойтись и транзистором.
По поводу модуля, нужно смотреть, не знаю.

tretyakovsa, 2017-09-01:
Вообще датчик 18b20 сейчас установлен жёстко на пине 14. Это прописано в скетче. Задать его нельзя без перекомпиляции. Это связно с библиотекой <OneWire.h>, она написана, так что нельзя переключить её на лету.

Paly4, 2017-10-18:
а 2 и более 18в20  можно подключить паралельно на 14 пин ?
в другой прошивке как то реализовано без перекомпиляции , в вэбе выставляешь пин для этого датчика , сохраняешь , потом идешь в другой пункт меню > 1-wire< жмем сканировать и прописывается какой то адрес и датчик работает
[image]