tretyakovsa/Sonoff_WiFi_switch #32: Заметки чайника. Что, как, где получается и не очень.

AVL4GITHUB, 2017-09-22:
Всем привет!
Начну общую рубрику что на мой взгяд (чайника) работает или не работает.
Или, возможно и работает, но чайнику трудно ЭТО понять.
Вот на картинке, при попытке обновления, могла б быть информация какя сейчас у меня версия прошивки и на какую апгрейдиться будем.
Также часть кнопок с круглями углами, а часть с прямым углом. Режет глаз.
http://picsee.net/upload/2017-09-22/9e29b07ba31e.png

AVL4GITHUB, 2017-09-22:
На главной странице жму ДОБАВИТЬ СЦЕНАРИЙ и система висит на вращающемся круге.
http://192.168.0.204/scenery.htm
К жизни возвращаю переходом на предыдущую страницу в браузере.

renat2985, 2017-09-22:
По поводу обновления, хорошая идея. Сделаем. Чего еще не хватает?
Сценарии запустим в ближайшие недели. Как я вам и говорил ранее они пока не работают.

AVL4GITHUB, 2017-09-23:
Спасибо.
Описание видео с Ютуба очень не нравится. Что нового и что нового. А о чём старое?
Я ищу в видео информацию по МКТТ. Когда-то краем глаза это видел, а в каком видео?
Пересматривать все и дельть заметки в каком видео о чём говорится и "что нового"...
Скажите, пожалуйста, если кто знает, в каком видео о МКТТ.
Спасибо!

AVL4GITHUB, 2017-09-23:
И, подскажите, пожалуйста, как в пост вставить лисунок, а не линк на рисунок. Спасибо.
И по МКТТ.
Я настраиваю  его на сервисе www.cloudmqtt.com 
Между Андроид приложением и этим сервисом успешно идёт обмен информацией. Я и там и там вижу и от-туда и от-сюда могу управлять.
А как к этому делу прицепить НАШ проэкт?
А точнее, как ТОПИК прописать в нашем проэкте. Ибо данные сервиса, логин/пароль я ввёл в настройках, а далше - тупикнулся.
Спасибо!

tretyakovsa, 2017-09-24:
На телефон поставьте iot manager. В модуль забейте данные сервиса особенно внимательно с портом ставьте обычный порт. Как только все это будет сделано вы на телефоне увидите переключатель реле. В сервисе можно будет видеть все топики.

AVL4GITHUB, 2017-09-24:
Любий iot manager?
Ибо, я поставил MQTT DASHBOARD. И в телефоне и на веб я все чудесно взаимодействую, но даних с нашего устройства - нет.

AVL4GITHUB, 2017-09-24:
Спасибі. Поставил именно синий, а не красній IOT MANAGER с Плеймаркета и МКТТ  заработал. Пожалуйста, укажите , сто именно с єтой программой работаем, или необходиміе топики для других приложений. Тестирую дальше. Спасибо.

renat2985, 2017-09-24:
@AVL4GITHUB хорошо, постараюсь называть более информативно видео ролики.
В MQTT работает на данный момент только реле.

AVL4GITHUB, 2017-09-25:
Спасиба за ответы.
Тестирую дальше.
Есть проблеммы: IOT MANAGER работает только по ВайФай. В 3Ж сети не может связаться даже с брокером.
Попробую на других Андроид приложениях и отпишусь, ибо год назад ДРУГИЕ приложения работали нормально. Это раз.
Два: тестировал на 5 и 12 выходах управление гаражными воротами.
Полный завал.
Если не перегружать девайс, то всё ок. Но если ресетнуть девайс с вебки или по питанию, ворота самопроизвольно открываются или закрываются в момент перегрузки девайса.
Но это даже прикольно, ибо, если я потерял ключи от дома, пульт от ворот и телефон, то я подхожу к силовому трансформатору моего района, обесточиваю район рубильником, опять подаю питание, то мои ворота (а может и не только мои) откроются и я смогу попасть домой. Технологии, однако :)
Если нельзя избавитьсяот этого глюка (я когда-то, где-то, читал, что, вроде на 14 или 16 пине нет такой беды, то можно-ли сделать, что б по команде на пине появлялся меандр. Тогда я отвяжусь от постоянного тока и смогу через последовательный конденсатор подать команду на управление чем-либо. То-есть у нас будет такая логика: есть меандр - логическая 1. Нет меандра - логический 0.

AVL4GITHUB, 2017-09-25:
Ещё глюк, на мой взгляд.
Ночьо отключали питание 220 вольт, на районе.
Сутра девайс - точка доступа.
Почему после запитки всего дома, наш девайс к роутеру не подцепился автоматически?
Кроме того, считаю нужнім  на один из девайсов в нашей внутренней сети поцепить часі реального времени, и что б после обесточивания всего дома и оитсутствия интернета, наша система знала время.

renat2985, 2017-09-25:
https://youtu.be/7z_xcVtL0aE?list=PL6NJTNxbvy-IPTDQk8XjTV41oRrFafrRi&t=25
Здесь ответы на ваши вопросы.

AVL4GITHUB, 2017-09-25:
У меня в морде нет выпадающего меню параметров подключения к роутеру:
http://picsee.net/upload/2017-09-25/804cf951805e.png

AVL4GITHUB, 2017-09-25:
Спасибо, у меня не догружалась файловая система.
Принудительной перепрошивкой теперь стала нормально файловая. Появились все опции в морде. 
Но значения параметров перезагрузки не сохраняются в памяти.
Я ставли или 30 секунд или минуту, а также указываю  мигальный пин 13.
Несколько раз жму СОХРАНИТЬ, но при обновлении страницы получаю старое значение (10сек) и светодиод не мигает при перезагрузке.

Тем временем поставил другое МКТТ приложение. Так вот, это другое приложение без проблемм работает в сети 3G. IotManager ж работает только в сети Вай-Фай. Я понимаю, что это недоработка не наших ребят, но акцентирую на этом внимание, что этот ИотМанагер нам пока-что не подходит.

renat2985, 2017-09-25:
Ясно, спасибо за информацию.

AVL4GITHUB, 2017-09-25:
Пожалуйста.
Но у меня в морде не сохраняются настройки переподключения к точке доступа. Я ещё раз перепрошил модуль, не помогло.
И, пожалуйста, скажите пару слов, как ПРАВИЛЬНО настроить статический ИП адрес. Видео мне не помогло. Какие цифры вписывать ? Ибо, при некоторых цифрах устройство замедляется в 200 раз, примерно. Станицца грузится минуты две.
Спасибо.

renat2985, 2017-09-25:
@AVL4GITHUB, Давайте пока подождем Сергея, он со дня на день должен приехать. :) У него были мысли сделать координальные изменения. Потом придется все заново тестировать. :)

alxnovo, 2017-09-28:
@AVL4GITHUB,  у меня долго грузится страницы ESP'шки когда запущен торрент на ноуте через WiFi. Если что поставил на закачку...всё до ESP не пробиться.
Разницы в скорости между статическим адресом и DHCP у меня нет.
В статическом адресе обычно используются первые 3 цифры от адреса роутера и четвёртая цифра лучше от 130 (мой роутер раздаёт сам от 100, поэтому и рекомендую от 130)
Например, как обычно у всех, адрес роутера **192.168.0**.1 значит статический адрес делай **192.168.0**.130

r-o-p-i, 2018-03-24:
Не знаю где написать  напишу тут, использую ваши прошивки , очень понравился проект уже от донатил Сергею на Яндекс . Вопросс заключается в том что спустя 3 дня или 2 точно не измерял, невозможно попасть на страницу управления [image] , а модуль при этом работает, реле переключается.После перезагрузки все становиться в норму. В чем может быть проблема? и соответственно в MQTT перестают передаваться данные.

renat2985, 2018-03-24:
Спасибо за donate. Странно. Не замечал такого. Попробуйте открыть вот эту страничку http://192.168.1.48/page.htm?starting и сделать скрин.

r-o-p-i, 2018-03-24:
192.168.1.48 реле 
2018/03/18 21:34:21 | External System | manually
-- | -- | --
2018/03/18 21:50:10 | External System | manually
2018/03/19 18:41:41 | External System | manually
2018/03/19 18:43:10 | SoftwareSystem restart | manually
2018/03/21 04:23:10 | Power on | manually
2018/03/21 19:07:35 | External System | manually
2018/03/21 19:08:23 | SoftwareSystem restart | manually
2018/03/22 04:23:54 | External System | manually
2018/03/22 19:54:56 | External System | manually
2018/03/24 12:57:45 | External System | manually

192.168.1.38 термодатчик dht22

2018/03/20 19:57:42 | SoftwareSystem restart | manually
-- | -- | --
2018/03/20 19:58:34 | SoftwareSystem restart | manually
2018/03/21 19:05:32 | Hardware Watchdog | manually
2018/03/21 19:05:40 | External System | manually
2018/03/23 20:21:45 | Exception | manually
2018/03/24 13:43:15 | External System | manually
2018/03/24 17:24:47 | Hardware Watchdog | manually

renat2985, 2018-03-24:
Пишите в skype: renat2985