tretyakovsa/Sonoff_WiFi_switch #34: При компиляции текущих исходников не работает главная страница

alxnovo, 2017-09-26:
Если скомпилировать текущие исходники, то получаем вот такую главную страницу если переключиться на Jalousie 

[image]

Через бинарники всё нормально работает, а вот через исходники нет

renat2985, 2017-09-26:
Не стоит самому компилировать. Используйте готовые bin файлы. :) Просто на данный момент в исходниках еще не готовый код, мы внедряем новые функции. Как только мы их доделаем, протестируем, мы сразу выкладываем bin файлы в папку build. После чего у вас в ESP появляется красный треугольник. Так же все доступные сборки вы можете посмотреть в ESP на странице "Конфигурация".

alxnovo, 2017-09-27:
Ну, мне всё-таки нужно немного подредактировать код (в жалюзях) для обесточивания всех реле через тайминг пока вы не реализовали в основном коде. 
Ладно пока буду использовать нестабильные исходники :)

mikemdr, 2017-12-19:
>Не стоит самому компилировать. Используйте готовые bin файлы. :)
>"Компилируя исходники самостоятельно, вы как правило будите получать нестабильную сборку."
Следует ли это понимать как намёк на то, что проект является "как-бы закрытым" и исходники не соответствуют бинарникам, в коих в последствии (или уже?) могут появится "недокументированные фичи"?

renat2985, 2017-12-19:
@mikemdr нет. Просто вся разработка ведется в master ветке. То есть как только мы более менее протестировали сборку мы выкладываем bin файлы. Вот в это время вы можете забирать исходники и компилировать самостоятельно. Только зачем? Далее мы снова начинаем добавлять новые функции изменяя код, после этого исходники могут быть не работоспособны, т.к. не всегда получается сразу сделать задуманное. Иногда исходники неделями лежать неработоспособными пока мы не придумаем как решить ту или иную задачу. Поэтому исходники качать не рекомендуется, у вас так же могут быть установлены не те библиотеки которые используем мы что тоже может повлиять на работу ESP.

alexandersciv, 2017-12-20:
**@alxnovo**  Я столкнулся с подобным интерфейсом на смартфоне. В браузере штатном   подобная картина как у Вас, в Опера мини всё нормально. При этом в браузерах одинаковые настройки.

mikemdr, 2017-12-20:
Под Win7 в IE, FireFox46 и ЯндексюБраузере - как у ТС.