tretyakovsa/Sonoff_WiFi_switch #35: DHT 22

alexeyko424, 2017-09-30:
Доброго дня. Можете поделиться рабочей библиотекой для DHT22? Выдает ошибку в показаниях как значение -3276.70 С. При подключения другой - просто ошибка при компиляции.

alexeyko424, 2017-09-30:
Почему то, библиотека DHT, не заработала. Подкинул SimpleDHT.h - работает, но вес большой получился

renat2985, 2017-09-30:
Ниже к этому сообщению приложил архив библиотеки которую используем мы.
[arduino-DHT.zip](https://github.com/tretyakovsa/Sonoff_WiFi_switch/files/1346463/arduino-DHT.zip)

alexeyko424, 2017-09-30:
Благодарю Вас.

nagornyak, 2017-11-20:
Подскажите пожалуйста, данные с DHT по MQTT только у меня не передаются?

alexeyko424, 2017-11-21:
Доброго утра.
MQTT я использую в своем написании (там много того, что вам не надо).
В Вашем проекте я MQTT не использовал, но желание было. Сейчас я на работе
(работаю вахтовым методом), проверить не могу. Брокер стоит на linux, плата
*Orange* *Pi* PС.