tretyakovsa/Sonoff_WiFi_switch #36: Ошибка при прошивке новых версий SONOFF.

johnes33, 2017-09-30:
Мучался с такой проблемкой:
Прошиваю модуль sonoff basic и после прошивки модуль не работает. В консоль пишет всякий мусор. Причем от прошивки не зависит - не работает с любой прошивкой. Но если консоль настроить на скорость 74880 то там есть сообщение, о том что возникает ошибка проверки контрольной суммы (chksum). В итоге набрел на решение данной проблемы [тут](https://letscontrolit.com/forum/viewtopic.php?t=3071): . Если в краце то для новых версий нужно выставлять режим работы с флеш памятью в DOUT ...... убил на это два вечера ((

P.S. может включите данную инфу в readme файл проекта, а то решение банальное, но совсем не очевидное.

renat2985, 2017-09-30:
Спасибо за информацию.

Paly4, 2017-10-18:
тоже был столкнулся с таким , весь мозг себе чуть не вынес за 2 дня пока не нашел решение https://esp8266.ru/forum/threads/proshivka-sonoff.1891/page-17#post-41064
и через это(DOUT) не работает обновление через треугольник в вэбе

renat2985, 2017-10-18:
@Paly4, насколько я понимаю Sonoff сейчас перешли на ESP8285, именно для этих версий необходим DOUT. Стоит ли нам переходить на DOUT? Он поддерживается ESP8266, но он немного меленее чем DIO.
Но эту проблему действительно нужно решать, Sonoff с ESP8285 становится все больше.

https://github.com/arendst/Sonoff-Tasmota/wiki/Theo's-Tasmota-Tips
https://github.com/letscontrolit/ESPEasy/issues/474

Paly4, 2017-10-19:
renat2985 , глянул , установлена все таки 8266 , но заводская прошивка установлена с DOUT , 
[image]
прошил , обновился через ссылку , вроде бы прошло , открывается главная страница , горит треугольник обновления , в настройки , управление с главной не могу войти , крутится все время кружок 
[image]

sultan5448, 2017-10-19:
извиняюсь за оффтоп. на видео что на одной странице несколько устройств и сценарии между ними. как добавить несколько устройств на одной странице ведь у них разные адреса?

renat2985, 2017-10-19:
@sultan5448 они сами друг друга находят и появятся у вас на 1 странице. Пожалуйста, в будущем создавайте новые issues.

@Paly4, ясно, спасибо.

alexandersciv, 2017-10-29:
У меня несколько модулей Sonoff Basic и Sonoff Basic / RF с TH на плате (5 отверстий). Три модуля Sonoff Basic прошил [https://github.com/tretyakovsa/Sonoff_Loader](url) без проблем. В ожидании новой версии прошивки решил прошить Sonoff Basic / RF. Флешер модуль не увидел, хотя настройки и USB_TTL не менялись. Проверил контакты, проверил и поменял USB_TTL. Без результата. В терминале модуль сообщает
[image]
Пробовал все рекомендации по скорости и смене режимов. Модуль не виден. Подключил ещё один Sonoff Basic - прошился сразу. Подключил ещё один Sonoff Basic / RF, тоже не виден.
Установлена 8266 как у **@Paly4** 
Как решить проблему?