tretyakovsa/Sonoff_WiFi_switch #37: Управление газовым котлом отопления. Нужен такой девайс.

AVL4GITHUB, 2017-10-02:
Всем привет!
Нужен девйс для управления газовым колом отопления в квартире.
Исходные данные:
Есть фирменный котёл  отопления. В нём предусмотрен изготовителем внешний вход управления (два контакта). Если контакты замкнуты, то котёл греет, разомкнуты - не греет.
Наша задача: сделать управление температурой в квартире.
Нужно иметь возможность задать ТРИ температуры:
1. Все спят - 18 градусов.
2. Бодрствуют - 20 градусов.
3. Никого нет дома - 12 градусов.
Также нужно контролировать не завис-ли котёл. Качество газа желает быть лучшим, и, иногда, котёл падает в аварию.
Если в данный момент котёл должен греть, а температура горячей трубы на выходе с котла не поднимается, то   перезапускаем котёл.
Итого, нам нужно два датчика температуры и два реле. Естественно, что по временному сценарию  это упрвляется и предусмотрены всякие интернет-плюшки.
Всё железо у меня готово и кое-как даже работает.
Делаем?

alexandersciv, 2017-10-02:
**AVL4GITHUB** тоже занимаюсь своим газовым котлом и тёплым полом. Какие есть наработки?

AVL4GITHUB, 2017-10-02:
Уже года с три, ближе к зиме, пробую довести до ума, то один, то другой проект слизанный с Интернета. В прошлом году запустил на основе http://homes-smart.ru/. Но слишком то решение не унифицировано под котёл и уже не помню во что упёрся, ибо весна пришла. Я решил попробовать что-то новое, вот ЭТОТ вариант пробую. А как у вас успехи?

tretyakovsa, 2017-10-02:
Я сейчас код переписываю и сценарии добавляю.
Все это можно будет реализовать.
С вас описание подключения всего этого. 
Что куда вы в котле подключаете?
Где расположены датчики и какие?
 Какой должен быть интерфейс страницы управления?

AVL4GITHUB, 2017-10-02:
Даю описание.
Сделано всё это на основе SONOFF, как у Вас.
Но контакты внутреннего реле нашего девайса отрезаны  от сети 220 вольт и коммутируют управление газовым котлом (ГПИО 12).
Нутрь нашего девайса впихнуто твердотельное ркеле, которе подаёт питание на котёл (розетка). Управляется это реле через транзистор (ГПИО5).
Зелёный светодиод (ГПИО13)
Кнопка (ГПИО0)
18В20, 2 шт (ГПИО14)
Котёл у нас двухфункциональный. Он независимо греет воду для рукомойников и для отопления.
Летом греет только для рукомойников. Зимой и туду и туда. Рукомойники нас никак не интересуют.
Котёл, значит включён в розетку (ГПИО5). Это включение нам нужно, что б ресетить котёл, если он зависнет от некачественного газа. Тогда на 5 секунд выключаем котёл и опять включаем.
Предусмотренный с завода вход управления котлом подключён к контактам внутреннего реле нашего девайса (ГПИО12). Мы смотрим какое время и день недели, меряем температуру и включаем или выключаем котёл.
Датчиков температуры - минимум два, максимум три.
1-й меряет температуру в помещении.
2- меряет температуру горячей трубы на выходе с котла. Ибо мы сейчас греем помещение, а температура трубы на выходе не поднялась выше температуры помещения - значит котёл завис и мы его ресетим.
3 - температура на улице. Не оьъязательно, но приятно её знать, коль  мы делаем высокотехнологический девайс.
Интерфейс: любой. Заходим с браузера. Локальное и через интернет мониторинг-управление.
Это базовое задание.
Дальше - можно в нескольких комнатах врезать в трубу отопления електроклапан с ВайФай и подрегулировать температуру конкретно в этих комнатах. Но это всё ПОТОМ, если БАЗАОВЫЙ вариант будет надёжно работать.
Потом можно и датчик движения прилепить, как охранная и сервисная функция.
Если я, к примеру приехал с работы раньше, а дистанцтоннно по МКТТ не включил подогрев, то, как только я зашол в помещение - автоматически подогрев включился.

alexandersciv, 2017-10-02:
Re: [tretyakovsa/Sonoff_WiFi_switch] Управление газовым котлом отопления. Нужен такой девайс. (#37)

Здравствуйте!
В целом Ваш сценарий работы котла понятен.
Я получил письмо и от tretyakovsa, содержание письма ниже.
Готовлю описание своего сценария и по готовности отправлю.
Ваш сценарий могу перенаправить tretyakovsa.
Сообщите адрес Вашей почты. Мой alexandersciv@gmail.com
 от - tretyakovsa
==Я сейчас код переписываю и сценарии добавляю.
Все это можно будет реализовать.
С вас описание подключения всего этого.
Что куда вы в котле подключаете?
Где расположены датчики и какие?
Какой должен быть интерфейс страницы управления?
===
Вы писали 2 октября 2017 г., 19:24:44:

Даю описание.
Сделано всё это на основе SONOFF, как у Вас.
Но контакты внутреннего реле нашего девайса отрезаны от сети 220 вольт и коммутируют управление газовым котлом (ГПИО 12).
Нутрь нашего девайса впихнуто твердотельное ркеле, которе подаёт питание на котёл (розетка). Управляется это реле через транзистор (ГПИО5).
Зелёный светодиод (ГПИО13)
Кнопка (ГПИО0)
18В20, 2 шт (ГПИО14)
Котёл у нас двухфункциональный. Он независимо греет воду для рукомойников и для отопления.
Летом греет только для рукомойников. Зимой и туду и туда. Рукомойники нас никак не интересуют.
Котёл, значит включён в розетку (ГПИО5). Это включение нам нужно, что б ресетить котёл, если он зависнет от некачественного газа. Тогда на 5 секунд выключаем котёл и опять включаем.
Предусмотренный с завода вход управления котлом подключён к контактам внутреннего реле нашего девайса (ГПИО12). Мы смотрим какое время и день недели, меряем температуру и включаем или выключаем котёл.
Датчиков температуры - минимум два, максимум три.
1-й меряет температуру в помещении.
2- меряет температуру горячей трубы на выходе с котла. Ибо мы сейчас греем помещение, а температура трубы на выходе не поднялась выше температуры помещения - значит котёл завис и мы его ресетим.
3 - температура на улице. Не оьъязательно, но приятно её знать, коль мы делаем высокотехнологический девайс.
Интерфейс: любой. Заходим с браузера. Локальное и через интернет мониторинг-управление.
Это базовое задание.
Дальше - можно в нескольких комнатах врезать в трубу отопления електроклапан с ВайФай и подрегулировать температуру конкретно в этих комнатах. Но это всё ПОТОМ, если БАЗАОВЫЙ вариант будет надёжно работать.
Потом можно и датчик движения прилепить, как охранная и сервисная функция.
Если я, к примеру приехал с работы раньше, а дистанцтоннно по МКТТ не включил подогрев, то, как только я зашол в помещение - автоматически подогрев включился.
—

С уважением,
Иванов Александр.

alexandersciv, 2017-10-02:
Re: [tretyakovsa/Sonoff_WiFi_switch] Управление газовым котлом отопления. Нужен такой девайс. (#37)

Здравствуйте!
Готовлю описание своего сценария котельной и тёплого пола, по готовности отправлю.
Сообщите адрес Вашей почты. 
Мой alexandersciv@gmail.com
Вы писали 2 октября 2017 г., 18:55:55:

Я сейчас код переписываю и сценарии добавляю.
Все это можно будет реализовать.
С вас описание подключения всего этого.
Что куда вы в котле подключаете?
Где расположены датчики и какие?
Какой должен быть интерфейс страницы управления?
—

С уважением,
Иванов Александр.

alexandersciv, 2017-10-02:
Извиняюсь, что не знал о дублировании переписки по эл.почте в топике. 
Вот моё видение темы:
Принял решение в основном автоматизацию домом строить на модулях ESP8266 и как одно из законченных решений Sonoff от https://www.itead.cc/ru/smart-home/sonoff-wifi-wireless-switch.html (рассматриваю ESP32 OLED Wi-Fi Модули + Bluetooth
 https://ru.aliexpress.com/item/ESP32-OLED-WiFi-Modules-Bluetooth-Dual-ESP-32-ESP-32S-ESP8266-OLED-ESP32-OLED-for-Arduino/32813166670.html как продвинутый вариант, в том числе управления голосом по Bluetooth).
Систему взаимодействия устройств (разные самые бюджетные но продвинутые, типа Broadlink) строю на OpenHab + MQTT + MajorDoMo.
Частично реализованных идей много, в частности по котельной.
Имею двухконтурный газовый котёл Baxi Slim 1.49iN, установлен в цоколе (0 этаж). +Смонтированные теплые полы на 3 этажах дома. 
+На каждый этаж смонтирован насос. 
+На каждом этаже гребёнка с ручными регуляторами по комнатных секций.
Задача 1. Подключить управление котла по температуре в самом нужном помещении (спальня 2 этаж). При этом терморегулятор включает котёл и через пускатели все 3 поэтажных насоса.
Заказал =терморегулятор 868 мГц беспроводной котельной контроллер термостат 
= https://ru.aliexpress.com/store/product/868MHz-Wireless-Boiler-Room-Controller-Heating-Thermostat-Weekly-Programmable-With-Backlight-App-Remotely-Control-With-Gateway/1986435_32590299248.html?spm=a2g0v.12010612.0.0.e6c13c5CidPHX = в комплекте со шлюзом = https://ru.aliexpress.com/store/product/868Mhz-gateway-for-wireless-programmable-wireless-digital-wifi-room-thermostat-smart-radiator-valve-gas-detector/1986435_32733898219.html?spm=a2g0v.12010612.0.0.7afff8e58cyYml =
(Хочу имея проверенное решение, разобрать его и разобраться как интегрировать в систему OpenHab + MQTT).
Задача 2. Поскольку тёплые полы очень инертны, не смог придумать простой сценарий поэтажного управления температурой кроме как, на поэтажной гребёнке в ручную выставляется приоритет температуры по помещениям. В самом удалённом помещении на этаже ( 0 цокольный+ 1 этаж) устанавливается Sonoff с датчиком температуры (при этом он может выполнять как реле дополнительный сценарий) передающий на Sonoff подключённый к пускателю насоса 0 или 1 этажа, показания температуры, и включать или выключать насос, увеличивая прокачку теплоносителя.
Ещё много сценариев практических действий планирую возложить на Sonoff в котельной.

grossbel, 2017-10-02:
Вообще то в нормальном котле все это практически есть и недельный таймер почасовой, где можно его запрограммировать на снижение температуры и авто сброс в случае ошибки.
В любом газовом котле предусмотрено подключение внешнего термостата. Возможно кому нибудь еще это и пригодиться, но это пару человек и нельзя единичные хотелки воплощать в целый алгоритм. Необходимы простые сценарии из которых можно составить сложный алгоритм.
Например как здесь http://ab-log.ru/forum/viewtopic.php?f=1&t=1130
очень легко написать сложный алгоритм допустим для одного термодатчика и нескольких исполнителей. 
Разработчикам этого чудного устройства: у меня есть достаточно много опыта испытаний устройств на ESP да и не только для умного дома, всегда подхожу ко всем устройствам как пользователь который ни чего не понимает и может нажать не ту кнопку. У Вас направление движения правильное, но немного запутанное, если хотите, то тогда прислушивайтесь к моей критике (прошу, только не обижайтесь, это будет только на пользу проекту) и у Вас получиться одно универсальное устройство, с одной прошивкой на разные случаи домашней автоматизации.
Желание топикстартера не такое же и сложное
1. недельный термостат - у вас уже есть таймеры на одни сутки, осталось сделать раскладку на семь дней
2. режимы (статус) дома их не так уж много и надо, пригодятся и для охранной сигнализации и для энергосбережения и для котла в том числе. Что это значит: необходимо создать какие либо виртуальные порты в которых значение можно изменить (физической кнопкой либо виртуальной, либо кодом пульта) при этом устройство раздаст этот режим всем остальным устройствам и соответственно жалюзи закрылись, ворота закрылись, лишний свет погас, утюг выключился потому что розетка обесточилась, котел перешел в эконом режим
3. сравнение температур, пока что под вопросом, тут пока еще не хватает простых функций, типа какая кнопка нажалась и что включить, да и с интерфейсом пока еще не все гладко, до сих пор нет статуса выхода и не понятно в каком состоянии у тебя выход пока не перегрузишь страницу, это не удобно.

grossbel, 2017-10-02:
alexandersciv не знаю какой алгоритм вашего котла, но в моем если снять управляющую перемычку, то он не только перестает нагревать теплоноситель, но и отключает насос циркуляционный, обратите на это внимание.

grossbel, 2017-10-02:
Поддерживаю alexandersciv, функция передачи команды от устройства к устройству должна быть, либо это будет посредством MQTT, либо общение модулей прямиком между собой по их какому либо ID (например это будут названия в шапке) или IP жестко привязанные как в MEGAD.
Например:
должен быть конфиг устройства, где можно четко выбрать GPIO и обозначить кто он вход или выход. 
Если он вход, то ты должен выбрать какой он  (кнопка, переключатель или датчик) в зависимости выбора сразу должна видоизменяться страница с нужными полями, далее вписываем необходимую температуру и действие которое необходимо произвести например локально на этом же устройстве, либо на удаленном - ID такой-то, GPIO такой-то, 1(включить) 0(выключить)

grossbel, 2017-10-02:
Продолжу с вашего позволения, при функции термостата должна быть возможность прописания нескольких сценариев, то есть должно быть несколько полей чтоб можно было на разные температуры одного датчика вписать разные действия например:
темп 25 GPIO x1 -1(включить), GPIO х2 -1(включить)
темп 45 GPIO x3 -1(включить), GPIO х2 -0(выключить)
темп 60 GPIO x4 -1(включить)
темп 90 GPIO x5 -1(включить)
при этом в модуле таймеров должен работать GPIO x6 который включается некоторое количество в час.
Это живой пример алгоритма реализации в котельной частного дома с газовым котлом и камином с водяной рубашкой.

grossbel, 2017-10-02:
На первый взгляд, может это и сложно выглядит, но все это разрозненно реализовано на других устройствах на базе ESP и прочих контролеров, но у всех существует идеология единого сервера, а это противоречит направлению интернет вещей, Ваш проект отличается структурой построения и это большой шаг, ведь даже если передавать команды между устройствами посредством MQTT, то брокера можно попробывать поднять и на каком нибудь роутере, ну в крайнем случае расбери повесить в сеть.

alexandersciv, 2017-10-03:
**grossbel** : Как мне пояснили специалисты, управление замыканием и размыканием перемычки именно включается или отключается как газовая горелка, так и насос.
Поэтому в **задаче 2** для себя я выделил алгоритм который позволит  на большой площади более тонко регулировать состояние температуры по этажам за счет того, что общий насос и горелка выключены, а включаются поэтажные насосы повышая температуру на этаже за счет увеличения протока теплоносителя.

alexandersciv, 2017-10-03:
**grossbel** изложил очень правильный алгоритм взаимодействия устройств на этой прошивке. Хочу добавить. Анализируя работу терморегулятора, получается, что в общем это алгоритм заложенный в моей **задаче2**. А именно. Одно устройство  Sonoff устанавливается удалённо от котла с датчиком температуры ( возможно с дисплеем как это  https://ru.aliexpress.com/item/new-NODEMCU-wifi-NodeMCU-forArduino-ESP8266-wemos-for-OLED/32802182457.html ) и передаёт данные на   Sonoff, который установлен непосредственно возле котла и через реле управляет его включением. 
Интересное предложение **AVL4GITHUB**   
== 2- меряет температуру горячей трубы на выходе с котла. Ибо мы сейчас греем помещение, а температура трубы на выходе не поднялась выше температуры помещения - значит котёл завис и мы его ресетим.==
Как привязать данные уличной температуры, пока не решил, но видимо отдельным устройством буду запускать метеостанцию.
Прошивка уникальная, но у меня в дополнительной информация теряется сама кнопка. Может выделить её цветом и размером шрифта? Очень нравится идея высказанная в теме доработки интерфейса: на экране названия устройства и кнопки, остальное в выпадающем меню (окне).

AVL4GITHUB, 2017-10-03:
Не слишком забегайте вперёд.
Давайте запустим самый простой вариант, обкатаем, а потом уже можно будет обволакивать его плюшками. Ибо, как показывает практика, даже простые вещи могут глючить :)

AVL4GITHUB, 2017-10-03:
Вот только-что глюк пойцмал по базовому варианту. 
У меня к девайсу лампочка подключена.
Несколько раз в день, я просто так включаю\выключаю лампочку по МКТТ.
Так вот, только что пробую поиграться, а лампочка не реагирует. Попробовал кнопкой на девайсе - переключает.
Смотрю на на МКТТ брокере - данные идут, а переключения - нет.
Передёрнул питание и заработало.

grossbel, 2017-10-03:
Фундамент не сделаешь, пока не запроектируешь крышу, так и здесь, направление движения вроде есть, но необходимо знать что взять с собой в дорогу

alexandersciv, 2017-10-03:
Пока ожидаем, что скажет уважаемый **tretyakovsa** на наши предложения, может есть у кого наработки управления кранами на базе Sonoff или ESP8266?

grossbel, 2017-10-03:
Смотря для какой цели кран, я использую клапана herz с термоэлектрической головой на 220 вольт https://www.herz.by/catalog/termostaticheskiy-klapan-gerts-ts-90-fv/herz-ts-fv-prokhodnoy/
, управляю простым китайским термостатом за 7.5 usd, если скорость открытия или закрытия не критична, то все это работает безшумно. 3-5 минут его скорость открытия, клапана есть с разным kvs (пропускная способность) можно в связке использовать непосредственно на батарее, сам привод можно воткнуть на гребенку теплого пола (если она позволит), я использую просто отдельно для подключения теплообменника в теплоаккамуляторе при избытке тепла от твердотопливного.

alexeyko424, 2017-10-04:
Доброго дня. Занимался котлами и колонками, только не на esp. Одна из глобальных проблем - как это все зарегистрировать. Пожарные и газовщики "дадут по голове", это в лучшем случае. Если от вашей "поделки" пострадает человек, это уже статья. И не будут смотреть, что дом Ваш и т. д.
Всё, что мы сами делаем, это индикаторы, а не измерители.

alexandersciv, 2017-10-04:
**grossbel**:  В отношении кранов. Сразу оговорюсь, не буду больше писать, что для котельной частного дома, поскольку все решения подойдут и для квартиры. А краны нужны для холодной и горячей воды, возможно газа. С радиоуправлением нашёл только для газа (https://ru.aliexpress.com/store/product/868Mhz-carbon-monoxide-co-de-LNG-Natural-Gas-leak-Smart-Detector-Monitor-Alarm-sensor-With-Voice/1986435_32697769366.html?spm=a2g0v.12010612.0.0.6239803fPlV2Lo ). Хотя сценарий управления можно самый простой. Датчик протечки + Sonoff + кран + сценарий в прошивке: при повышенной влажности кран закрылся, на смартфон уведомление. Можешь сам принять решение о включении-выключении.

alexandersciv, 2017-10-04:
**alexeyko424:** Безопасность и надёжность должна быть превыше всего. При этом каждый в меру понимания этого вопроса может поступать как посчитает нужным. Можно долго дискутировать что такое индикаторы и измерители, смысл один: получить достоверную информацию о выполнении сценария прописанного в приборе, проанализировать её и вмешаться с ручным управлением при проблемах. В отношении сдачи объекта соответствующим службам, имею негативный опыт включения сертифицированного оборудования по причине материальной заинтересованности инспектора.

grossbel, 2017-10-04:
alexandersciv, если для протечки воды или газа, то есть готовые решения, отлаженные на заводе и с промышленными температурными диапазонами. Например в нашей стране все новые котельные оснащаются сигнализаторами метана и угарного газа, опционально сигнализатор позволяет управлять запорным клапаном, а так же выдавать сигнал на пульт диспетчера, стоимость его порядка 60 USD и газоанализирующие датчики там ну ни как не ардуино и даже не Китай. По системам протечки тоже есть готовые решения.

grossbel, 2017-10-04:
http://minskgas.by/Catalog/products/gassupply/%D0%A1%D0%B8%D0%B3%D0%BD%D0%B0%D0%BB%D0%B8%D0%B7%D0%B0%D1%82%D0%BE%D1%80%D1%8B--%D0%B3%D0%B0%D0%B7%D0%B0/%D0%A1%D0%B8%D0%B3%D0%BD%D0%B0%D0%BB%D0%B8%D0%B7%D0%B0%D1%82%D0%BE%D1%80-%D0%B3%D0%B0%D0%B7%D0%B0-%D0%A1%D0%93-1%D0%A1%D0%9D.aspx не реклама, там же и клапана

alexandersciv, 2017-10-04:
grossbel  Большинство готовых решения мне известны. Они в большинстве своём не радио управляемые и не работают с одной информационно-исполнительной базой. Даже такое решение как Broadlink не имеет кранов и не имеет обратной связи. В нашем случае, я вижу потенциал интеграции  устройств в такую систему на базе Sonoff под управлением прошивки из нашего топика. Остаётся только Ренату и Сергею вынести вердикт возможности прописать в ихней прошивке выше указанные сценарии.

alexeyko424, 2017-10-04:
Доброго дня. Ребята. по роду своей работы, на данный момент, [используем клапаны ](http://teplosmile.ru/category/krany_s_elektroprivodom/) с обратной связью и без. Как предложение: может использовать контроллер типа ОВЕН и с него, в проект по modbus кидать данные. Надежно и в проект можно подцепиться без проблем. Извините если, что не так.

AVL4GITHUB, 2017-10-04:
На мой взгляд, для СОНОФФ удобнее всего, использовать 220-ти вольтовые клапана.
Я прав?

alexeyko424, 2017-10-04:
SONOFF - без обратной связи по клапану. Связь только на выполнение команды. Если Вы не желаете иметь состояние клапана, что для дома это актуально, то да, 200В -  это оптимальный вариант.

renat2985, 2017-10-04:
Может есть какие-то моторы, которые ставятся прямо на стандартные краны? Что бы не вызывать сантехника. Можете что-то предложить?

alexeyko424, 2017-10-05:
Утро доброе. На установленную запорную арматуру, которая не имеет редуктора, ставить (из опыта и форумов) не советую. Тут 50/50. На новых (шаровых) кранах - работает, но надо учитывать нагрузку рычага. Если вода Г**, то высока возможность поломки. Лучше ставить готовое решение.
Да, на алике есть накладные редукторы, но 50/50. Много параметров надо учитывать.

alexandersciv, 2017-10-05:
Уважаемый **alexeyko424**  затронул  очень важную, даже принципиально важную тему =обратная связь устройств= которая и определят надёжность системы. Было сказано, что **=SONOFF - без обратной связи по клапану=** 
-Во первых, готовых заводских решений с кранами управляемыми по WiFi, тем более с обратной связью, мне найти не удалось. 
-Во вторых, SONOFF это доработанное устройство на модуле ESP8266 (думаю в перспективе ближайшего будущего  на ESP32) в котором заложена масса не использованных возможностей, от подключения датчика температуры до датчика движения. По этому получение подтверждения от выполнения посланной команды, зависит только от сценария и прошивки. К примеру: по температуре – команда –включить газовый котёл – датчик на трубе – нагрелась труба – система отработала штатно – не нагрелась – перезапуск котла. Так и по воде и.д.

alexeyko424, 2017-10-05:
alexandersciv - Вы правильно описали ПИД. Но, такое косвенное выполнение задания, сильно зависит от инерции/интеграла i. Описать такой сценарий думаю проблематично. И самое главное - это надежность исполнительных устройств. Работаю по АСУ ТП. Сейчас почти все исполнители без обратной связи - качество позволяет.
ESP32 - шикарная штука.
Разводить дискуссию не буду. Если, что - korzhavin@gmail.com.

alexandersciv, 2017-10-05:
По кранам. Опять таки уважаемый **alexeyko424** очень правильно  и полно изложил основные сложности по работе с электрокранами. На самом деле я тоже не нашёл законченного решения, но есть у нас Кулибины: https://www.youtube.com/watch?v=VHbmbaw-t_A   https://www.youtube.com/watch?v=7bwKAQn02PU  = защита от протечек воды в квартире. Антипотоп на Arduino своими руками=.
Сам механизм = https://ru.aliexpress.com/item/DC8V-DC16V-Intelligent-Gas-pipeline-robot-only-robot-with-automatic-manual-clutch-conversion-for-smart-home/32808249211.html?aff_platform=aaf&cpt=1507196510820&sk=Ee2JQzjUz&aff_trace_key=2851e2281aa044d5afeba4e6f38195cd-1507196510820-07986-Ee2JQzjUz&terminal_id=8efd676d65dd4962934ae4c20417db45 =
Есть такие решения: https://ru.aliexpress.com/item/Ball-valve-actuator-ball-valve-electric-ball-valve-coil-controller-Three-line-two-way-control-AC220/32820899030.html?ws_ab_test=searchweb0_0,searchweb201602_2_10152_10065_10151_10068_10344_10342_10343_10340_10341_10307_10301_10060_10155_10154_10056_10055_10054_10059_10534_10533_10532_100031_10099_10338_10339_10103_10102_10169_10052_10053_10142_10107_10050_10051_10084_10083_10080_10082_10081_10110_10111_10112_10113_10114_143_10312_10313_10314_10078_10079_10211_10128_10073_10129_10125,searchweb201603_16,ppcSwitch_4&btsid=31158798-99e1-4294-856e-a6258d091430&algo_expid=f5223749-65da-49f1-8044-e1f81a2ca41d-18&algo_pvid=f5223749-65da-49f1-8044-e1f81a2ca41d= 
Большинство заказывает https://ru.aliexpress.com/item/DN20-G-3-4-AC220V-electric-actuator-brass-ball-valve-Cold-hot-water-Water-vapor-heat/32786907292.html?ws_ab_test=searchweb0_0,searchweb201602_2_10152_10065_10151_10068_10344_10342_10343_10340_10341_10307_10301_10060_10155_10154_10056_10055_10054_10059_10534_10533_10532_100031_10099_10338_10339_10103_10102_10169_10052_10053_10142_10107_10050_10051_10084_10083_10080_10082_10081_10110_10111_10112_10113_10114_143_10312_10313_10314_10078_10079_10211_10128_10073_10129_10125,searchweb201603_16,ppcSwitch_4&btsid=de36db9a-de66-40cd-9bff-f69bf797ffd8&algo_expid=5c2274e7-4df4-460a-bd90-cef1701761d3-0&algo_pvid=5c2274e7-4df4-460a-bd90-cef1701761d3 =
Пользуясь случаем, вопрос к Ренату. Какую информацию ещё нужно для доработки прошивки? Когда можно надеяться на первые результаты?

renat2985, 2017-10-05:
@alexandersciv сегодня создал issues: https://github.com/tretyakovsa/Sonoff_WiFi_switch/issues/38
Если есть желание можете попробовать как работают сценарии.
В ближайшем будущем добавим сценарии для освещения, температуры и сенсора движения. Потом Сергей хочет переехать на асинхронный веб сервер. Это позволит на лету без refresh видеть смену состояния устройств. Потом все остальное. Скорее всего как-то так будем двигаться.

alexandersciv, 2017-10-05:
Новую прошивку установил. Установка прошла штатно. Появился значок треугольника в левом углу. Появились сценарии. Устройство у меня запущено пока только 1 из 10, завтра подключу ещё. Интерфейс  стал загружаться ощутимо быстрее. Работу сценариев проверить смогу после запуска 2 устройства. Судя по видео проделана большая работа по сценариям и само направление, с моей точки зрения, правильное. Вот только без опроса устройства перед действием будут происходить накладки при механическом или действием второго пользователя (жена, дети) со смартфона. У меня сейчас – реле включено с кнопки (горит лампа) – в интерфейсе =включить=. В интерфейсе нажимаю =включить= -лампа гаснет в интерфейсе =включить=. В интерфейсе нажимаю =включить= -лампа загорается. Для света не очень кардинально, так как после первого нажатия кнопка показывает реальное состояние, а для серьёзных приборов желательно дополнительную кнопку =опрос=. В Вашем видео больше десятка устройств, уверен они ещё прибавятся, список уже длинный, при этом с права и слева 2/3 информации которая не важная. Я про выпадающее меню. 
Ещё про название кнопки =включить= и =выключить=. Я, так в начале соображаю в каком состоянии прибор. Думаю =выключен= и =включен= отражает реальное состояние и последующее действие.
С Интернет радио Вы великолепно придумали, голосовое уведомление очень оживит систему. А голосовое управление, думаю тоже скоро. :))

renat2985, 2017-10-05:
@alexandersciv Не пишите в эту ветку, я буду отсюда все удалять. Ваш пост в другой ветке.

N1neTa1ls, 2024-11-18:
Ребят, в итоге получилась какая нибудь аппа из этого? Мне тут по сухому контакту надо на удалёнке котлом рулить, а везде если и предлагается, то только в составе вундервафли по управлению целым домои

renat2985, 2024-11-18:
> Ребят, в итоге получилась какая нибудь аппа из этого? Мне тут по сухому контакту надо на удалёнке котлом рулить, а везде если и предлагается, то только в составе вундервафли по управлению целым домои

Полноценно работает только управление по локальной сети. Удаленное управление только с танцами и бубном..

N1neTa1ls, 2024-11-18:
> Полноценно работает только управление по локальной сети. Удаленное управление только с танцами и бубном..

Ну подозреваю, что комп понадобится, а приложение то по итогу есть или нет?)

pastukhov, 2025-03-01:
У меня такие сценарии работают с помощью зигби термостата и сервера  Home Assistant
Если с НА связываться не хочется, можно поставить "облачный" термостат с WiFi и управлением из инета типа такого https://aliexpress.ru/item/1005005890137082.html?sku_id=12000034720636212&spm=a2g2w.productlist.search_results.8.7dcd7fd1VZUnCM
Тоько подберите правильную модель для управления котлом посредством сухого контакта

N1neTa1ls, 2025-03-01:
> Если с НА связываться не хочется, можно поставить "облачный" термостат с WiFi и управлением из инета типа такого

Спасибо, но я тогда только знакомился с этим всем, хотелось бы ошибки котла читать в случае чего, так как пока там не живу. но по сухому контакту этого не сделать, сухой контакт это чисто по температуре врубать, вырубать для экономии газа)

pastukhov, 2025-03-02:
Да, с сухим контактом ошибки котла не обработать. Тут надо смотреть какая у котла шина opentherm, e-bus или что-то ещё