tretyakovsa/Sonoff_WiFi_switch #38: Бета тестеры.

renat2985, 2017-10-05:
В последнее время вижу здесь активность. Это радует. :)
Мы практически доделали сценарии, цепочки на включение\выключение уже работают. Но их нужно тестировать.
Те у кого есть желание могут попробовать их, обновится можно по этой ссылке:
http://Ваш_IP_Устройсва/upgrade?spiffs=http://backup.privet.lv/spiffs.0xBB000_flash_size_1Mb.256Kb_2017.10.05.bin&build=http://backup.privet.lv/build.0x00000_flash_size_1Mb.256Kb_2017.10.05.bin

Обязательно посмотрите видео, как это работает: https://youtu.be/u8qw-qW9VVo
Просьба писать о всех проблемах сюда.

AVL4GITHUB, 2017-10-05:
Пишусь в бету.

renat2985, 2017-10-05:
>  alexandersciv Новую прошивку установил. Установка прошла штатно. Появился значок треугольника в левом углу. Появились сценарии. Устройство у меня запущено пока только 1 из 10, завтра подключу ещё. Интерфейс стал загружаться ощутимо быстрее. Работу сценариев проверить смогу после запуска 2 устройства. Судя по видео проделана большая работа по сценариям и само направление, с моей точки зрения, правильное. Вот только без опроса устройства перед действием будут происходить накладки при механическом или действием второго пользователя (жена, дети) со смартфона. У меня сейчас – реле включено с кнопки (горит лампа) – в интерфейсе =включить=. В интерфейсе нажимаю =включить= -лампа гаснет в интерфейсе =включить=. В интерфейсе нажимаю =включить= -лампа загорается. Для света не очень кардинально, так как после первого нажатия кнопка показывает реальное состояние, а для серьёзных приборов желательно дополнительную кнопку =опрос=. В Вашем видео больше десятка устройств, уверен они ещё прибавятся, список уже длинный, при этом с права и слева 2/3 информации которая не важная. Я про выпадающее меню.
> Ещё про название кнопки =включить= и =выключить=. Я, так в начале соображаю в каком состоянии прибор. Думаю =выключен= и =включен= отражает реальное состояние и последующее действие.
> С Интернет радио Вы великолепно придумали, голосовое уведомление очень оживит систему. А голосовое управление, думаю тоже скоро. :))

Я не понимаю как вы работаете с веб интерфейсом. Когда человек открывает веб сайт с прошивкой у него все блоки грузятся заново. Сделайте себе ярлык на рабочем столе, и когда вы будите нажимать на этот ярлык у вас откроется сайт и все данные будут актуальными. На телефоне можно сделать точно так же.
Авто обновление статуса устройств появится после того как мы переедем на асинхронный веб сервер.

Не пял какие вы считаете блоки лишними с лева и право. Делайте скриншоты.

renat2985, 2017-10-05:
Обновил сегодняшние бинарники.
[image]

@AVL4GITHUB переведете на украинский?
```
 "Langrgbon":"Включить лед",
 "Langrgboff":"Выключить лед",
 "Langrgbnot":"Переключить лед",
 "Langrelayon":"Включить реле",
 "Langrelayoff":"Выключить реле",
 "Langrelaynot":"Переключить реле",
 "Langjalousieopen":"Открыть жалюзи",
 "Langjalousieclose":"Закрыть жалюзи",
 "Langjalousienot":"Переключить жалюзи",
 "Langjalousiestop":"Остановить жалюзи",
 "Langjalousieturn":"jalousieturn",
 "LangstateJalousie":"Состояние жалюзи",
 "LangstateRGB":"Состояние лед",
 "LangstateRelay":"Состояние реле",
 "LangturnSensor":"Сенсор оборотов",
 "Langtime":"Время",
 "Langweekday":"День недели",
```

alexandersciv, 2017-10-05:
Перепрошил по ссылке http://Ваш_IP_Устройсва/upgrade?spiffs=http://backup.privet.lv/spiffs.0xBB000_flash_size_1Mb.256Kb_2017.10.05.bin&build=http://backup.privet.lv/build.0x00000_flash_size_1Mb.256Kb_2017.10.05.bin. Обновление не появилось.

renat2985, 2017-10-05:
@alexandersciv попробуйте обновить (F5) страницы на которых вы не видите изменений.
Последнее мое обновление которое коснулось надписей в выпадающих меню, оно пока работает только в русской версии. 
Если изменения не появились после refrash (нажатия на F5) попробуйте зайти на IP/lang/lang.ru.json и обновить. У вас скорее всего проблема с кешем.

AVL4GITHUB, 2017-10-06:
"Langrgbon":"Включити  LED",
 "Langrgboff":"Виключити LED",
 "Langrgbnot":"Переключити LED",
 "Langrelayon":"Включити релє",
 "Langrelayoff":"Виключити релє",
 "Langrelaynot":"Переключити релє",
 "Langjalousieopen":"Відкрити жалюзі",
 "Langjalousieclose":"Закрити жалюзі",
 "Langjalousienot":"Переключити жалюзі",
 "LangstateJalousie":"Стан жалюзі",
 "LangstateRGB":"Стан LED",
 "LangstateRelay":"Стан релє",
 "LangturnSensor":"Сенсор обертів",
 "Langtime":"Час",
 "Langweekday":"День тижня",

renat2985, 2017-10-06:
@AVL4GITHUB спасибо. И еще парочку:
```
 "LangcolorRGB":"Цвет лед",
 "LangspeedRGB":"Скорость лед",
 "LangbrightnessRGB":"Яркость лед",
 "LangmodeRGB":"Динамический режим лед",
```

AVL4GITHUB, 2017-10-06:
"LangcolorRGB":"Колір LED",
 "LangspeedRGB":"Швидкість LED",
 "LangbrightnessRGB":"Яскравість LED",
 "LangmodeRGB":"Динамічний режим LED
",

renat2985, 2017-10-06:
Спасибо. Таким образом мы сейчас можем Включать\Выключать\Переключать устройства, по отслеживаемым данным:
Реле: Состояние (Включено\Выключено)
LED: Состояние (Включено\Выключено) Установленный цвет, скорость, яркость, динамический режим
Жалюзи: Состояние (Включено\Выключено), сенсор оборотов.

Скоро добавим слежку за сенсором движения, температурой и освещением.

alexandersciv, 2017-10-06:
Установил прошивку с исправлениями. Установка прошла штатно. Изменения появились и я считаю, что это дополнение переводит прошивку на качественно новый уровень. Теперь появилась возможность управления одного устройства другим по его показанию и согласно перечню сценариев, уже готовых и возможно изменённых пользователем. При этом все устройства в едином локальном пространстве даже без создания сервера управления.

renat2985, 2017-10-06:
@alexandersciv, да. Пробовали сами создавать сценарии, какие-то глюки заметили?

alexandersciv, 2017-10-06:
Пока готовлю второе устройство. После включения и тестирования обязательно отпишусь. Заметил, что в крайней версии прошивки не сохраняется настройка в =Подключение к Wi-Fi роутеру= BLINK GPIO...

AVL4GITHUB, 2017-10-06:
Етот глюк был и раньше, обещали исправить.

renat2985, 2017-10-06:
@alexandersciv спасибо! Исправил, теперь BLINK сохраняется.
@AVL4GITHUB хм, я не видел что писали о проблеме с BLINK.

Выложил новые bin файлы. Так же появился фильтр на команды:
[image]

alexandersciv, 2017-10-06:
**@renat2985**  очень радует скорость доработки при обнаружении недостатков в прошивке, создаётся мнение о волшебстве :) 
Как доработать SONOFF RF и подключить датчики я разобрался, поскольку много информации в Интернете. А вот как подключить правильно датчик движения и ЛЕД ленту, не нашёл. Подскажите, кто знает.

renat2985, 2017-10-06:
@alexandersciv, :) стараемся.
По поводу Sonoff-rf создайте для него другую тему, отвечу.

AVL4GITHUB, 2017-10-06:
В морде не работало изменение/сохранение настройки  времени и пина подключения к ВайФай роутеру. Я писал об этом, цитирую:
Я ставли или 30 секунд или минуту, а также указываю мигальный пин 13.
Несколько раз жму СОХРАНИТЬ, но при обновлении страницы получаю старое значение (10сек) и светодиод не мигает при перезагрузке.

renat2985, 2017-10-06:
@AVL4GITHUB прошу прощения не помню такого. Проверьте сейчас все ок с этим, у меня вроде работает.

alexandersciv, 2017-10-06:
**@renat2985** Прошился крайней версией. Очень радует хорошая скорость перепрошивки и отзывчивости интерфейса. Изменения стали сохраняться, но  BLINK GPIO 12 при запуске включается реле и светодиод, после запуска тухнет и отключается. Во всех остальных случаях светодиод не включается.
Прошу прокомментировать значения BLINK GPIO, в видео где то упоминалось, не смог найти.
У меня  SONOFF RF.

alexandersciv, 2017-10-06:
Запустил второе устройство. Прошил последней версией. Прописал простой сценарий. При вкл 1 = вкл 2 устройство.  Тамже Второй сценарий: При выкл1 = выкл2 устройство. 
Реально работает! Даже оптимизм появился, что можно запустить удалённый термоморегулятор и другие устройства.
Молодец Ренат! Спасибо за труды!

alexandersciv, 2017-10-06:
Работает даже когда задаёшь противоречивый сценарий: Там же Третий для устройства2:  откл2 когда вкл 1. 
Реально не хватает кнопки =**опросить**= поскольку при изменении статуса с другого устройства кнопка не меняется.

renat2985, 2017-10-06:
@alexandersciv  https://youtu.be/7z_xcVtL0aE?list=PL6NJTNxbvy-IPTDQk8XjTV41oRrFafrRi&t=35

> Реально не хватает кнопки =опросить= поскольку при изменении статуса с другого устройства кнопка не меняется.

Не понял вас. Объясните подробней. Если вы о главной странице, пользуйтесь кнопкой обновить:
[image]
Просто не оставляйте странице открытой в браузере, заходите на нее как я вам предлагал ранее. Тем самым вы будите получать актуальную информацию. Вообще не вижу в этом проблемы.

alexandersciv, 2017-10-07:
По кнопке =опросить= вопрос снимается. =Если вы о главной странице, пользуйтесь кнопкой обновить:= этого достаточно. Проверено, изменение состояния механически кнопкой, на смартфоне другим пользователем при нажатии =обновить= показывает реальный статус устройства.

tretyakovsa, 2017-10-07:
Уже сделали. И можно подключить приемник 433 мГ.

alxnovo, 2017-10-07:
Что сделали?

renat2985, 2017-10-07:
@alxnovo это вот такие железки: :)
[image]
Они могут поймать частоты 433mhz или 315mhz. Это как правило пульты от кондиционера, некоторые китайские сигнализации, так же некоторые пульты от розеток. 

Заходим в сценарии, выбираем "Беспроводной приемник", далее нажимаем на пульте и с право на лупу. У вас появится код кнопки с пульта и делаете дальше сценарий, что должно произойти после того как вы нажали на кнопку пульта. :)
[image]

alxnovo, 2017-10-07:
Понял, вы про Sonoff RF

А к каким пинам подключать RF-ресивер (на картинке вроде ресивер) и получается он умеет как принимать, так и передавать сигнал?
У меня по RF могут управлять некоторые люстры и iRobot, если сделать ещё и, наоборот, нажимая кнопку в интерфейсе через RF управлять люстрами, розеткам и другими устройствами (в моём случае пылесосом iRobot) было бы совсем супер - ночной режим освещения, запуск робота когда все ушли или удалённо.

P.S. я пользуюсь исходниками с GitHub, мне бинарники для управления ролетами не подходят

renat2985, 2017-10-07:
> А к каким пинам подключать RF-ресивер (на картинке вроде ресивер) и получается он умеет как принимать, так и передавать сигнал?

Это вам нужно установить "Тип устройство: manually", там увидите RCSwitch, поставьте на него галочку и установите нужный вам порт.
Эта железка умеет только принимать. Для отправления нужна другая. - Скоро внедрим и ее.

tretyakovsa, 2017-10-07:
Это не про sonoff RF там так сделать нельзя.

alexandersciv, 2017-10-07:
**@tretyakovsa** если это не =Это не про sonoff RF= то про какое устройство?

alexandersciv, 2017-10-07:
Перепрошился. Новые сценарии не появились. Старые настройки сценариев исчезли (как их сохранять?). Изменения стали сохраняться и в  BLINK GPIO.

renat2985, 2017-10-08:
Выложил новые бинарники.
Теперь в любом типе устройства доступны все модули +- другие мелочи.

http://192.168.1.15/upgrade?spiffs=http://backup.privet.lv/spiffs.0xBB000_flash_size_1Mb.256Kb_2017.10.08.bin&build=http://backup.privet.lv/build.0x00000_flash_size_1Mb.256Kb_2017.10.08.bin

alxnovo, 2017-10-08:
С новой прошивкой, как добавились сценарии, Safari 8.0.7 перестал отображать интерфейс управления :( Safari 10 всё нормально. 
Вообще, Appl'овские девайсы, точнее их браузеры, не очень то "любят" эту прошивку...с большим скрипом открывают страницы, потом, как прокэшируют, открываются с терпимой скоростью.

[image]

renat2985, 2017-10-08:
@alxnovo  Попробуйте другие дизайны. Я какраз в используемый вами дизайн добавил экспериментальны код.

alexandersciv, 2017-10-08:
Подключил 3 модуль с датчиком температуры. Прошил последней версией все модули, настройки сценариев не сохранились, в версии прошивки 171001. Сценарии по включению и выключению лампочек работают. По датчику температуры  и влажности сценарии не нашёл.

renat2985, 2017-10-08:
@alexandersciv сценарии после обновления пока действительно не сохраняются. Сценарием на температуру и влажность пока нет.
На сегодняшний день актуальная прошивка: http://192.168.1.15/upgrade?spiffs=http://backup.privet.lv/spiffs.0xBB000_flash_size_1Mb.256Kb_2017.10.08.bin&build=http://backup.privet.lv/build.0x00000_flash_size_1Mb.256Kb_2017.10.08.bin

alexandersciv, 2017-10-08:
**@renat2985** у меня именно эта прошивка, но показывает
[image]

renat2985, 2017-10-08:
@alexandersciv то что вы мне показываете это официальные стабильные прошивки. Там нет того что я выкладываю здесь :) здесь для бета тестеров :)
И почему вы именно BUILD открылb f yt UPGRADE BUILD & SPIFFS?

http://192.168.1.15/upgrade?spiffs=http://backup.privet.lv/spiffs.0xBB000_flash_size_1Mb.256Kb_2017.10.08.bin&build=http://backup.privet.lv/build.0x00000_flash_size_1Mb.256Kb_2017.10.08.bin

Обратите внимание на эту ссылку, здесь указано 2 bin файла 1 это файловая система spiffs.0xBB000_flash_size_1Mb.256Kb_2017.10.08.bin
другой прошивка build.0x00000_flash_size_1Mb.256Kb_2017.10.08.bin
В имени файлов можно увидеть что они скомпилированы для ESP с 1Mb flash памятью. И в конце ее дата сборки.

Только что сделал еще 1 фикс и обновил сегодняшние bin файлы. Если вы еще раз обновитесь, то нажав на красный треугольник увидите установленную вами прошивку. Но не жмите на "ОК" т.к. он вам откатит на стабильную версию.
[image]

alexandersciv, 2017-10-08:
Перепрошился. Всё показывает как надо. 
Когда можно ожидать сценарии с датчиками температуры?

renat2985, 2017-10-08:
По этим вопросам к Сергею. Думаю скоро.

alxnovo, 2017-10-08:
@renat2985 
> попробуйте другие дизайны, они работают нормально?

На других дизайнах такая же петрушка с интерфейсом, НО это на сборке из исходников на GitHub'е от 07 октября. 

_Вчера прошил 2 модуля на основе этих исходников - встало с первого раза, а на третьем никак не мог правильно поставить, постоянно вот такой интерфейс был во всех браузерах. Что только не делал и заливал пустышку в первый мегабайт и во все 4 мегабайта и использовал разные варианты прошивок с разными файловыми сборками, часов 6 провозился, в итоге поставил через установку сначала старой прошивки и файловой системой, а сверху уже последнюю (прошивки компилировались из исходников)_

[image]
[image]

А так 3 модуля работают совместно, простейшие сценарии (на модулях "жалюзи" работают. Блин, накаркал, перезагрузил модуль в спальне (тот третий, что мне до 4 утра не давал спать), так теперь на него заходишь, он показывает другие модули, но не себя - это в Chrome где проблем с интерфейсом не было.

[image]

а если зайти на другие устройства, то "Спальня" нормально видна

[image]

И вот фиг поймёшь кто или что виноват :( Модуль в спальне отличается от других только тем что у него статический адрес, при этом он самый медленный из всех по интерфейсу и самый далеко расположенный от роутера, но пингуется он нормально, после 3-4 запросов (у всех модулей первые пинги долгие)

[image]

AVL4GITHUB, 2017-10-08:
Меня тоже, для начала, интересует исключительно реакция на датчики температуры.

renat2985, 2017-10-08:
@alxnovo проблемную ESP попробуйте обновить следующем образом, посмотрите видео, оно с привязкой по времени на нужном вам моменте: https://youtu.be/D0Fu8etyVg0?list=PL6NJTNxbvy-IPTDQk8XjTV41oRrFafrRi&t=260

alxnovo, 2017-10-08:
@renat2985 "Я женат, мне нельзя в bin'льдяжки" ;)
Нельзя мне бинарники ставить пока. Если их ставить без доработки под ролеты, то у меня китайские релюшки прикипают и семисторы горят, я так чуть мотор в ролетах не сжёг когда на все обмотки ток пошёл. Повезло что в моторе защита оказалась (я так думаю) иначе бы потёмках сидел, т.к. ролеты только электроприводом открываются.

renat2985, 2017-10-09:
@alxnovo тем самым вы смогли бы понять, проблема в вашем компиляторе (Arduino) или прошивке.

alxnovo, 2017-10-09:
Я бы с удовольствием протестировал, но боюсь потом замучаюсь возвращать рабочую версию прошивки. Надо доехать до магазина и купить ещё пару модулей Witty Cloud на поиграться, текущие все задействованы на ролетах и отказываться от их незаметной работы уже не хочется, к комфорту быстро привыкаешь :)

renat2985, 2017-10-11:
Всем спасибо, бета тестирование закрываем. Все bin файлы удалили. Сегодня, добавлю их в официальный релиз.

closed 2017-11-13 by renat2985