tretyakovsa/Sonoff_WiFi_switch #39: Автоматическая подсветка лестницы

nagornyak, 2017-10-05:
Ivan Nagornyak
Случайно наткнулся на ваш проект и не понимаю как не видел его раньше)) Спасибо, замечательная идея и реализация! А теперь вопрос. Не планируете ли создать автоматическую подсветку лестницы на его основе с использованием двух сонаров? Пинов у есп вроде как раз хватает.﻿
Коротко алгоритм работы. Если освещение меньше заданного включается режим "ночь" и слегка подсвечивается первая и последняя ступеньки. При наближении к одному из сонаров включается плавное зажигание каждой из ступенек с определенным интервалом в направлении движения. через определенное время в том же направлении ступеньки плавно гаснут, а первая и последняя притухает. 
В данный момент есть поделка на ардуино и HC-SR04 (HC-SR05 говорят надежнее) и обычной одноцветной ленте, но хочется перейти на RGB (в зимние праздники можно использовать как гирлянду) и ws2811 дабы из 16 проводов сделать не 46 а всего 3))  . Pir не совсем подходит так как улавливает много движения возле лестницы. Согласен, что если использовать его и сценарии, то можно все организовать штатными средствами без переделок.

renat2985, 2017-10-05:
Датчик света поддерживает наша прошивка. Но он пока не поддерживается в сценариях, дума в ближайшие недели добавим.
Далее, подсвечивать первую и последнюю ступень. Идея конечно интересная, но я не думаю что кто-то это будет делать, т.к. там понадобятся дополнительные настройки. На данном этапе проект работает с WS2811/WS2812/NeoPixel LEDs и поддерживает около 44 динамических режимов, в каждом из них можно настроить скорость, яркость, цвет, там так же есть бегущие огоньки, я думаю вы обязательно что-то подберете из этого количества себе.
Но тут опять есть проблемы, если вы будите использовать более 100-130 лед (это где-то 3-4м), динамические режимы будут немного подтормаживать, ESP тяжело справляться с большим количеством огней.

Санар пока не поддерживает, идея интересная, думаю когда ни будь обязательно добавим и его.

Я в свое время брал колпачок от духов отрезал 1 сторону, получалась трубка 3-4см длинной и одевал ее на PIR сенсор, за счет чего можно значительно уменьшит его радиус чувствительности... так же как правило там можно настроить чувствительность.

nagornyak, 2017-10-06:
Спасибо за рекомендации. Диодов у меня будет всего 15 (маленький модуль ws2811 с обвязкой и силовым ключом на каждую ступеньку, а дальше почти метр обычной ргб ленты), так что есп справится без проблем. В принципе я сам перенесу все это из ардуино на есп, основной функционал итак будет работать. Хотелось просто веб-интерфейс и дополнительные эффекты, собственно так и наткнулся на вашу прошивку и уж очень она мне понравилась))) Свой функционал думаю сделаю в ближайшие дни, как время появится, а вы при желании и по возможности интегрируете его в свой проект))