tretyakovsa/Sonoff_WiFi_switch #40: Название данного проекта ???

grossbel, 2017-10-05:
Уважаемые разработчики, свое детище называть устройством, ну как то неправильно, ДАЙТЕ ИМЯ ПРОЕКТУ. Не можете сами придумать, объявляйте конкурс, вдруг кто что интересное предложит.

renat2985, 2017-10-05:
Сделаем. Назовем, название уже есть. Потерпите. :)
Просто первоначально проектов было 3. Прошивка конкретно для sonoff с очень маленьким функционалом, прошивка для жалюзи и проект для led ленты.
https://github.com/tretyakovsa/Sonoff_WiFi_switch
https://github.com/renat2985/rgb
https://github.com/tretyakovsa/jaluzi
Они никак с друг другом небыли связаны. Со временем так получилось что больше активности стало именно в этом проекте. И проект уже разросся и поглотил 2 других проекта. По сути в них сейчас находится полностью идентичный код что и в этом репозитории.

Доделаем сценарии +- что-то еще, потом переедем на другой репозиторий. :) И возможно bin файлы уже там будут собираться для 4Mb flash памяти, т.к. 1Mб уже становится мало.

AVL4GITHUB, 2017-10-06:
Я сразу закупил 4 метровых флешек для апгрейда девайса. Стоят копейкы, а  "запас памяти костей не ломит".