tretyakovsa/Sonoff_WiFi_switch #41: SONOFF доработка и применение

alexandersciv, 2017-10-06:
В Китае появилось пока несколько производителей выпускающих готовые решения на базе модуля ESP8266, думаю в ближайшее время появятся и на базе ESP32. Один из  них https://www.itead.cc/smart-home.html под маркой  SONOFF. Считаю что эти устройства заслуженно пользуются популярностью за свою функциональность и небольшую стоимость (сейчас заказать с фабрики дешевле, чем у других продавцов).
Для своих задач автоматизации дома перебрал много разных вариантов устройств и прошивок, в свете новых доработок прошивки из этого топика. Из совокупности цена + функционал + доработка + компактность, лучший вариант  SONOFF + прошивка Sonoff_WiFi_switch.
Предлагаю в этой теме публиковать варианты доработки и применения SONOFF.

alexandersciv, 2017-10-06:
Как доработать SONOFF RF и подключить датчики температуры и влажности я разобрался, поскольку много информации в Интернете (могу дать ссылки и поделиться своими решениями). 
А вот как подключить правильно датчик движения и ЛЕД ленту, не нашёл. 
Подскажите, кто знает.

renat2985, 2017-10-06:
В SONOFF RF к сожалению есть только 1 дополнительный PIN GPIO 14 к которому вы можете подключить 1 датчик.

[image]

В нашей прошивке это делается следующем образом. Т.к. вы прошивали Sonoff скорее всего на первой странице настройки вы выбрали "Sonoff-rf / Sonoff / Wi-Fi Smart socket", зайдя на страницу конфигурации, вы можете сменить его на другой тип или настроить нажав голубую кнопку "EDIT"

[image]

По дефолту на 14 ножке стоит датчик тимпературы. Как только вы его подключите к пину, он сразу появится на главной странице. Там же есть возможность снять галочки с датчиков тимпературы и поставить на сенсор движения. Но нужно тестировать эту функцию, возможно это все не работает, попробуйте.

Вы так же можете сменить Тип устройства допустим на Manually, и тогда у вас появится больше возможностей, на 14 ножку вы сможете поставить лед ленту. Но опять же с sonoff это нужно тестировать. Я не проверял.

Да и в бета версии появилось возможность в реле ставить его состояние при включение. То есть если вы хотите что бы реле автоматически включалось после запуска ESP вам нужно поставить единичку, если хотите что бы оно было в разомкнутом состояние ставьте 0.
[image]

tretyakovsa, 2017-10-07:
Кроме этого можно использовать RX и ТХ для подключения. TX это GPIO1. RX это GPIO3.
Попробуйте RX на датчик движения, TX для LED,

renat2985, 2017-10-07:
Именно так, совсем забыл про них. TX выход цифра 1, RX выход цифра 3.

alexandersciv, 2017-10-07:
Постарался найти что про GPIO SANOFF пишут в Интернете. Очень хорошая статья https://mysku.ru/blog/china-stores/45793.html  
SONOFF TH10/16
GPIO0 — кнопка 
GPIO12 — красный светодиод и реле
GPIO13 — синий светодиод
GPIO14 и GPIO4 выведены на разъем подключения датчиков.
Причем оба датчика однопроводные и используют GPIO14 
GPIO4 еще нужно включить перемычкой на плате
GPIO2 и GPIO15 имеют подтягивающие резисторы на плате, можно подпаяться к ним.
GPIO5 и ADC вообще никуда не распаяны и нужно подключаться прямо к микросхеме ESP.

alexandersciv, 2017-10-07:
https://mysku.ru/blog/china-stores/47080.html
SONOFF LED - умный светодиодный драйвер
[image]

alexandersciv, 2017-10-08:
Буду выкладывать ссылки на материалы которые нахожу в Интернете. Надеюсь кому то будет полезно. А у нас найдутся специалисты которые на практике смогли сделать доработку и поделятся своими решениями на русском языке, поскольку в основном все решения не на русском.
**Добавление RF в Sonoff**  http://tinkerman.cat/adding-rf-to-a-non-rf-itead-sonoff/ 
[image]

alexandersciv, 2017-10-09:
Предполагаю, что в ближайшее время прошивка в 1 МБ памяти на модуле  Sonoff помещаться не будет. Есть несколько решений по перепайке схемы памяти: 4 МБ ( https://tech.scargill.net/32mb-esp01/ ) и 16 МБ ( http://www.instructables.com/id/ESP-03-Upgrade-Flash-Memory-to-128-M-Bit/ ).
Но  эти модули должны поддерживаться прошивкой. По скольку большой разницы 4 или 16 в стоимости и работе по перепайке нет, вопрос к Ренату. 
**Какую память будет поддерживать прошивка?**

renat2985, 2017-10-09:
@alexandersciv, не знаю. Я бы на вашем месте посмотрел лучше на другие модули. Например то что предлагаем мы http://www.onclick.lv/ru/shop-detail-chip.html или что-то подобное. 
Мне не нравится sonoff у них плохая антенна, они часто отваливаются от роутера, за частую если его поставить в другую комнату он вообще не увидит роутер и не будет работать. Я стараюсь использовать модули с внешней антенной которые стабильно могут пробить через 2-3 стенки, у которого много дополнительных выходов. На которые можно поставить и фото резистор и много других датчиков и реле. На sonoff этого сделать нельзя.

alexandersciv, 2017-10-09:
**@renat2985** Безусловно, Ваш модуль более функционален, на него можно повесить массу задач. Но для большинства случаев достаточно 2-3 действия: по датчикам (свет,движение, температура) включить или выключить нагрузку sonoff как законченное коробочное решение вполне справится+цена вполне умеренная. С проблемами WiFi я борюсь следующим образом. В каждое помещение подтянута витая пара для локалки + умдом(с HAME MPR-A1 на конце). Если нет возможности локалки, репитеры на базе HAME MPR-A1 (http://4pda.ru/forum/index.php?showtopic=377187&st=5160#entry65809862). Сеть настроена параллельно (https://esp8266.ru/esp8266-openwrt-mosquitto-mqttwarn-thingspeak-email-android-ios-twitter-cloudmqtt/#wifi). Для более сложных задач буду использовать ESP32. 
Вот если очередная Ваша прошивка не влезет в стандартный  sonoff, будет плохо. Поэтому заранее хочу увеличить память. На сколько?

alxnovo, 2017-10-09:
@alexandersciv не осилил написанное на 4pda, чем хорош этот HAME MPR-1? Как я понял у вас в каждую комнату, где это возможно, проложено несколько витых пар...к одной паре подключен HAME MPR-A1 как репитер (с провода на wifi), к другой паре (или парам) УД + по WiFi подключаться модули на ESP
Первое - не лагает такая сеть у которой на концах HAME MPR-1, мобильные девайсы не тупят при переходе с одной зоны на другую (когда между роутерами идёт переключение)? Или на HAME MPR-1 построена параллельная сеть для IoT?
Второе - если что-то случится с УД и/или ESP то как организовали ручное управление хозяйством?

Память в идеале увеличивать до 4Mb, т.к. это сейчас золотая середина в ESP и все будут ориентироваться на этот объём в ближайшем будущем.
Хотя если SONOFF будет как исполнительное устройство или как датчик, то смысл воротить на него многофункциональную прошивку, отключили лишние модули и ненужные страницы и в текущий объём всё влезет.

По поводу DIY-модуля с ESP для своих экспериментов с ролетами я изначально рассматривал [Wifi IoT Relay Board Based on ESP8266](http://www.electrodragon.com/product/wifi-iot-relay-board-based-esp8266/) Не помню как на него попал, но я тогда не знал про SONOFF (я вообще про такие гаджеты ничего не знал) и когда попалось описание SONOFF, то мне он показался каким-то уж простым и отсталым по сравнению с модулем от Electrodragon:
- ESP12 со всеми вытекающими...
- 2 реле на борту (мне это изначально нужно было чтобы поднимать и опускать ролеты)
- предусмотрено сразу место под DHT-22 - впаял и работает
- есть свободные пины для подключения дополнительных датчиков
- есть впаянные пины для прошивки
- стоит столько же как и SONOFF

минусы по сравнению с SONOFF:
- чуть длинее на 2см и капельку шире
- нет изначальной френдли-прошивки и нативного приложения для смартфонов

Сам никогда ни один девайс в руках не держал, опыт чисто на теории :)

P.S.  сейчас наигрался раздельными модулями, параллельно хочу найти большой модуль на ESP c 8-ю реле (чтобы управлять 4 ролетами с одного устройства), наподобие [Sonoff 4CH](https://www.itead.cc/smart-home/sonoff-4ch.html) только на 8CH, никто такой модуль не встречал?

alexandersciv, 2017-10-10:
**@alxnovo**  по порядку: HAME MPR-1 привлекает тем, что полнофункциональный роутер, небольшие размеры, не большая цена (https://ru.aliexpress.com/item/COTS-Hot-Mini-Portable-USB-3G-4G-WiFi-150Mbps-Wireless-Router-WCDMA-CDMA-Modems/32748325942.html =покупал за 405 руб.) не большое энергопотребление, есть версии со встроенным аккумулятором (нет света), навороченная прошивка (http://4pda.ru/forum/index.php?showtopic=709298) в неё даже встроили пакеты для домашней автоматизации .
Моё выполнение автоматизации в начальной фазе. Был твёрдо намерен строить на MegaD-2561 (https://www.ab-log.ru/smart-house/shop) очень интересный проект, много разумного, а главное реализованное Андреем пригодилось мне при осмысливании практического применения решений. Поэтому я на стадии строительства подтянул в каждое помещение  2 витых пары. Одну для локальной сети, вторую для 1-wire.  Всё свёл в этажный щит, а из него в серверную (как оказалось построил опорную сеть :)) WiFi сеть по дому уже работает стабильно в одном адресном пространстве. Модули УмДома будут включены в режиме моста, сиды скрытые ссылку дублирую (https://esp8266.ru/esp8266-openwrt-mosquitto-mqttwarn-thingspeak-email-android-ios-twitter-cloudmqtt/#wifi)
Везде электроника будет дублироваться механическим управлением (как сейчас и работает).
Хотя Ренат пока и не дал чёткую рекомендацию, как кой объём памяти необходим, я тоже склоняюсь к 4 МБ, а поскольку я только учусь и для меня пока сложно отключать и включать дополнительные модули в прошивке я жёстко привязан к наработкам Рената и Сергея. Вообще решения этих ребят очень оригинально, хотя их прошивку встретили на www.ab-log.ru прохладно, но там кроме ESP8266 MegaD api (https://www.ab-log.ru/forum/viewtopic.php?f=1&t=1130&sid=0abcfa40a9f17692e0cf572579900cb6#p20670 ) как то вяло реагируют на радио решения. Для меня, как новичка, эта прошивка идеальна, даже wifi-iot.com и Blynk не имеют такого функционала.
Готовые решения на http://www.electrodragon.com/product/wifi-iot-relay-board-based-esp8266/ я рассматривал, но они мало популярны и считаю малонадёжны (статья http://forum.amperka.ru/threads/Сделать-управление-отоплением.9164/).
У SONOFF, как отмечалось в статье (https://mysku.ru/blog/china-stores/45793.html ) =Самый большой недостаток данных модулей то, что часть GPIO не разведены на разъемы. Ну что стоит сделать компании ITEAD гребенку с 8 pin и завести туда все свободные GPIO от ESP. Я думаю, популярность бы таких модулей при существующей цене сильно бы выросла.=
Вроде мне удалось расшевелить фабрику, обещали подумать.

alexandersciv, 2017-10-10:
**_Шлюз WI-FI-nRF24L01 на базе ESP8266 для обмена данными с nRF24L01 и nRF24LE1_**
[http://homes-smart.ru/index.php/oborudovanie/bez-provodov-2-4-ggts/63-shlyuz-wi-fi-nrf24l01-na-baze-esp8266-dlya-obmena-dannymi-s-nrf24l01-i-nrf24le1](url)
Обновлено 31.03.2016 15:50 | Автор: MaksMS |
=Не смотря на огромные возможности ESP8266 этот чип не всегда подходит для всех задач, например в плане энергоэкономных датчиков.  При использовании модуля "RF24 хаб" в конструкторе прошивки ESP8266 возможно использование датчиков и других устройств на базе nRF24LE1 или nRF24LO1+Arduino(или любой другой микроконтроллер). ESP8266 выступает в качестве хаба, который принимает данные с таких устройств с возможностью передачи данных на сервера, а так же отображение данных на главной странице модуля.=

alexandersciv, 2017-10-12:
В теме https://github.com/arendst/Sonoff-Tasmota/issues/484 предлагается, как я понимаю, решение подключения обычного выключателя к Sonoff. Может есть знающие, рабочие эти схемы?
[image]
[image]

renat2985, 2017-10-12:
@alexandersciv схемы правильные, только зачем. В sonoff и так GPIO мало. Лучше подпаяйте провода к уже существующей на нем кнопке и подключайте к нужному вам выключателю.

[image]

alexandersciv, 2017-10-12:
**@renat2985**  У меня все выключатели с фиксированным положением, кнопка в Sonoff не фиксируется, поэтому и приходится жертвовать GPIO.

Paly4, 2017-10-13:
я пробовал подпаиваться паралельно к кнопке , провода сантиметров по 15 , но получается полная ерунда , малейшая искра при включении в соседнюю розетку вызывает срабатывание реле
вот как бы побороть эту проблему ? ...пробовал даже экранированным проводом - не помогает
@alexandersciv . есть выключатели , на вид как обычные , но без фиксации - звонковые называются

alexandersciv, 2017-10-13:
**@Paly4** Про =звонковые= выключатели понятно, но задача подключиться к обычному выключателю. У некоторых моделей выключателей есть возможность доработки, у меня нет.

edgiru, 2018-09-07:
@alexandersciv @renat2985 тоже интересует эта тема причем хочется оставить выключатель прямо на месте как резерв а соноф как дополнение удобное. Как сделать?

ndv27, 2019-03-21:
> **@Paly4** Про =звонковые= выключатели понятно, но задача подключиться к обычному выключателю. У некоторых моделей выключателей есть возможность доработки, у меня нет.

Обычные выключатели (не звонковые) начинают работать как переключатели без фиксированного положения. Но нужно настраивать для этого в прошивке режим работы  GPIO14. Там есть разные варианты.

tretyakovsa, 2019-03-23:
@ ndv27 Дима ничего себе какая встреча! Сколько лет мы с тобой не виделись. Звони на скайп пообщаемся.

ndv27, 2019-03-25:
Привет, Серега!

Мелькнула мысль, что это ты, но проехала.
Куда звонить-то? Ника Skype не нашел...

tretyakovsa, 2019-03-25:
@ ndv27 Скайп vtplus