tretyakovsa/Sonoff_WiFi_switch #42: Управление температурой в квартире (регулятор для радиаторов/батарей)

alxnovo, 2017-10-08:
Начался отопительный сезон и в квартире начали шпарить батареи, подняв, температуру до некомфортно высокой. Самый простой способ это ограничить горячую воду в батареи с помощью ручных и автоматических клапанов ([термоголовок](http://otoplenie-guide.ru/oborudovanie/komplektuyushie/termostaticheskaya-golovka))
[image]
Ручной это когда крутя ручку ты ограничиваешь теплоноситель на постоянку, термоголовка закрывает-открывает поток в зависимости от внешней температуры и установки необходимой температуры на нём.
Теоретически термоголовка должна автоматически решать 90% потребностей в регулировании температуры, но как показывает моя практика термоголовка это аналог ручного вентиля не более. Почему так происходит:
- слишком близко к батареи находится термоголовка. Тут ничего не поделаешь, как установили сантехники клапан (на который одевается либо ручной регулятор, либо термоголовка) тем и пользуйся. При такой установке термоголовка нагревается от самой батареи, а не реагирует на температуру в комнате.
- почти всегда, те же сантехники или родственники ставят клапаны вертикально и, соответственно, термоголовка на них ставится в том же направлении. Получается на входной трубе (самой горяечей) стоит вертикально термоголовка и теплые потоки от трубы её нагревают, опять она регулирует совсем не ту температуру.
- сами радиаторы в квартирах обычно закрыты сверху подоконником, а с спереди шторами или мягкой мебелью (у некоторых радиаторы закрыты экранами или утоплены внутрь стены) и микроклимат и температурный режим возле батареи сильно отличается от того, что происходит в комнате. Установленная на радиаторе термоголовка опять же оперирует не той температурой.

Раньше считалось (да я и сам так считал) что ничего страшного в том, что термоголовка в другой температурной зоне находится, просто её нужно открыть больше на несколько градусов (на разницу температур между комнатой и у батареи) и она будет нормально работать. Но всё равно нужно было всё время подкручивать - то холодно, то жарко. Прошлой зимой мне сантехник объяснил, что температура теплоносителя варьируется в котельной в зависимости от температуры на улице, чем ниже температура на улице, тем горячее теплоноситель и батареи шпарят по полной и термоголовка закрывает поток, т.к. находится в "своём" микроклимате, хотя в комнате может быть холодно. Когда нибудь температура в комнате и батареи сравняется, но за это время уже успеешь замёрзнуть и подрегулировать настройки термоголовки - сначала температура поднимется до комфортной и продолжит подниматься, ведь настройки сделали на более высокую температуру и так по кругу...то холодно, то жарко. Термоголовка, конечно, лучше регулирует температуру, чем ручные вентили, но всё-равно, не идеально.
В итоге я пришёл к выводу что нужна регулировка батарей с выносным датчиком температуры. Есть термоголовки с выносными датчиками, но это те же яйца, только в профиль (датчик можно вынести на несколько десятков сантиметров, поэтому его расположить можно только на той стене где и батареи, а это опять за шторами и т.п.)

Это я всё плавно подвожу под проект Сергея и Рената :)

Нужен электрический "наболдажник" на радиатор которым что-то (например, ESP8266 c прошивкой smart-room) управляет в зависимости от температуры в комнате. Алгоритм очень похоже на тот, что хочет @AVL4GITHUB в теме [Управление газовым котлом отопления. Нужен такой девайс](https://github.com/tretyakovsa/Sonoff_WiFi_switch/issues/37) - по датчику температуры регулировать электропривод клапана радиатора.

Мало того, что нужно дописать под это устройство прошивку, так нужно это устройство ещё и найти. Никто не интересовался этой темой, может уже есть решения и я изобретаю велосипед?

Основные мои требования от этого устройства:
- поддерживать **постоянную** заданную температуру в комнатах за счёт отопления
- возможность устанавливать разную температуру по времени суток (днём выше, ночью ниже - чтобы спалось лучше).

alxnovo, 2017-10-08:
По устройствам...

Начал искать устройство которое может работать автономно, как по логике, так и по питанию. Важнее чтобы работал от своего питания, вести проводку к батареям не хочется (когда делал ремонт хотел к батареям подвести слаботочку, но забыл в процессе).
Так же это устройство должно управляться извне, в идеале, по WiFi, но для работающего от своего питания это вряд ли возможно. Тогда по любому радиоканалу который можно прикрутить к ESP и smart-room.
Внешний эстетический вид тоже важен, всё-таки в комнате устанавливать надо, поэтому китайские редукторы на запорную арматуру не подходят.

Первое что начал искать это на китайской алибабе, либо плохо искал, либо у них таких нет. Нашёл только типа [сантехнического реле](https://ru.aliexpress.com/item/Electric-thermal-actuator-for-manifold-underfloor-heating-system-room-radiant-radiant-valve-calefaccion-suelo-radiante-actuador/32790811518.html?spm=a2g0v.search0104.3.1.H0c4jF&ws_ab_test=searchweb0_0,searchweb201602_3_10152_10065_10151_10068_10344_10342_10343_10340_10341_10307_10301_10060_10155_10154_10056_10055_10054_5370015_10059_10534_10533_10532_100031_10099_10338_10339_10103_10102_10169_10052_10053_10142_10107_10050_10051_10084_10083_10080_10082_10081_10110_5590015_10111_10112_10113_10114_10312_10313_10314_10078_10079_10211_10128_10073_10129_10125,searchweb201603_30,ppcSwitch_5&btsid=dafa91ec-ac97-41b2-9cc4-a5c39f17e56a&algo_expid=8b4967ca-d834-4ce3-9a5f-61313d8ede25-0&algo_pvid=8b4967ca-d834-4ce3-9a5f-61313d8ede25) которое может быть в двух состояниях открыто или закрыто, да ещё с питанием 220
[image]
Отбросил пока китайский рынок, начал искать в рунете...оказывается у нас продаются удовлетворяющие моим требования терморегуляторы. 

Самый подходящий оказался [TECHEM FHKR](https://www.vodokontrol.ru/shop/product_info.php?products_id=30)
[image]
- внешне не страшный
- производитель мне неизвестные, но Германия :)
- живёт от батареек до 3 лет
- высокая точность установки температуры
- сам себе профилактику делает от засора клапана
- управляется по радиоканалу 868,95МГц. **Через RC-шилды для ESP можно по этому радиоканалу работать?**
- программируется на компьютере через [USB-свисток](https://www.vodokontrol.ru/shop/product_info.php?products_id=35)
- Есть для него [термостаты](https://www.vodokontrol.ru/shop/product_info.php?products_id=31) и дистанционные датчики температуры.

Всем хорош этот терморегулятор и цена не кусачая. Остался вопрос можно ли через ESP управлять им и не одним, а по всей квартире. _Моих технических и програмистких навыков для этого недостаточно :(_

Думаю купить это устройство, покрутить его. Но также хочется широты выбора, на одном устройстве не хочется останавливаться...

[Короткое описание](http://www.speppers.ru/11-novostnye-stati/50-umnoe-otoplenie.html) системы этой немецкой фирмы с участием терморегулятора.

alxnovo, 2017-10-08:
Нашёл ещё беспроводной терморегулятор [Dаnfoss living connect ](http://domoton.ru/shop/termoregulyator-danfoss-living-connect/)
[image]
Хоть Danfoss мировой лидер терморегуляторов, даже для него цена за 7к заоблачная, сказываются неофициальные поставки. В [инструкции](http://domoton.ru/uploads/2015/04/instrukciya-danfoss.pdf) к нему написано что работает на той же частоте что и TECHEM FHKR 869МГц по протоколу Z-Wave, можно предположить что и TECHEM FHKR на протоколе Z-Wave и это может быть тупиком для реализации управления через ESP8266, сходу нашёл только проект [Z-Uno](https://geektimes.ru/company/zwave/blog/268980/) (стоит плата 3-4k), но не разбирался с ним

alexandersciv, 2017-10-09:
Решения на Аliexpress мне попадались. Например https://ru.aliexpress.com/item/Heating-Thermostat-Thermal-Electric-Actuator-for-Manifold-in-Under-flooring-Heating-System/32757490866.html?spm=a2g0v.10010108.1000014.9.3251eff0apozJg&traffic_analysisId=recommend_3035_null_null_null&scm=1007.13338.80878.000000000000000&pvid=b7ee5d36-cb15-475f-9b78-8d7b4c0fefcf&tpp=1
или https://ru.aliexpress.com/item/Programmable-Thermostatic-Radiator-Valve-Digital-Energy-Saving-Controller-220V/32822163446.html?spm=a2g0v.10010108.1000013.1.3251eff0apozJg&traffic_analysisId=recommend_2088_1_90158_iswistore&scm=1007.13339.90158.0&pvid=3937dfa4-d5b1-4cb5-bf48-02d0de38873b&tpp=1
Самая основная проблема, на мой взгляд, долго срочно и надёжно кран с питанием от батареек работать не будет. У себя планирую кран (https://ru.aliexpress.com/store/all-wholesale-products/1885035.html?spm=a2g0v.12010608.0.0.2f67bb27Gahk8X) + SONOFF с датчиком температуры + прошивка (когда Сергей допишет :))

alxnovo, 2017-10-09:
@alexandersciv [Этот электронный клапан](https://ru.aliexpress.com/item/Programmable-Thermostatic-Radiator-Valve-Digital-Energy-Saving-Controller-220V/32822163446.html) на Али, который указали единственный, других вариантов для радиаторов когда есть шаговый двигатель, внешне красив и долго работает на автономной питании нет. Стоимость его зашкаливает даже для среднего полёта бренда. 
Клапаны что вы хотите использовать с SONOFF дискретные: либо открыты полностью, либо закрыты. Для моих целей такой клапан недопустим без промежуточных положений. Можно, конечно, сделать трещётку из такого клапана, аналог ШИМ только для пропускания воды, а не тока, но это за гранью комфорта в квартире :)

grossbel, 2017-10-10:
alxnovo, не пишите ерунды, какой ШИМ с трещеткой, термоэлектрический привод не работает как электромагнит щелк - щелк, от закрытого состояния, до открытого проходит примерно 3-5 минут, столько же от открытого, до закрытого. Система отопления очень инерционная по времени и в ней нет необходимости тысячи раз в секунду контролировать и регулировать.

alxnovo, 2017-10-10:
@grossbel, будет вам, перегнул я ассоциацией :) 
Пристыдили меня, пришлось искать инструкции для аналога китайского клапана, нашёл у Danfoss'а тех. описание на [термоэлектрические приводы серии TWA](http://heating.danfoss.com/PCMPDF/datasheet_TWA.pdf), правда, перемещаются около 3 минут, китайские, допускаю и 5 минут будут эту же операцию делать.

Только, всё равно, у этого клапана 2 состояния: закрыт или открыт. Как таким клапаном регулировать температуру радиатора? За 3-5 минут алюминиевый радиатор остынет. Да, помещение так быстро не остынет, но ведь остынет и он опять должен открыться. Если на улице будет очень холодно, то клапан цикл будет повторять каждые 15-30 минут.

Или я не прав? Может этот клапан можно остановить в разных положениях прекратив подачу питания? Тогда зачем у этих клапанов пишут "нормально открытый или нормально закрытый".

Я так понимаю у вас есть опыт эксплуатации таких девайсов, что тогда посоветуете использовать для управления температурой батарей с помощью ESP?

alxnovo, 2017-10-10:
Пока искал описания этих китайских актуаторов, которые я назвал водяным реле, нашёл сначала на алибабе, а потом и сайт производителя нашёл WiFi-терморегулятора [SASWELL SEA800- APP](http://russian.saswell.com/details.php?id=151)
[image]

grossbel, 2017-10-11:
alxnovo, механически клапан для термоэлектрической головы работает следующим образом, шток вверху - поток идет, шток вдавили - поток закрыли. Теперь про голову NC- нормально закрытый, значит при отсутствии напряжения на голове, шток на клапане будет вдавлен, потока не будет. Можно достаточно точно (для отопления) регулировать поток теплоносителя прерывая сигнал для открывания, не дожидаясь пока он полностью откроется. Батарея не остынет за пять минут, а комната тем более. Для информации: комната 64 квадратных метра высота потолков от 280 до 460 см. кубатуру  посчитайте сами, ночью было около 6 грд. за 8 часов температура теплоносителя снизилась с 70 до 45 грд, температура в помещении с 24 упала до 22 градусов, при этом подключено только 3 конвектора (а это просто трубка с пластинами). В Вашем случае лучше брать голову NO, пусть лучше будет жарко если управляющая электроника откажет, а управлять можно чем угодно, к чему душа лежит. Поверьте поиграешься с этим месяц, пока отрегулируешь, потом даже подходить не будешь.

AVL4GITHUB, 2017-10-22:
Какая  ситуация с датчиками температуры и сценариями.
Уже похолодало, я давно готов тестировать на реальном котле у меня в доме.

tretyakovsa, 2017-10-22:
Переписываю весь код, скоро будет.

AVL4GITHUB, 2017-10-22:
Спасибо!!!

Shepty, 2018-04-04:
Вставлю и свои 5 копеек. 
Устанавливаем в каждой комнате (или на каждую батарею) по электроклапану, типа китайского зеленого, как на первом фото. На каждый клапан вешаем по esp8266 с датчиком температуры:
Батарея1, комната 1, темпер1, reley11
Батарея2, комната1, reley12
Батарея3, комната2,темпер2, reley2
Батарея4, комната3, темпер3, reley3
Esp-шки электроклапанов через mqtt получают два значения (желательно с сохранением значений в ПЗУ ): "температура минимальная N", "температура маeксимальная N". И через mqtt отдают два значения: "темпер N", "reley N" (0 или 1). Сценарий например для Батарея1 (электроклапан лучше ставить нормальнооткрытый, чтобы в аварийном режиме батареи нагревались):
Iff темпер1 < температура минимальная1 {reley11=0}
else iff темпер1>температура максимальная1 {reley11=1}.

На котел тоже вешаем esp8266, управляющее контактом включения котла : Котел, топочная, reley0.
По mqtt esp получает и отдает значения: "разрешение работы" ("виртуальная" переменная, позволяющая принудительно отключить отопление, желательно с сохранением в ПЗУ), "reley0".
Сценарий для котла:
Iff reley11==1and reley12==1 and reley2==1 and reley3==1 and разрешение работы==1 {reley0=0} 
else {reley0=0}
(сценарии придется сделать с "масштабированием" пользователем количества задаваемых параметров и добавлением логических операторов между параметрами).
Можно добавить сценарий с отключением отопления при увеличении уличной температуры выше чем самая высокая  "температура минимальная" из набора заданных.
Ещё один момент: после того, как закроются все электроклапаны, проток воды через теплообменник котла прекратится. Поэтому в отопительной системе необходимо оставить одну из батарей (там где точная температура не принципиальна) без электроклапана, прикрутив через нее проток регулировочным краном. 
Система получится автономным модулем для отопления, но в тоже время может управляться внешним сервером через mqtt , например тем же majordomo ( не сочтите рекламой, просто использую как голосовое управление :)).
Электроклапана (термоклапан) имеют большую задержку. У меня не получилось добиться регулирования лучше 0,8 градуса в режиме терморегулятора с петлей гистерезиса. Возможно надо добавить ПИД регулятор с ШИМ на управление электроклапана (думаю с периодом 3-4 секунды). Алгоритм не тяжёлый. Когда-то использовал его на Меге 8.
Как-то так.... :)

ajanulis, 2018-09-24:
@alxnovo kupitj možno na aliexpress.com:

https://www.aliexpress.com/snapshot/0.html?spm=a2g0s.9042647.0.0.77a44c4dbPPANw&orderId=94706870584688&productId=32835324647

No vot kak k nemu dostučatsa po MQTT - poka ne razobralsja.

alxnovo, 2018-11-10:
@ajanulis по твоей ссылки ничего не открывается, можешь другую ссылку указать или написать название по которому можно найти?

ajanulis, 2018-11-11:
> Вот такой нашёл

Da, i ja takoj pakupal. Dva komplekta - odin Gateway byl miortvyj, prodavec
- merzkij, no nadejusj poluchit obratno dengi. Seichas reshil poprobovatj
drugoj variant - na z-wave i sdelatj svjazku s MQTT

alxnovo, 2018-11-11:
Это дорогой вариант, на мой взгляд. Я пытался как-то подключить Techem о котором писал в начале, но безуспешно. Зато выяснил, что донором моего Techem оказался eQ-3 MAX! у которого есть хаб, в свою очередь хаб имеет LAN и WiFi и есть API для управления

shafr, 2018-11-21:
Долго размышлял над подобным сам, пока не натыкнулся на Aliexpress сегодня. При наличии прямых рук, 3D принтера - можно попробовать собрать самому.

А может лучше сделать проект самим с нуля:
* мотор `28BYJ-48` - порядка доллара за штуку. (800гр\см)
* драйвер `L293D` - $.50 за штуку
* ESP-01 или Witty-cloud или еще какой-то - $2
* Блок питания на 9-12 вольт, $3-5

Распечатать накладку (или на подобии) на вентиль радиатора к которому через шестерню будет подключатся мотор. Интегрировать c Home-assistant вообще будет просто по Wifi.

renat2985, 2018-11-21:
@shafr а кто-то уже из ребят которые пользуются нашей прошивкой делал самостоятельно такие краны, покупал моторы, печатал на 3Д корпус. И получилось достаточно неплохо.
Сергей как его звали? Дай ему линк на эту тему, пускай поделится опытом.

a-x-, 2019-01-05:
@shafr, а можете ссылку на блок питания дать? Я только более чем за 1к₽ пока нахожу на чипдипе.
У меня есть ШД на 4кг.см и cnc shield с входом на 12В, примерно 2В и 2А, т.е. видимо 500мА бп с запасом будет норм. 
Только я пока не понимаю какой нужен импульсный или трансформаторный со стабилизированным напряжением

a-x-, 2019-01-05:
@alxnovo, по ссылке на eQ-3 Smart Heizkörperthermostat Bluetooth цена уже сильно выше и нет доставки в Россию. Вы его уже используете? Как подключаете? Используете какой-нибудь bluetooth модуль или esp32?

a-x-, 2019-01-05:
Используя [калькулятор](http://rosfrezer.com/kalkulyator-shd) получил 24В 2.5А
[Такой импульсный бп стоит 1300₽ в chipdip](https://www.chipdip.ru/product/gst60a24-p1j)

a-x-, 2019-01-13:
@shafr, 
> Распечатать накладку (или на подобии) на вентиль радиатора к которому через шестерню будет подключатся мотор

проще снять вентиль (обычно это просто, у меня например даже ключ нужного размера нашёлся)
и через [муфту](http://www.r23d.ru/mufta_soedinitelnaya?search=муфта) соединить оси 

Я почти собрал, вот [тут всё записываю](https://www.notion.so/thismoment/Steppers-Servos-8b545e278fd748ffb778dd43c8f9d657#36a0e34b0cb54d55a4109df80a2062db)

a-x-, 2019-01-13:
Хорошо бы для таких issues сделать специальный лейбл, наприме [project]

renat2985, 2019-01-13:
@a-x- сделал.

a-x-, 2019-02-02:
@shafr,
Шаговик на 4.5КГ*см, который у меня был конечно не справился с вентилем, надо какую-нибудь мощную серву, но не знаю как определить подходящий крутящий момент. >= 10КГ*см думаю надо

Gio0566, 2019-08-01:
1 797,37 руб.  41%СКИДКА | Wifi технологичный клапан система домашней автоматизации клапан управления для газа или воды Голосовое управление работа с Alexa Echo Google Home
https://s.click.aliexpress.com/e/cOTORU9c привод для шаровых кранов 
Пробовал использовать электрические термоголовки( зелёные)  с ШИМ и ПИД регулированием - получилась точность около 2 градуса и циклическое нагревание и остывание радиатора.  без переделок эти головки даже при ШИМ регулировании фактически работают в режиме оn/off только с большим периодом, использовал головку 220 в и твердотельное реле.

shafr, 2020-01-13:
А у меня не получилось нормально использовать `Eq-3 BLUETOOTH® Smart Radiator Thermostat`. Заказал 2 штуки с Amazon.de, проблем с блютусом или интеграцие с умным домом нету. Есть 2 но:
* В термоголовке нету встроенного термометра, тоесть он работает в слепую. При условии что у нас в доме котельная работает по-разному и температура батареи может отличатся - не понятно что делать. Ну ладно если бы не П2:
* Регулировка не работает вообще. Ни в автоматическом ни в ручном режиме. Ни в ручную ни через Bluetooth приложиние, ни с Home Assistant. Посли инициализации и того как погудит мотор на протяжении пару минут, он больше не работает кроме как в режиме `Boost`. Тоесть при выставлении температуры вообще ничего не происходит...

a-x-, 2020-05-25:
а тут есть список других термоголовок
https://v2.sprut.ai/catalog?tags=Термоголовки

многие не дорогие, но я ещё не пробовал

у меня есть [свой список с комментами](https://www.notion.so/thsm/Steppers-Servos-8b545e278fd748ffb778dd43c8f9d657#1a3459e4dc6f4c159cd08122371fd706)

а если колхозить, я так и не понял как крутящий момент подобрать и вообще какую модель сервы надо.

a-x-, 2020-05-25:
@shafr, eq-3 и eq-3 max, это разные термоголовки?

shafr, 2020-05-26:
Не знаю даже, я брал самую дешевую - тем более что доставка была бесплатно. Надо будет что-то опять придумывать ближе к осени - наверное возьму просто 2 головки любого производителя, лишь бы работали с моими батареями.

Detsle, 2025-04-25:
Добрый день, в квартиру заведён коллектор отопления на 3 радиатора, и 3 радиатора соответственно, которые установлены в нишу и закрыты защитными экранами. Не хотелось бы каждый раз открывать экраны, чтобы сделать теплее или холоднее.
_Подскажите, как можно удалённо через телефон, регулировать температуру, либо радиаторов, либо коллекторов?_
**НО**
**1)** Нужно, чтобы прибор был на постоянном токе, без каких либо батареек.
**2)** Чтобы регулировать можно было не на вкл/выкл, а немного более детально.
**3)** Чтобы не было внешнего терморегулятора.