tretyakovsa/Sonoff_WiFi_switch #43: Сценарии

renat2985, 2017-10-11:
Сделали сценарии, прежде чем обновляться посмотрите видео https://youtu.be/E_llovvoZe4

renat2985, 2017-10-11:
Просьба писать все ошибки сюда.

tretyakovsa, 2017-10-11:
Сделали сценарии, прежде чем обновляться посмотрите видео https://youtu.be/E_llovvoZe4

alexandersciv, 2017-10-12:
Как уже сообщал, сценарии на включение отключение света в цепочке из 3 устройств Sonoff работают хорошо, проблем не обнаружил. Но мне сейчас нужно выполнить 3 задачи на этих устройствах.
1. Управлять одним устройством Sonoff с показаний температуры другого устройства Sonoff .
2. Включать свет по датчику движения.
3. Включать свет по датчику освещенности (день-ночь)
Нашёл решение включать устройство Sonoff обычным выключателем (в теме SONOFF доработка и применение) как его реализовать в Вашей прошивке?
Когда ожидать выполнения сценариев?

renat2985, 2017-10-12:
@alexandersciv, а что нам будет за это :) мы дорогие разработчики если делать на заказ ;)

alexandersciv, 2017-10-12:
**@renat2985** присутствие коммерческой составляющей в любом проекте для меня это нормально. Но здесь должна быть определённость. К примеру https://wifi-iot.com/p/esp8266/ продвинутая, стабильная прошивка, с решением практически любых желаний по работе с ESP8266 имеет как свободные, так и платные функции. 100 рублей и модуль универсальный. Есть еще несколько примеров. 
Поэтому Вам решать, что будет платным или бесплатным.

vobis77, 2017-10-12:
Очень плохо работает RF приёмник: выход включается и выключается как хочет, мигает во время нажатия кнопки на пульте.

tretyakovsa, 2017-10-12:
@vobis77 Напишите сюда строку сценария связанную с этой ошибкой.

vobis77, 2017-10-13:
@tretyakovsa
Если Received = 15 тогда ESP8266-Test relaynot

vobis77, 2017-10-13:
Нашёл решение проблемы:
Если включению и включению присваивать отдельные кнопки то всё работает нормально.
 С одной кнопкой на ПЕРЕКЛЮЧЕНИЕ чёрт знает что творится.

alexandersciv, 2017-10-16:
Большая просьба к разработчикам поправить ==Обновился до крайней версии 2017.10.11. Перестали сохраняться настройки = Задать статический IP=

alexandersciv, 2017-10-16:
Прошил прошивкой 2017.10.16. настройки = **Задать статический IP**= не сохраняются, BLINK сохраняется.
Сохраняются только на прошивке 2017.10.08.

lyovav, 2017-10-20:
Будут ли добавлены  в сценарии операции с  данными датчиков на устройстве?

renat2985, 2017-10-20:
@lyovav, да. Скоро сделаем.

Paly4, 2017-10-20:
Подключил сегодня  ИК приемник , работает хорошо , это отлично расширяет функциональность , молодцы !
смотрел видео на ютубе , там вы говорите что пока не работает режим Переключить , не знаю почему , но у меня работает он и вроде бы нормально

renat2985, 2017-10-20:
@Paly4 мы уже и ИК передатчик сделали. Скоро выложим новую версию. :) По поводу режима переключить, я оговорился. Он не работает только в том случае если у вас включен модуль RF приемник и IR приемник. Если вы включаете только 1 из них, режим переключения работает.

Paly4, 2017-10-20:
не , нашел все таки что не работает при режиме Переключить - с пульта работает нормально , одной кнопкой вкл/выкл , а вот управление через  вэб тогда не работает правильно , проверял на RGB 
например захожу в управление , у меня диод горит , жму на красный цвет , он выключается , жму на другой цвет - включается , кнопка вкл/выкл не работает от слова совсем, удаляю сценарий с переключением , остается 2 сценария на включение и выключение разными кнопками - тогда в вэбе работает вроде бы как и положено

Paly4, 2017-10-20:
еще заметил , удалил все сценарии , настроил с пульта только одну кнопку на включение
RGB включено (горит) , стоит нажать в это время запрограммированную на включение кнопку на пульте и в вэб опять кнопка не работает , что бы заработала , нужно еще раз нажать кнопку на пульте

renat2985, 2017-10-21:
@Paly4 спасибо! Исправим.

massed24, 2017-12-10:
Всем привет,
Имею два устройства Jalousie,- Ролета3 и Ролета4. Заметил проблемку в сценариях.
Сценарий Ролета4:
Если stateJalousie =1 Тогда Ролета3 jalousieopen
Если stateJalousie =0 Тогда Ролета3 jalousieclose
Так вот, когда нажимаю Ролета4 открыть, Ролета3 тоже открывается, все отлично, а вот если нажимаю Ролета4 закрыть, Ролета3 не закрывается. Это у всех так, или только у меня?
Версия прошивки 2017.10.16

massed24, 2017-12-14:
Откатился на версию 2017.10.11, попробовал, проблема осталась, но потом обновился до последней 2017.10.16 и все стало на свои места, сценарий работают как надо. Не понял конечно, ну и ладно, главное работает.

augusev, 2019-01-20:
Две платы wemos. Сделан сценарий, при включении реле по таймеру на первой плате, мигает светодиод на второй плате. При выключении реле на первой плате, светодиод на второй плате гаснет. Таймер на первой плате включает реле на 1 минуту, затем выключает на 1 минуту. В принципе все работает, но сигнал на включение светодиода на второй плате проходит не всегда. Т. е. реле на первой плате включилось, а светодиод на второй плате не включился. Иногда наоборот, реле на первой плате отпустило, а светодиод на второй плате продолжает моргать. Все собрано на столе в метре от WI-FI-роутера, так,что на уровень сигнала грешить нельзя. Кто виноват? Что делать?

augusev, 2019-01-20:
Даже когда вручную нажимаю кнопку включить/выключить реле на первой плате, светодиод на второй плате включается/выключается не всегда.