tretyakovsa/Sonoff_WiFi_switch #44: Дополнительное реле

AleksandrGromov, 2017-10-12:
Есть ли в планах добавление дополнительных реле с полным функционалом, таймеры, сценарии
Было бы удобно назначать определенному GPIO функцию управление реле.

Использую прошивку на wemos d1 r2 для управления освещением аквариума (используется одно реле) по таймеру, плюс RGB лента для ночной подсветки. Хотелось бы для второго реле добавить по условию, к примеру если температура воды в аквариуме по датчику DS18S20 выше +28 С° включить через реле обдув воды, а при понижении температуры до +25 С° выключить реле.
Сейчас приходится дёргать блок питания на вентилятор вручную.

renat2985, 2017-10-12:
Скоро добавим поддержку нескольких реле на 1 устройстве. 
Сценариев связанных с температурой пока нет. Но скоро тоже сделаем.

AleksandrGromov, 2017-10-13:
Спасибо за вашу работу!
Буду ждать обновлений в прошивке.

tretyakovsa, 2017-10-13:
@AleksandrGromov 
> Хотелось бы для второго реле добавить по условию, к примеру если температура воды в аквариуме по датчику DS18S20 выше +28 С° включить через реле обдув воды, а при понижении температуры до +25 С° выключить реле.
> Сейчас приходится дёргать блок питания на вентилятор вручную.
обдув воды это что? Это вентилятор стоящий у аквариума или это аэратор опущенный в воду? 
Или может быть правильно выключить нагрев воды и включить аэратор.

AleksandrGromov, 2017-10-13:
Обдув реализован обычным компьютерным кулером на 12v
Подключен к блоку питания 5v из за этого меньше обороты и его практически не слышно.
Кулер направлен на поверхность воды и установлен в крышке аквариума в которой стоят лампы освещения, они и подогревают воду. При включении обдува температура воды заметно падает  с 30 до 22 градусов. Это единственный способ охлаждения воды который я мог придумать и он работает. Обогреватель выключен, температура воды и без него 30 градусов.
Так вот обитателям моего аквариума комфортная температура воды 24-25 градусов.
Приходится вручную регулировать температуру воды включением вентилятора

TimKaSANOFF, 2017-10-17:
Поддерживаю на таких платах как nodemcu вообще можно до 8 реле подключить,так что жду появления поддержки нескольких реле(поставил новую баню(сруб)...из которой летом буду делать-Умную баньку...типа включение света при подходе к бане, включение выключение электро котла, управление электро-клапаном подачи воды+датчик уровня воды в резервуаре(баке)...типа нажал кнопку пошел набор воды как только замкнулся датчик уровня воды(датчик дождя подача пекратилась и.тд.). Так что без потдержки нескольких реле никак:)