tretyakovsa/Sonoff_WiFi_switch #45: Не работает режим работы реле по таймеру после перегрузки модуля ESP

lyovav, 2017-10-16:
При перегрузке модуля не соблюдаются условия режимов состояния ВКЛ ВЫКЛ.
К примеру включить по таймеру с 7 утра и выключить в 7 вечера и если в этот промежуток времени перегрузить модуль то соблюдается только условие ВКЛ ВЫКЛ при старте модуля ESP а условия таймера игнорируются.

tretyakovsa, 2017-10-16:
Я так понимаю вы хотите, чтоб после перезагрузки система вернулась к состоянию до перезагрузки.
И таймеры тут ни пи чем. Я правильно понял?

lyovav, 2017-10-16:
Ну как бы есть конфликт с опцией ВКЛ ВЫКЛ при старте (рестарте) модуля ESP и самим условием таймера. Хотелось бы оставить все как и есть но в одним условием когда таймер активен это условие при старте (рестарте) модуля ESP не соблюдалось а соблюдались условия таймера.

Теперь поясню на пальцах)... Включен таймер с 7 утра ВКЛ  а с  7 вечера ВЫКЛ и что б это условие соблюдалось и при перегрузке модуля ESP. Тоесть  если между с 7 утра ВКЛ и до  7 вечера был рестарт то рестарт не должен влиять на условие таймера. А если условия таймера нет в задании то активно только условие ВКЛ ВЫКЛ при старте модуля ESP.

tretyakovsa, 2017-10-16:
Причем здесь таймер? Вот у вас есть состояние устройства, что то включено, что то выключено.
Модуль перезагрузился и вернулся в то состояние в котором был до перезагрузки. Так?

lyovav, 2017-10-16:
Хорошо)))  Включен таймер с такими параметрами

relay | вкл | Всегда | 07:00:00 |  
-- | -- | -- | -- | --
relay | выкл | Всегда | 19:00:00

Оно идеально выполняется при одном условии если на модуль с той или иной причиной не перегрузился.

Перегружаю модуль ESP  и срабатывает настройка при включение ESP

> Первая ячейка - PIN, вторая - состояние реле при включение ESP

и ей абсолютно пофиг  какое там было состояние таймера

Если уже у меня активный таймер то его состояние должно сохранятся при перегрузки модуля ESP и - опция состояние реле при включение ESP должна игнорироваться.

lyovav, 2017-10-16:
Одним словом на состояние реле не должно влиять перегрузка и опция в настройках 

> -

 состояние реле при включение ESP

renat2985, 2017-10-16:
Состояние реле не сохраняется в файл после его смены. Поэтом после рестарта, ESP не знает реле у вас было включено или выключено.
Сохранять состояние реле в файл тоже не стоит. Т.к. таким образом вы достаточно быстро затрете свою flash память на ESP.
Поэтому пока нет решения вашей задачи.

lyovav, 2017-10-16:
Ну как бы можно состояние реле записывать в EEPROM ESP при включении в ручную и средствами заданий таймера... Переделывал скетч,взятый на просторах Интернета, для себя под wifi розетку средством Arduino на основе модуля ESP вполне работоспособно.

tretyakovsa, 2017-10-16:
@lyovav 
EEPROM ESP это вы что имеете ввиду? Flash?

lyovav, 2017-10-16:
Да Flash.
Я не силен в программировании и их терминологии и делаю в основном интуитивно на основе уже готового скетча (кода)
В среде разработки Ардуино есть 

> библиотека  #include <EEPROM.h>
с ее помощью реализована запись всех данных настроек в том числе и состояние реле.

renat2985, 2017-10-16:
@lyovav я вам ответил ранее, сохранять в Flash не целесообразно. Т.к. спустя пол года, максимум год вы затрете свою ESP и вам придется ее выкинуть.

lyovav, 2017-10-16:
@renat2985 Что Вы имеете в виду "затрете свою ESP"  для хранения настоек используется определенная область во Flash куда записываются данные. Каждый день записывать туда данные никто не собирается, если идется о ограниченных циклах записи-чтения самой флеш памяти...
А для работы хватит раз настроить и пусть себе работает.
Не понимаю Вас, если с помощью  средства разработки  Arduino это уже реализовано и работает (разработан скетч, залит в ESP, работает нареканий нет ) почему тут то как то так?

tretyakovsa, 2017-10-16:
В общем, мы совсем не понимаем друг друга.

lyovav, 2017-10-16:
По даташиту сама флешка, котраявпаяна в esp8266 выдерживает 

> – More than 100,000 erase/program cycles 
> – More than 20-year data retention

grossbel, 2017-10-16:
С одной стороны Ренат прав, разработчики всегда стараются как можно реже писать в память, для продления жизни устройства, наблюдая за промышленными разработками, я часто наблюдал когда EEPROM умирал и в бытовой технике есть такая проблема, но эта проблема вылазит через 10 - 15 лет эксплуатации.
Теперь давайте произведем подсчет, по даташиту, запись/перезапись 100 000  циклов, в данном модуле достаточно 4 выходов, каждый из них изменит свое состояние 10 раз в день, получаем 4х10х365дней= 14 600 в год, итого устройство проживет 6,85 года

grossbel, 2017-10-16:
Микрочип вообще 100 лет хранения обещает, а не 20

grossbel, 2017-10-16:
А теперь изменим математику по полной, вымышленной нагрузки 8 выходов, 30 раз в день получаем 87600 циклов за год, итого примерно модуль сдохнет по истечению года и трех месяцев.

grossbel, 2017-10-16:
Поэтому резюме такое: если выход важный необходима галочка рядом с ним - сохранять текущее состояние при перезагрузке.

lyovav, 2017-10-16:
@grossbel Согласен, а то ставить резервное питание на ЕСП както не комильфо.

grossbel, 2017-10-16:
Сергей, ну что тут не понятного, представьте ситуацию, у Вас включен насос, дернули питание в деревне, включили через некоторое время, ESP загрузилась а Ваш насос не работает, Вы приезжаете вечером домой, а там холодно. Мое мнение такое - необходимо писать состояние выходов в память после каждого изменения его состояния, для продления ресурса я уже предложил, важность выбирает пользователь

alexandersciv, 2017-10-16:
Соглашусь с **@grossbel**  - запись в память очень важная функция. В профильной теме на 4pda даже предлагали подключить батарейку для сохранения настроек при выключении питания.

tretyakovsa, 2017-10-16:
@grossbel   Я спросил: "Нужно вернутся к состоянию в котором модуль был до перезагрузки?" И услышал, что то о таймерах. Этого я не понял.

tretyakovsa, 2017-10-16:
@alexandersciv Действительно при некоторых задачах стоит ставить батарею. Батарея будет сохранять данные в  RTC и по включению нормального питания все приводить в норму.
Но для этого нужно правильно поставить задачу. И понятно, что это не стоит  делать на готовых устройствах без этого режима. Таких как Sonoff

grossbel, 2017-10-16:
Да, кстати про RTC, Сергей как часто модуль часов сверяется с сетью и корректируется?

tretyakovsa, 2017-10-16:
@grossbel Я не разбирался с этим вопросом это должно быть определено в библиотеке. В первых вариантах кода, я сам запускал этот процесс один раз в сутки. Затем выбросил этот код, уже не помню но увидел что библиотека делает это сама. Если замечаете, что часы уходят, пишите.

grossbel, 2017-10-16:
Спасибо, понял. Думаю что для домашнего хозяйства разбежка в пару минут в сутки не критична.

v222008, 2017-10-19:
Думаю оптимальным решением будет использование mqtt сервера  еще и в роли устройства памяти состояния реле
Тут (https://www.youtube.com/watch?v=J38SOv_to88) парень подробно описал как он ето реализовал

alexandersciv, 2017-10-19:
**@v222008** Использование mqtt сервера также как Domoticz и другие, очень хороший вариант для подготовленного специалиста, мне так и не удалось, как простому пользователю,самому настроить или найти пошаговую инструкцию. Было бы хорошо увидеть Ваше решение.

Paly4, 2017-10-19:
> Использование mqtt сервера также как Domoticz и другие, очень хороший вариант для подготовленного специалиста,

это 100% , нужно очень не мало прочитать и изучить ...

grossbel, 2017-10-19:
Друзья, получаю я сегодня посылочку от Китай тружеников с недельными таймерами для дин рейки, вскрываю коробку и вижу, таймер - то работает, по паспорту батарея на 15 дней, но шли они больше, поэтому если цепь критичная и надо чтоб она была в состоянии вкл. , то можно и батарейку вкорячить, но это уже усложняет цепи питания и увеличивает в размерах модуль.