tretyakovsa/Sonoff_WiFi_switch #47: Работа над ошибками в обновлении прошивок.

alexandersciv, 2017-10-16:
У меня сейчас на обкатке 3 модуля. Один прошит 2017.10.08 - у него сохраняется установка  **=Задать статический IP=**
Второй прошивал утром прошивкой 2017.10.16, у него не сохраняется установка  **=Задать статический IP=** и нет треугольника обновления прошивки (хотя бинарники поменялись на новые - как принудительно прошивать если нет треугольника?)
Третий я прошил с прошивки 2017.10.11 (у него треугольник был) но **=Задать статический IP=** не сохраняется.

tretyakovsa, 2017-10-16:
Я все проверю. Странно, что сама галочка сохраняется, а вот адрес модуль получает все время динамический это я поправлю. А вот чтоб, галочка не сохранялось не могу получить такого эффекта.

alexandersciv, 2017-10-16:
**@tretyakovsa** У меня и на утренней и на вечерней прошивках убранная галочка сохраняется (остаётся отключенной).

renat2985, 2017-10-16:
Обновляемся, пробуем. Только что выложил новые биранкики. 
[image]

alexandersciv, 2017-10-16:
Обновился. =Задать статический IP= не сохраняется.

renat2985, 2017-10-16:
@alexandersciv значит у вас проблемы с файловой системой. Попробуйте подключить ESP к компьютеру и обновится через nodemcu flasher https://github.com/tretyakovsa/Sonoff_WiFi_switch/releases/tag/2.2

grossbel, 2017-10-16:
Все сохраняется, проверил 3 раза. 
Теперь вылезла проблемка, модуля  не видно в закладке Windows - Сеть, а раньше он был :)

alexandersciv, 2017-10-16:
Вроде решил вопрос. Из меню прошил модули прошивкой 2017.09.20. Установил статический IP. Обновил по треугольнику. Пока работает.

vobis77, 2017-10-17:
У меня показания датчиков на главной странице показываются два раза если они находятся на другом  модуле:

vobis77, 2017-10-17:
[image]

renat2985, 2017-10-17:
@vobis77 такое бывает когда какое-то из ваших устройств не обновлено до последней версии. Проверьте. Если все ОК, попробуйте просто перезагрузите устройство.

vobis77, 2017-10-17:
Все устройства обновлены, перезагрузил, пока вроде все нормально. Спасибо

vobis77, 2017-10-17:
@renat2985 
Пару часов спустя та же проблема только теперь датчики показываются втройне. На других устройствах которые я перестартовал все нормально.
[image]

renat2985, 2017-10-17:
@vobis77, то есть вы сделали рестарт как я вам посоветовал первый раз, после этого все работало нормально. Потом спустя 2 часа снова появились на том же устройстве дубликаты? Вы что-то настраивали в промежутке этих двух часов на каком либо из этих устройств. Мне важно понять после чего такое происходит.

vobis77, 2017-10-17:
@renat2985 
Да всё именно так как вы описали выше. С первый раз делал рестарт всех устройств. Много раз заходил в настройки  разработчика но ничего не менял. Имя одного устройства изменил.

renat2985, 2017-10-17:
@vobis77 если заметите из за чего конкретно это происходит пишите, исправим.

sultan5448, 2017-10-20:
при использовании темы block показания датчика не отображает в режиме реального времени

renat2985, 2017-10-20:
@sultan5448, показания датчиков обновляются в режиме реального времени только в Life и Default.

AVL4GITHUB, 2017-10-24:
Всем привет!
После этой прошивки http://192.168.0.204/upgrade?spiffs=http://backup.privet.lv/esp/sonoff/spiffs.0xBB000_flash_size_1Mb.256Kb_2017.10.16.bin&build=http://backup.privet.lv/esp/sonoff/build.0x00000_flash_size_1Mb.256Kb_2017.10.16.bin
пропала возможность конфигурации пина релэ.
Возможность конфигурации пина МИГАНИЯ ПРИ КОНЕКТЕ К РОУТЕРУ осталась.
И, после апдейта "слетела" настройка собственно этого пина реле. 
Перепошивался 5 раз, ситуация не изменилась.
Что делать, или, откорректируйте прощивку, пожалуйста.
Спасибо.

renat2985, 2017-10-24:
@AVL4GITHUB исправим. Если в новой версии проблема останется. Напомните пожалуйста еще раз.