tretyakovsa/Sonoff_WiFi_switch #49: Добавитьт использование своего IP NTP-сервера

lyovav, 2017-10-18:
Во избежании лишнего Интернет трафика было б удобно использовать локальный  NTP-сервер к примеру свой роутер. (не у всех есть анлимитный Интернет по мобильной связи)

tretyakovsa, 2017-10-18:
Нет в большинстве роутеров NTP- сервера.

lyovav, 2017-10-18:
Поверьте, по специфике своей работе знаю что есть.

alexandersciv, 2017-10-18:
**@lyovav** Тогда поясните как его настраивать.

lyovav, 2017-10-18:
Любое железо на прошивке Openwrt, роутеры MikroTik но суть не где настроить а опционально добавить в устройстве

renat2985, 2017-10-18:
> Поверьте, по специфике своей работе знаю что есть.
@lyovav, сочувствую вашей квалификации. Практически в любом бюджетном роутере нет NTP сервера.

lyovav, 2017-10-18:
Ну дякую, дякую не думав що перейдемо на особисті образи. Бачу у Вас тут найперше прийнято обісрати людину з ніг по голову аж потім, можливо, відповісти по суті...

renat2985, 2017-10-18:
@lyovav, во первых если вы уважаете тех кто здесь пишет, в чем я уже сомневаюсь, и хотите что бы вам помоги пишите на Русском. Я ведь с вами не общаюсь на своем родом Латышском.
Вас никто не опускал. Скорее вы первый намекнули что Сергей некомпетентен в этих вопросах. На что вам еще раз ответили что далеко не в каждом роутере есть NTP сервер.

Krendelyok, 2017-10-19:
А что, если прикрутить в прошивке поддержку RTC-модуля? Поднять NTP-сервер? Или каким-то другим способом передавать значения времени остальным модулям с модуля, где установлен RTC с батарейкой. Тогда система не упадет даже при отключении Интернета.

tretyakovsa, 2017-10-19:
Мы можем дать возможность настройки адреса NTP сервера. Но как включить NTP сервер в сети мы точно рассказывать не будем. Поддержку RTC-модуля мы тоже делать не будем. Если количество тех кто хочет пользоваться прошивкой без доступа в интернет будет значительное количество, а не один два человека, тогда можно не совсем точное время выдернуть из любого роутера.