tretyakovsa/Sonoff_WiFi_switch #51: Глюк с сценариями при смене имени устройства на котором они  установлены

lyovav, 2017-10-18:
Настроил ИК пульт на включение и выключение реле на 2х устройствах.
1 Устройство которое включает и  выключает и в котором есть ИК приемник
2 Устройство которое включает и  выключает свет.

После смены имени у 1 устройства перестает работать его реле по ИК пульту но управляет ИК пультом устройством 2.

tretyakovsa, 2017-10-18:
Да это есть. И эту задачу, я ещё не решал.

massed24, 2017-12-15:
В моих устройствах jalousie, после добавления сценариев, заметил глюки, которые могут повредить механизмы моих рулет, закрытого типа. Заключается это в следующем. Если на устройстве добавить сценарий, то оно начинает работать заторможено, а именно, после срабатывания датчика оборотов или как в моем случае герконового концевика, мотор продолжает некоторое время(1-2с) вращаться, хотя контакты концевика, уже замкнуты. Чем больше сценариев я добавляю, тем больше время задержки, в итоге конец рулеты поднимается выше заданных пределов и упирается в корпус. Без сценариев мотор после срабатыванию датчика останавливается резко, рулета останавливается строго в заданных пределах.
Просьба создателей проекта, или других пользователей, протестировать работу своих рулет со сценариями и без них. Заметили ли вы, продолжение работу мотора, после срабатывания датчика, при добавлении сценариев. 
Если у вас все четко, буду искать причину у себя, если есть какие советы, буду вам благодарен.

renat2985, 2017-12-15:
ESP8266 07, к устройству подключены жалюзи (кнопка, сенсор оборотов), Relay, PIR сенсор и сенсор освещения. Задействованы 4 сценария:
1. Если жалюзи = 0, опускаем в другой комнате другое жалюзи.
2. Если жалюзи = 1, поднимаем в другой комнате.
2. Если Relay = 1 включается по PIR сенсору, включаем другое реле на другом устройстве.
4. Если Relay = 0 выключаем подсветку на другом ESP устройстве.
Глюков не замечено, PIR сенсор отрабатывает сразу же, сценарии срабатывают с задержкой менее 1сек. Останавливается жалюзи в одном и том же месте.

Схема: https://raw.githubusercontent.com/tretyakovsa/jaluzi/master/tutorial/jalousie.png
Видео: https://www.youtube.com/watch?v=I2lZExMvrtA&list=PL6NJTNxbvy-IPTDQk8XjTV41oRrFafrRi&index=16

massed24, 2017-12-15:
А вы можете, специально для меня, удалить сценарии и сравнить остановятся ли жалюзи в том же месте или хоть немного, проедут, я бы знал, что мне делать дальше. 
В жалюзях такого типа как вас, не критично, если они проедут дальше на 1см, а у меня за секунду они проедут 4см.