tretyakovsa/Sonoff_WiFi_switch #52: Вопросы по RGB

Paly4, 2017-10-18:
не нашел подходящей темы , открыл новую
есть несколько вопросов , может предложение по функционалу
1 - возможно ли сделать что бы при включении в сеть включался выбранный режим  , цвет свечения диодов?
2 - можно ли сделать что бы при включении статического режима любого цвета , лента загоралась не полностью , а постепенно , диод за диодом пока не засветится полностью  , кажется это было бы красиво и для глаз более приятно , не сразу врубается на полную, а постепенно

renat2985, 2017-10-18:
1. То есть ESP было по какой-то причине без питания. Вы подключили снова питание и нажав кнопочку "Включить" хотите видеть прошлый режим, цвет, яркость? К сожалению эти настройки не сохраняются. И после рестарта сбрасываются на дефолтные. 
Возможно вам подойдет следующее решение. Когда вы выбираете нужную вам динамику, цвет, на странице создается URL с этими параметрами. Ее можно скопировать, сделать ярлык и в будущем заходя по этой ссылке у вас будет включатся тот настроенный вами режим.
Но в этом случае поставьте галочку на "Задать статический IP", т.к. IP может поменяться со временем. 
[image]

2. Мысль интересная, но пока это сделать нельзя. Если у вас есть опыт в программирование вы можете попробовать добавить этот режим. Мы используем эту библиотеку для динамики:
https://github.com/renat2985/WS2812FX
Если вы сделаете я добавлю. :)

Paly4, 2017-10-20:
к сожалению опыта и знаний в программировании нет  ...
есть еще предложение , если конечно возможно  , добавить в RGB микрофон и режим светомузыки

Gostin0, 2017-11-18:
Привет, а есть возможность что появится прошивка для обычной RGB ленты не адресноию. Буду благодарен за прошивку если таковая имеется. модуль esp8266 01