tretyakovsa/Sonoff_WiFi_switch #53: устройства и vpn

sultan5448, 2017-10-20:
имею два устройства sonoff. находятся в разных концах города но в одной vpn сети. имеют адреса 192.168.2.138 и 192.168.4.215.возможно ли чтоб оба устройства были на одной странице и между ними выполнялись сценарии? и второй вопрос эти устройства имеют порт помимо адреса? спасибо

tretyakovsa, 2017-10-20:
Возможно или нет  не знаю. Если SSDP протокол работает при такой конфигурации сети, то может и работает.

renat2985, 2017-10-20:
@sultan5448, на ваш второй вопрос, каждое устройство запускает http сервер. Который работает на 80 порте.
Очень интересно получится ли у вас связать 2 устройства в VPN сети. Вы уже пробовали это сделать?

sultan5448, 2017-10-20:
оба устройства в сети. но не связаны. каждое доступно по своему адресу. но на одной странице оба не отображаются.

renat2985, 2017-10-20:
Можете попробовать создать у себя на компьютере файл: ssdp.list.json
С таким содержимым:
`{"Здесь имя устройства с страницы конфигурация":"192.168.2.138","Здесь имя устройства с страницы конфигурация":"192.168.4.215"}`

Далее зайдите на каждом устройстве в http://ip/edit и закачайте этот файлик.

sultan5448, 2017-10-20:
буду дома попробую. но не уверен что справлюсь. надо создать текстовый документ с расширением json? {"Sonoff":"192.168.2.138","Sonoff1":"192.168.4.215"}
правильно?

renat2985, 2017-10-20:
Если у вас имена Sonoff и Sonoff1 на странице конфигурации, то да, правильно.
Я для вас сделал файлик ssdp.list.json распакуйте и попробуйте закачать.
[ssdp.list.json.zip](https://github.com/tretyakovsa/Sonoff_WiFi_switch/files/1401147/ssdp.list.json.zip)

sultan5448, 2017-10-20:
спасибо. если можно поправьте его. я напутал ip адреса. 192.168.4.193 и 192.168.2.138

renat2985, 2017-10-20:
`{"Sonoff":"192.168.4.193","Sonoff1":"192.168.2.138"}`
[ssdp.list.json.zip](https://github.com/tretyakovsa/Sonoff_WiFi_switch/files/1401222/ssdp.list.json.zip)

sultan5448, 2017-10-20:
с андроид загрузить файл на устройства не получилось. захожу по адресу..... /edit  выбираю файл .....json нажимаю upload.потом смотрю  файл  в левой колонке но он не изменился. надо с компа дома

sultan5448, 2017-10-20:
к сожалению не получается и на компе тоже. скачал файл который вы прислали. указываю путь нажимаю upload. левая колонка на странице перезагружаетя но сам файл ssdp. list. json не меняется. пробывал и правая кнопка мыши потом edit и руками прописывать. сохраняется адрес только одного устройства.

renat2985, 2017-10-20:
Надеюсь вы распаковали zip архив? Попробуйте зайти http://ip/ssdp.list.json

sultan5448, 2017-10-20:
конечно распаковал

sultan5448, 2017-10-20:
вижу ('sonoff':'192.168.2.138') скобки и кавычки другие

renat2985, 2017-10-20:
Попробуйте с другого браузера закачать файл.

sultan5448, 2017-10-20:
ie и chrome. отрицательно. Подключил оба устройства в одной сети роутера. Оба устройства Видят друг друга. в файле.json оба устройства прописаны. но тоже не меняются. Вы у себя попробуйте поменять

sultan5448, 2017-10-20:
если этим путем невозможно,можно но ли путем например жёсткой прописки этих адресов в. bin файле до прошивки.

renat2985, 2017-10-20:
В таком случае вам лучше разобраться с VPN, выяснить почему там не работает SSDP.

sultan5448, 2017-10-20:
верно. судя по прочитанному в интернете это возможно.