tretyakovsa/Sonoff_WiFi_switch #54: Общие вопросы

Paly4, 2017-10-20:
не увидел , или плохо смотрел как подключить датчик освещенности и какой он должен быть ?

еще вопрос-предложение , если собирать проект на модуле esp8266 , там много свободных gpio , в отличии от Sonoff и соответственно на один модуль можно например повесить RGB ленту и 
 светильник через реле и еще добавить например жалюзи
[image]

и все эти 3 точки сейчас в интерфейсе имеют общее название , в моем случае RGB , что совсем не информативно , а если в доме модулей на ESP много , то еще и забудешь что за что отвечает
возможно ли сделать что бы каждому присваивать свое название ?
[image]

renat2985, 2017-10-20:
Фото резистор я подключаю следующим образом. Через 1кОм.
[image]

По поводу вашего второго вопроса, пока нельзя. Я их называю так: "TV-RGB-and-Relay", "Zaluzi-i-poliv".
Но мы думаем об этом. Возможно в будущем что-то и придумаем.

Paly4, 2017-10-20:
еще предложение , сделать по умолчанию в RGB цвет по умолчанию более нейтральный , например **ffecc0** , это второй квадрат в желтой гамме , сейчас, имхо, довольно желтый по умолчанию

> Я их называю так: "TV-RGB-and-Relay"
названия на кирилице будут влиять ? или лучше латиница ?

renat2985, 2017-10-21:
@Paly4 можно кириллицей но до 8 символов, латиницей до 15. В новой версии можно будет писать имя только латиницей.
По поводу цвета, мне больше нравится желтый :)

alxnovo, 2017-10-22:
Также **нельзя одинаково называть модули**, что для меня было не столь однозначно и я месяц ломал голову почему у меня модули друг-друга редко находят или когда набираешь адрес одного модуля показывается страница другого.
Сегодня после очередного зависания ESP (пришлось откатится на старую прошивку, чтобы отвис), заново делал настройки и присвоил временное название модулю и на главной странице корректно стали отображаться все модули.
Называть по разному одно и то же устройство в разных комнатах неудобно для использования, поэтому пока выкручиваюсь тем, что называю модули буквами из разных языков визуально идентичными русским буквам.

Не будет работать:
```
Имя модуля-1: Ролета; Пространство: Кухня
Имя модуля-2: Ролета; Пространство: Спальня
```

Работает:
```
Имя модуля-1: Ролета; Пространство: Кухня (первая буква "P" английская)
Имя модуля-2: Ролета; Пространство: Спальня
```

Как разбираться где-какое устройство если подключить к одному модулю 4 ролеты (реальная необходимость для меня) с текущей прошивкой пока не придумал. Об этом как раз говорит @Paly4

tretyakovsa, 2017-10-22:
@alxnovo 
SSDP имя должно быть уникально. По нему мы определяем IP адрес устройства.
SSDP это не название функции которую выполняет модуль. Не нужно туда писать Ролета.
Вы опишите, свою конфигурацию, где какие модули стоят и как вы хотите к ним обращаться.
Мы подумаем и сделаем.

alxnovo, 2017-10-22:
@tretyakovsa Мы наверно про разное говорим. Я про название модуля в настройках
[image]

Для того кто пишет прошивку понятно что нельзя дублирующие названия писать, для непосвящённого пользователя это неочевидно, да и неудобно в пользовании

Сейчас на главной странице у меня выглядит вот так

[image]

Визуально "Ролеты" выглядят одинаково, но написаны буквами из разных языков, раньше я не понимал что нельзя писать одинаковые имена модулям (теперь знаю что это уникальный SSDP)

В, принципе, "человекопонятный" SSDP нужен только для идентификации модуля через сетевое окружение (никогда этим не пользовался, в основном на Маке сижу, а заливаю прошивки через старый ноут на WinXP...ни на одном я SSPD не вижу), насколько это востребовано не мне судить.

Для идентификации модулей через Веб-интерфейс я бы сделал alis'ы к SSDP, которые автоматически присваиваются на основе, например, серийника ESP. Тогда уникальность будет всегда, независимо от действий пользователя и называть модули можно как угодно - хоть одинаково, хоть неограниченным кол-вом символов. Это просто мысли...

Если можно будет называть каждый управляющий блок в модуле своим именем - о чём писал изначально @Paly4 то как обозван модуль (SSDP) без разницы.

Моя будущая конфигурация: ESP, к которой подключено 8-ми канальное реле - по 2 на ролету, 4 выключателя-кнопки - для ручного управления. Получается с одного модуля буду управлять 4 ролетами, находящиеся в разных комнатах (питающая и сигнальная проводка от них приведена в одно место). Но это другая тема, я детали не продумывал, поэтому не пристаю с этой идеей. Просить начну когда реализуете в прошивки отключение реле управления жалюзи по времени, а не только по оборотам :)

tretyakovsa, 2017-10-22:
Да, в итоге мы приведем интерфейс к фото в первом посте.

sultan5448, 2017-10-23:
По поводу идентификации устройств. Насколько сложно сделать так чтобы поиск происходил не только по ssdp а вписывался в ручную?  при vpn полезно. на андроиде заметил слишком крупные графики в web

chanov, 2017-11-07:
Приветствую! Очень хорошее впечатление от прошивки, начинаю пробовать использовать с sonoff параллельно с самоделками с easyesp(esp8266.nu), у них просто из коробки есть подключение экрана и интеграция с Domoticz.

Как простое предложение к расширению функционала: добавить возможность случайного окна - "плюс минус время" ко времени срабатывания таймера - что бы можно было случайно, но в пределах например получаса включать-выключать свет и имитировать присутствие.

bArada007, 2017-11-11:
Ренат а может SSDP просто складывать из имени и зоны. 
например Ролета_Кухня, или Кухня_Ролета
тогда и путаница пропадет.

и есть предложение по поводу таймера
не понятно назначение "время работы". как мне казалось это должно быть временное окно в котором состояние реле соответствующее. 
Например если у меня уличное освещение и я задал вкл в 19:00 и время работы 06:00.
то логично предположить что все 6 часов у меня будет гореть свет. но если по каким то причинам в 21:00 вырубят электричество и включат в 22:00, то реле уже свет не включит, а хотелось что бы при старте оно проверяло временные окна. можно что то подправить в коде. хотя бы на этапе старта.
в идеале надо проверять именно "временное окно" и в каком состоянии должно находиться реле.
с другой стороны ручное управление должно отменять автомат.

bArada007, 2017-11-11:
странно, но у меня главная
[image]

логичней было бы попадать на страницу управления реле
[image]
а уже с нее на все остальные

не подскажите почему у меня главная не прописана? честно говоря случайно попал (методом тыка) на страницу реле, через сценарии

renat2985, 2017-11-11:
@bArada007, остальные страницы нормально работают? Проблемы только с главной страницей?
Попробуйте сменить дизайн. Помогло?
[image]

Мне нужно больше информации о проблеме...

bArada007, 2017-11-12:
вроде остальные страницы отображаются нормально. смена стиля ничего не меняет, кроме расположения злементов (внешнего вида).
[image]
[image]
[image]
[image]

все остальные страницы нормально. и стартовая была нормальной, появилась правда только после первой прошивки. пробовал перепрошивать с разной разметкой 1М(512к) и 1М(256к).
хотел ещё узнать, а где этот модуль запоминает сеть WiFi, потому что после прошивки он её помнил?

bArada007, 2017-11-12:
первое что я залил на Sonoff была прошивка jaluzi-master - в ней уже была проблема, через конфигурацию я перешел на sonoff, но ничего не изменилось. потом пробовал ещё два раза прошить но уже сборку Sonoff_WiFi_switch-master последнюю.

bArada007, 2017-11-12:
на всякий случай выкладываю список файлов на модуле, может не хватает какого файла json?
[image]

bArada007, 2017-11-12:
на пробу прошил WeMos D1. вот её скрин
[image]

bArada007, 2017-11-12:
попробовал залить старую сборку - на ней всё ОК. правда перестал видеть второй модуль.

renat2985, 2017-11-12:
@bArada007 насколько я понимаю вы закачиваете bin файлы, или сами собираете проект в Arduino?
А что у вас здесь: http://192.168.*.*/ssdp.list.json

bArada007, 2017-11-12:
{}
это всё.
я собираю в Arduino.

bArada007, 2017-11-12:
сравнивал вкладки SSDP в двух сборках
в той что не работает строка:
ssdpList = jsonWrite(ssdpList, jsonRead(configOptions, "SSDP"), WiFi.localIP().toString());

а в той что работает:
ssdpList = jsonWrite(ssdpList, jsonRead(configSetup, "SSDP"), WiFi.localIP().toString());

может дело в этом?

bArada007, 2017-11-12:
да. и ещё. на стартовой странице показан только второй модуль, себя не отображает

bArada007, 2017-11-12:
нашел причину.
[image]
закоментил верхнюю строчку и все теперь хорошо.
[image]

renat2985, 2017-11-12:
Не надо собирать самому. Заливайте bin файлы. В исходниках как правильно нестабильная \ не рабочая версия над которой мы работаем.

bArada007, 2017-11-12:
а где bin файлы брать? в папочке build? 
я просто когда вижу кучу бинарников теряюсь какой ставить. та и хочется разобраться чего для чего.
я этим модулем только знакомлюсь, да и с самой Arduino то же. раньше только Atmel программировал в CodeVision.

renat2985, 2017-11-12:
На главной странице репозитория:
https://github.com/tretyakovsa/Sonoff_WiFi_switch
Большими буквами Download Build & Spiffs (flash 1Mb 256Kb) 2017.10.11.zip
Ниже скриншот с какими парамитрами заливать а еще ниже ссылка на эту программу если у вас ее нет.
[image]

bArada007, 2017-11-12:
ок. спасибо.