tretyakovsa/Sonoff_WiFi_switch #55: Вопрос по командам устройства на сервер MQTT

lyovav, 2017-10-21:
Настраивал сервер MQTT и клиент на планшете и случайно обнаружил следующую картину

есть топик созданный устройством 
/IoTmanager/2311111-1458415/RELE_1_not/status  и контент  (1 вкл 0 викл)
что есть следствием нажатия кнопки на Веб интерфейсе устройства.
И создавая в клиенте MQTT  на планшете ту же кнопку обнаружил что клиентская кнопка на планшете помнит состояние (1 вкл 0 викл) и при этом создался параллельный топик

/IoTmanager/2311111-1458415/RELE_1_not/control и контент  (1 вкл 0 викл)

и каждый раз при нажатии кнопки обнвляется и status  и control

таким образом я почти  нашел что искал...

Ести устрйство заставить отправлять параллельно (вкл  викл) вместе два параметра на сервер MQTT
/IoTmanager/2311111-1458415/RELE_1_not/status и (1 вкл 0 викл)
/IoTmanager/2311111-1458415/RELE_1_not/control   (1 вкл 0 викл)
таким образом мы объединяем  управление с памятью состояния  конпки на клиентской MQTT части на планшете и на Веб интерфейсе устройства.
И если научить устройство при новом рестарте читать состояние в топике .../RELE_1_not/control 
(при условии что настроен сервер MQTT на устройстве )мы реши проблему запоминания состояния реле устройства до рестарта.
Как Вам такая идея?

tretyakovsa, 2017-10-22:
Так получается, что MQTT управление у меня остается на последнем месте. Я совершенно не понимаю как должен выглядеть интерфейс. Как правильно настроить виджеты и в какие топики что писать. MQTT остается всегда с самым низким приоритетом. И здесь есть только один выход. Если тот кому действительно нужен MQTT разберется в виджетах и алгоритме обмена с IoT managerom.

lyovav, 2017-10-22:
А можите для теста добавить строчку записи в топик  (файл Relay.ino) если Вас конечно не затруднит?
```
void topicPub(String topic, String data, boolean retain ) {
  client.publish(MQTT::Publish(prefix + "/" + chipID + topic,    "{\"status\":" + data + "}").set_retain(1));
  **client.publish(MQTT::Publish(prefix + "/" + chipID + topic,    "{\"contlol\":" + data + "}").set_retain(1));**

  }
```

tretyakovsa, 2017-10-22:
@lyovav 
Пока этого сделать не могу. Через несколько дней сделаю. Сейчас переписываю код еще раз, делая его более универсальным.
Вы пользуетесь IoTmanagerom? Может придумаете как будет выглядеть интерфейс от одного устройства и когда они включены все?

Paly4, 2017-10-23:
тоже вчера установил IoTmanager посмотреть что это такое
сервер MQTT локальный , выяснилось что приложение релюшку на сонофе переключает , а вот  на rgb никак не реагирует , думал хоть просто будет вкл/выкл
сегодня залил тестовый скетч в соноф , попробовал - понравилось как альтернатива управления не из домашней сети

> Может придумаете как будет выглядеть интерфейс от одного устройства и когда они включены все?
мне кажется придумывать ничего не нужно, приложение по меткам делает вкладки
но наверное для полноты управления через приложение или по mqtt нужно реализовать то , о чем говорили в теме Общие вопросы 
[image]
[IoTmanager2.zip](https://github.com/tretyakovsa/Sonoff_WiFi_switch/files/1408479/IoTmanager2.zip)