tretyakovsa/Sonoff_WiFi_switch #56: Нужна помощь - перевод.

renat2985, 2017-10-21:
Если у кого-то есть время, переведите пожалуйста. Нужен Английский, Латышский, Украинский.
Этот текст находится на странице модулей.
```
 "_comment": "---page.htm?configs---",
 "LangConfNTP":"Включаем время из сети",
 "LangConfDDNS":"Доступ по DDNS",
 "LangConfMQTT":"Для управления IoT Manager",
 "LangConfTIMERS":"Если требуются таймеры",
 "LangConfDHT":"Датчики температуры и влажности DHT11, DHT22, AM2302, RHT03, Si7021",
 "LangConfDS18B20":"Пока только на GPIO14, менять его нельзя",
 "LangConfA0":"Аналоговый вход",
 "LangConfTach":"Кнопка на GPIO0, доступно: rgbnot relaynot jalousienot",
 "LangConfRELAY":"Первая ячейка - PIN реле, вторая - состояние реле при включение ESP",
 "LangConfJALOUSIE":"Первая ячейка - PIN счетчик оборотов, вторая и третья - управление мотором",
 "LangConfRGB":"WS2811/WS2812/NeoPixel LEDs. Первая ячейка - PIN LED ленты, вторая - количество лед, терья - состояние LED при включение ESP",
 "LangConfRGBSHIM":"В стадии разработки, пока не работает",
 "LangConfMOTION":"Сенсор движения, доступно: relay rgb jalousie",
 "LangConfRF-RECEIVED":"Беспроводной приемник MHz частот",
 "LangConfRF-TRANSMITTER":"Беспроводной передатчик MHz частот, скоро.",
 "LangConfIR-RECEIVED":"Беспроводной приемник IR частот",
 "LangConfIR-TRANSMITTER":"Беспроводной передатчик IR частот.",
 "LangConfSerial":"Включить порт UART"
```

lyovav, 2017-10-22:
```
Укр версия перевода

 "LangConfNTP":"Включаємо час з мережі",
 "LangConfDDNS":"Доступ по DDNS",
 "LangConfMQTT":"Для керування IoT Manager",
 "LangConfTIMERS":"Якщо потрібні таймери",
 "LangConfDHT":"Датчики температури та вологості DHT11, DHT22, AM2302, RHT03, Si7021",
 "LangConfDS18B20":"Поки і тільки на GPIO14, міняти його не можна",
 "LangConfA0":"Аналоговий вхід",
 "LangConfTach":"Кнопка на GPIO0, працює з: rgbnot relaynot jalousienot",
 "LangConfRELAY":"Перша комірка - PIN реле, друга - стан реле при включенні ESP",
 "LangConfJALOUSIE":"Перша комірка - PIN лічильник оборотів, друга та третя - керування двигуном",
 "LangConfRGB":"WS2811/WS2812/NeoPixel LEDs. Перша комірка - PIN LED стрічки, друга - кількість лед ламп, третя - стан LED при включенні ESP",
 "LangConfRGBSHIM":"В стадії розробки, поки ще не працює",
 "LangConfMOTION":"Сенсор руху, працює з: relay rgb jalousie",
 "LangConfRF-RECEIVED":"Бездротовий приймач MHz частот",
 "LangConfRF-TRANSMITTER":"Бездротовий передавач MHz частот, вже незабаром.",
 "LangConfIR-RECEIVED":"Бездротовий приймач IR частот",
 "LangConfIR-TRANSMITTER":"Бездротовий передавач IR частот.",
 "LangConfSerial":"Включити порт UART"
 
 
  было еще не переведено 
 
  "LangDevices":"Головна"
  "LangLess":"Меньше",
  "LangMore":"Більше",
  "LangEqual":"Дорівнює",
  "LangNotEqual":"Не дорівнює",
  "LangOr":"чи",
  "LangTurn":"Оберти двигуна",
  "LangType":"Тип пристрою",
  "LangCloudUpload":"Завантажити з cloud",
  "LangCloudDownload":"Закачати в cloud",
  "LangSelect":"Виберіть",
```

renat2985, 2017-10-22:
@lyovav спасибо, все исправил. Sensors, да добавлю.