tretyakovsa/Sonoff_WiFi_switch #57: Настройка DDNS и IFTTT

alxnovo, 2017-10-22:
После просмотра [видео](https://youtu.be/v2VO3d31JRE) Рената про настройку DDNS и IFTTT захотел настроить закрытие ролет при заходе солнце.

Зарегестрировался на freemyip, прописал в прошивке hostname и service URL, указал 80 порт

[image]
открыл этот порт по IP модуля в роутере (IP статический в прошивке).

[image]
На всякий случай ESP и роутер перезагрузил.

Захожу на страницу управления, чтобы скопировать глобальную ссылку, а там ничего не появляться, только локальная ссылка.
[image]
Ладно, думаю, сам эту ссылку пропишу в IFTTT. Прописал в IFTTT 
http://name.freemyip.com:80/cmd?command=jalousienot 
[image]

При поверке реакция никакой нет.

[image]

Что я не так сделал и как проверить кто в цепочке ESP-Роутер-FreeMyIP виноват или где ошибка?

P.S. "name.fremyip.com" для скриншотов, реально DNS другой. Прошивка последняя из бинарников.

renat2985, 2017-10-22:
А вот эти ссылки работают:
http://name.freemyip.com:80/jalousieopen
http://name.freemyip.com:80/jalousieclose

alxnovo, 2017-10-22:
Не работают.

http://name.freemyip.com:80 как-то пинговаться должен? У меня он не пингуется что с указанием порта, что без

alxnovo, 2017-10-22:
После ручного ввода ссылки http://freemyip.com/update?token=0e6b4562fa7d8d1dc86131ec&domain=name.freemyip.com&myip=192.168.0.142

начал пинговаться http://name.freemyip.com 

и стали работать ссылки:

http://name.freemyip.com/cmd?command=jalousienot
http://name.freemyip.com/cmd?command=jalousieopen
http://name.freemyip.com/cmd?command=jalousieclose

Но IFTTT не заработал.

lyovav, 2017-10-22:
@alxnovo Вы указали открытый 80 порт а на роутер Вы все еще заходите 192.168.0.1:80 ?
или ВЕБку роутера перевели на 8080?

alxnovo, 2017-10-22:
@lyovav Не понял вопроса. Я в настройках прошивки указал порт 80, его же указал когда добавлял переадресацию. 

ВЕБку роутера никуда не переводил
192.168.0.1:80 заходит на роутер
192.168.0.1:8080 отбойник 

Чувствую вопрос с подвохом, но я на эти порты смотрю как баран на новые ворота, поэтому делаю по инструкциям не особо понимая смысла действий :)

lyovav, 2017-10-22:
@lyovav Я только хотел уточнить... 80 порт у Вас занят ВЕБкой роутера и все внешние соединения натыкаются на  ВЕБку роутера. 80 порт ВЕБки роутера надо переместить на другой порт к примеру 8080. Только тогда когда кто то будет стучатся к Вам на 80 порт тогда его перебросит роутер на  ВЕБку 192.168.0.142:80 (Ваше ESP  в локальной сети)

PS Никаких подвохов - только благие намерения)

tretyakovsa, 2017-10-22:
Вот почему я всем сердцем не хочу поддерживать DDNS. Слишком это сложно для тех кто не понимает.
Решение вижу такое. Запросы с IFTTT попадают в специальный топик MQTT сервера и спокойно  обрабатываются через MQTT. Только совершенно не понятно почему IFTTT  не работает с MQTT серверами?

alxnovo, 2017-10-22:
@tretyakovsa настроить MQTT сервер и топики тоже не самое простое для непосвящённых ;) Хотя более логичное.

@lyovav спасибо за пинок, бегло изучил порты, пробросы... Понял что не в прошивке (пока) дело. Я не понимаю как извне достучаться до той же ВЕБки роутера. 
У меня IP WAN'а статический и он не совпадает с моим внешним IP которые показывают разные сервисы, например, [Port Forwarding Tester](https://www.yougetsignal.com/tools/open-ports/).

В роутере (TP-Link) 80 порт открыт

[image]

Но извне по тому адресу что показывают сервисы не достучаться, ни по одному порту.
Как я понял, пока я не смогу для начала достучаться до своего роутера, ESP'шкой мне управлять не получится. И как это сделать я не знаю.

renat2985, 2017-10-23:
@alxnovo вам вы хотите открыть доступ к "Вебки роутера"? Вам не нужно это делать. :) Вам нужно открыть только порт к нужной вам ESP и насколько я понял вы это уже сделали т.к. вы писали выше что по этим ссылкам уже жалюзи работает:
http://name.freemyip.com/cmd?command=jalousienot
http://name.freemyip.com/cmd?command=jalousieopen
http://name.freemyip.com/cmd?command=jalousieclose

Далее вам нужно настроить IFTTT, судя по вашим фото у вас тоже все парильно. Теперь просто ждите заката в вашем городе. :) Точное время когда должен отправится запрос на ваше жалюзи вы можете узнать в google:  "Заход солнца в моем городе"

alexandersciv, 2017-10-23:
По настройке бесплатного DDNS сервиса имею полугодовую практику без проблемного использования ChangeIP.com. Настройку расписал здесь [http://cyber-place.ru/showpost.php?p=37236&postcount=18](url).
Там же как настроить интересный сервис [https://vpnki.ru/settings/before-you-begin/main-tech-info](url)

renat2985, 2017-10-23:
@alexandersciv скорее всего предлагаемый вари ресурс работает на https, ESP поддерживает только http, и те сервисы которые предлагаем мы значительно проще.

NLObP, 2018-03-26:
Такая же проблема. Глобальная ссылка пустая. Решения этой проблемы нет?

renat2985, 2018-03-26:
@NLObP, уже исправили. Все будет корректно работать в новой версии.

NLObP, 2018-03-26:
А новой версии еще нет?