tretyakovsa/Sonoff_WiFi_switch #58: HTTP Request Shortcuts - ярлыки на рабочий стол

alexandersciv, 2017-10-26:
Пока все находятся в ожидании новых версий прошивки, может у Рената найдётся время сделать инструкцию по программе HTTP Request Shortcuts?

renat2985, 2017-10-26:
У нас проблема, не можем выпустить новую версию. Не хватает места в ESP :) Решаем...

grossbel, 2017-10-26:
не влазит в 1 мв? код или web?

renat2985, 2017-10-27:
@grossbel код. Для обновления необходимо 50% и более свободного места в памяти. Иначе обновление просто не пройдет. Последняя версия занимает ровно 50%. И мы не можем выложить новую, т.к. она больше. Сергей на прошлой недели начал переписывать код, он должен стать легче. А я уже почти добавил возможность редактирования главной страницы :) так же мы уже сделали IR передатчик. Теперь можно управлять и телевизором. Думаю через 2-3 недели что ни будь придумаем и выложим новую версию.

i81, 2017-11-13:
Друзья, не уверен, что пишу в правильную тему, но у меня проблема: сегодня обновился через web морду и перестали работать http запросы. 
Нп http://192.168.1.169/relay и прочие один ответ "FileNotFound". Как вылечить? Остался без света.
П.с. В web тоже кнопки нет :(

chanov, 2017-11-13:
http://192.168.1.169/relayon и http://192.168.1.169/relayoff попробуйте

i81, 2017-11-13:
[image]
FileNotFound - всё так же...
[image]

i81, 2017-11-13:
[image]
[image]

i81, 2017-11-13:
В сценариях, если выбрать релле - вечная загрузка, а на главной странице просто нет кнопки релле...
Опять криво встала прошивка? Какой файл нужна скачать и откуда, что бы прошить через web update?

i81, 2017-11-13:
Перепрошил по этой инструкции https://youtu.be/D0Fu8etyVg0?list=PL6NJTNxbvy-IPTDQk8XjTV41oRrFafrRi&t=322 через update - помогло