tretyakovsa/Sonoff_WiFi_switch #59: Ошибка 403 в веб-интерфейсе при входе с другого устройства

shady2k, 2017-10-30:
Вхожу с компьютера в веб-интерфейс, все хорошо, страничка отображается. Затем пытаюсь открыть веб-интерфейс с ноутбука или с телефона, ошибка 403 Forbidden, говорит nginx. Мне нужно как-то выходить с одного устройства прежде чем открывать на другом? Если да, то это как-то не очень хорошо. Или я что-то не так делаю?

renat2985, 2017-10-30:
Насколько я понимаю эта проблема у вас на странице "Starting log"?
Такое очень редко но бывает. Просто попробуйте позже открыть и все будет хорошо.

shady2k, 2017-10-30:
Нет, стартовая страница. То есть, я вообще не указываю страницу, только IP. Вот скрин с телефона. Ожидание не помогает.
[image]

renat2985, 2017-10-30:
Этот IP принадлежит не ESP. Он принадлежит какому-то другому устройству на котором стоит веб сервер (nginx). Проверьте действительно ли с компьютера вы заходите на этот IP. Возможно вы указали двум устройствам один и тот же IP, попробуйте сменить его в ESP устройстве на странице конфигурация:
[image]

shady2k, 2017-10-30:
Ну как бы я уверен, что он принадлежит ESP. Другого устройства с таким IP у меня нет, адреса автоматически назначаются DHCP сервером.

[image]

renat2985, 2017-10-30:
Понимаете, скриншот который вы мне выслали 100% не может отдавать ESP, в нем нет nginx сервера. Его отдает какое-то другое устройство, которое сидит на этом-же IP, попробуйте в ESP устройстве сменить IP на другой, и перезагрузить ESP.

shady2k, 2017-10-30:
Теперь понял, спасибо, буду разбираться.

closed 2017-10-30 by shady2k

shady2k, 2017-10-30:
Разобрался, была проблема с маршрутизацией на роутере. Спасибо @renat2985 за помощь.