tretyakovsa/Sonoff_WiFi_switch #60: Не могу найти некоторых элементов, которые есть у других пользователей

shady2k, 2017-11-02:
@AleksandrGromov писал, что появилась возможность через веб-интерфейс установить статус реле при включении в этом issue https://github.com/tretyakovsa/Sonoff_WiFi_switch/issues/25. Подскажите, пожалуйста, где это в настройках? Все облазил, нет у меня такого пункта. Прошивка от 16.10.2017.
Еще заметил, что на некоторых скриншотах есть пункты, которых у меня нет. Например, нет кнопки Starting log, нет типа модуля.

renat2985, 2017-11-02:
А как вы определили что у вас прошивка за 16.10.2017? 
Может у вас в cache браузера сидит старый код, попробуй открыть страницу в приватном режиме браузера (инкогнито).

shady2k, 2017-11-02:
На странице конфигурация текущая прошивка выделяется жирным. Так ведь?

[image]

В инкогнито проверил, ничего не меняется. Кэш, куки почистил, нет эффекта.

renat2985, 2017-11-02:
Уже по вашему скриншоту вижу у вас, очень, очень старая версия.
Я встречал такие проблемы, почему такое происходит я не знаю.
Есть несколько решений:
1) Попробуйте еще раз обновится. Но я думаю вам это не поможет :)
2) Можете попробовать зайти на страницу /edit и обновит ручками все файлы, - долго но поможет.
3) Я думаю лучше всего вам попробовать обновится по этой ссылке: http://192.168.1.82/upgrade?build=http://backup.privet.lv/esp/loader/Sonoff_Loader.ino.generic.bin не забудьте заменить IP на свое устройство. 

Отпишитесь получилось ли.

shady2k, 2017-11-02:
Проверил 3 вариант. Вроде бы обновление прошло, но ничего не изменилось. И настройки все на месте. Похоже, что не получилось.

renat2985, 2017-11-02:
На какое устройство вы пытаетесь залить прошивку? 
Может там менее 1Mb flash памяти?
Если есть возможность, попробуйте залить через NodeMCU Flasher https://github.com/tretyakovsa/Sonoff_WiFi_switch/releases

У вас очень старая версия, обязательно пробуйте обновится, получите много плюшек и более стабильную работу.

shady2k, 2017-11-02:
У меня вот такой:

[image]

Наверное NodeMCU Flasher единственный вариант, завтра попробую.

renat2985, 2017-11-02:
Попробуйте обновится через NodeMCU Flasher именно по той ссылке что я вам дал выше, она затрет все и зальет заново.

shady2k, 2017-11-02:
Вот еще что хотел уточнить. По инструкции нужно заливать два файла build и spiffs. Я так понял, build я заливаю из той ссылки, что вы дали. А spiffs нужно заливать? И если да, то какой?

renat2985, 2017-11-02:
Не не не, заходите https://github.com/tretyakovsa/Sonoff_Loader/tree/master/ESP8266Flasher
Нажимаете там на кнопку: Download this file ESP8266Flasher.zip
Сохраняем себе на компьютер, распаковываем, заходим в появившуюся папку, запускаем там ESP8266Flasher.exe
И просто жмем загрузить в ESP. Все нужные bin файлы там уже будут указаны.

alexandersciv, 2017-11-03:
**@renat2985** По ссылке  http://192.168.1.82/upgrade?build=http://backup.privet.lv/esp/loader/Sonoff_Loader.ino.generic.bin  новая версия прошивки?

shady2k, 2017-11-03:
Так, ну я вроде бы прошился. Затем обновился до прошивки от 16.10.2017. Помогло.
Но так и не нашел где установить статус реле при включении.

shady2k, 2017-11-03:
Нашел. Если вдруг кто тоже не найдет, оно здесь: Конфигурация - Настройки разработчика - Дополнительные настройки - Тип модуля EDIT - RELAY.
// Первая ячейка - PIN реле, вторая - состояние реле при включение ESP

Спасибо за помощь!

closed 2017-11-03 by shady2k

renat2985, 2017-11-04:
@alexandersciv этот bin файл с репозитория Sonoff loader.