tretyakovsa/Sonoff_WiFi_switch #61: Необходимо отправлять статус реле через MQTT после подключения к WiFi

shady2k, 2017-11-03:
Сейчас после подключения к WiFi шлется только сообщение в топик /IoTmanager/XXXXXXXXX/config
{"id":0,"page":"buttons","descr":"Toggle","widget":"toggle","color":"green","topic":"/IoTmanager/XXXXXXXX/RELE_1_not"}
Можно ли как-то сделать так, чтобы через MQTT отправлялось сообщение о текущем статусе в топик /IoTmanager/ХХХХХХХХХХХ/RELE_1_not/status?

AVL4GITHUB, 2017-11-04:
Добавлю своих 5 копеек.
И не только что б статус реле отправляло, а и время последней отправки данных.
Отправлять можно раз в минуту или 5.
В таком случае, в МКТТ сразу будет видно, система ОНЛАЙН или где-то, что-то подвисло или не работает.

pastukhov, 2017-11-07:
Для мониторинга, что все онлайн в MQTT предусмотрено сообщение с типом LWT, оно объявляется обычно при старте системы, но становится доступно подписчикам только если модуль ушел в оффлайн, т.е. не выходит на связь с сервером в течении заранее определенного времени.

alxnovo, 2018-11-10:
MQTT заработало если скомпилировать из текущих исходников?

[image]

tretyakovsa, 2018-11-10:
Да статусы от есп будут попадать в топик /IoTmanager/XXXXXXXXX/relay1/config.
Мне кажется все очень медленно работает. А как у вас?

alxnovo, 2018-11-11:
Я пробовал где-то полгода назад MQTT запускать, видел в посмотрщике MQTT сообщения от ESP, но не смог отправляемыми сообщениями управлять. Подумал что должный быть какие-то хитрые сообщения и забросил тогда.
Ещё тогда заметил, если брокер не запущен, то ESP на вашей прошивке очень сильно тормозит, web-морда несколько минут открывалась, реакция на действие очень медленная была...до минуты.
Сейчас подключил роллершаттеры (еso + ваша прошивка) к Home Assistant и управляю по http. Но при такой связке, если через web или физическую кнопку управлять жалюзи, статус в Home Assistant не меняется, для этого и стал нужен MQTT
В идеале хотелось бы иметь логику в прошивке: если есть управление по MQTT, то девайс подчиняется только контролеру (серверу) несмотря на свои сценарии, таймеры; при пропадании связи с сервером, девайс переходит на локальные настройки и работает по заданным в прошивке сценариям и таймерам

tretyakovsa, 2018-11-11:
@alxnovo Может обсудим по skype или https://hangouts.google.com?

alxnovo, 2018-11-11:
Я только за, как связаться? Мой адрес <ник здесь>@gmail.com

closed 2022-07-16 by shady2k