tretyakovsa/Sonoff_WiFi_switch #62: Когда можно ожидать новую прошивку?

alexandersciv, 2017-11-10:
Когда можно ожидать новую прошивку?

tretyakovsa, 2017-11-11:
Пока не ясно. Я связки потянул. Ходить, сидеть очень трудно. Даже к компу не подхожу.

alexandersciv, 2017-11-11:
**@tretyakovsa** Знакомая ситуация. Сам недавно сломал надколенник. 
Быстрейшего выздоровления и новых решений в прошивке!

renat2985, 2017-11-11:
В понедельник постараюсь выложить тестовую версию. Появится возможность изменять внешний вид главных страниц через веб "Ctrl+M" + много других плушек. + протестируем как происходит обновление, т.к. были замечены некоторые проблемы.

alexandersciv, 2017-11-12:
**@renat2985**  Конечно ждём обновление. На 4pda.ru активно обсуждают Sonoff / eWeLink http://4pda.ru/forum/index.php?showtopic=807727 включили в обсуждение Вашу прошивку. Может есть необходимость для популяризации прошивки завести тему на 4pda, по аналогии с 1M Smartphone?

renat2985, 2017-11-13:
@alexandersciv чем больше пользователей тем лучше... Но у меня к сожалению нет времени на создание тем на других сайтах. 
Новая бета версия: https://github.com/tretyakovsa/Sonoff_WiFi_switch/issues/63