tretyakovsa/Sonoff_WiFi_switch #63: Бета тестеры. Налетай! :)

renat2985, 2017-11-13:
Выкладываю очередную бета версию, главной фишкой которой является возможность редактировать главную страницу через веб.

С начало смотрим видео: https://www.youtube.com/watch?v=K4lXwBxvwws&index=1&list=PL6NJTNxbvy-IPTDQk8XjTV41oRrFafrRi

Потом пробуем. Важно что бы у вас была установлена последня доступная версия. Т.к. обновлять с вами будем только файловую систему

http://192.168.1.21/upgrade?spiffs=http://backup.privet.lv/esp/spiffs.0xBB000_flash_size_1Mb.256Kb_2017.11.13.bin

Впечатления и найденные ошибки пожалуйста пишите ниже. И не забываем IP сменить на нужный вам.

alexandersciv, 2017-11-13:
**@renat2985**  Очень впечатляет возможность изменения интерфейса, но очень бы хотелось новых сценариев и в частности:
- включать реле с показаний датчиков на удалённом реле
- возможность действий на реле по нескольким подключенным к нему датчикам (движения, освещения, температуры)

renat2985, 2017-11-13:
@alexandersciv спасибо. :) Сценариями Сергей занимается. Я больше по веб части занимаюсь в проекте. Он же занимается Arduino. 
Как скоро появятся нужные вам возможности пока не знаю. Но они обязательно появятся. Сергей связки сильно подтянул. Пока оправится, + он еще начал переписывать свой старый код. В общем не знаю сколько это все времени займет. 
А пока можно заказывать только веб фичи. :)

alexandersciv, 2017-11-13:
**@renat2985** Моя постоянная мечта заняться веб дизайном, но пока сложно. Вот если объекто- ориентированные  решения можно применить, тогда проще.
Сергею я уже направлял пожелание быстрейшего выздоровления.

renat2985, 2017-11-13:
В будущем я думаю сделаю возможность создавать, редактировать интерфейс нажимая на кнопочки а не редактируя json код. Мы к этому и движемся. Чего еще не хватает в веб интерфейсе?

alexandersciv, 2017-11-13:
**@renat2985**  Лично меня пока всё устраивает, поскольку в тесте всего 3 санофа. Как прибавятся сценарии, тогда и добавятся пожелания.

ilyalush, 2017-11-13:
Чего очень не хватает, так это единой точки входа с аутентификации. Не безопасненько. :) А так все отлично! Сейчас пишу кроссплатформенное приложение к вашей системе, если интересно, то выложу (только чуть позже, пока очень сырое все).

renat2985, 2017-11-14:
@ilyalush единственной точки входа? Вы можете зайти на любое устройство и в нем увидите все другие устройства в вашей сети. Вы о чем, я не понимаю. Сделайте веб ярлык на рабочем столе любого из устройств и будет вам счастье.

Аутентификацию тоже не считаю нужным. Полноценно управлять устройствами можно только в локальной сети. А если у вас нет пароля на WiFi роутере поставьте его и у вас будет все безопасно. Попробуйте переубедить. 

Кросс платформенное приложение :) ну покажите.

ilyalush, 2017-11-14:
Под единой точкой входа я подразумеваю что-то вроде абстрактного класса php в index файле, который контролирует запросы и аутентификацию (это как пример). Понятно, что php для данного проекта не подходит, хотя после Вашего комментария про то, что не нужна аутентификация - это уже и не актуально.
По поводу необходимости аутентификации... лично для меня - это само собой разумеющееся. Система у Вас хорошая, многофункциональная и, думаю, получит широкое распространение со временем, а соответственно и разные решения, которые могут реализовываться, в открытых или плохо защищенных сетях, да и не все завязано на локальную сеть. 
Я высказываю сугубо свое мнение и тут Ваше, как разработчика, право делать что-то или нет. :)

alexandersciv, 2017-11-17:
**@tretyakovsa** Извиняюсь, за беспокойство выздоравливающего человека, но хотелось бы знать когда будет выпущена, и будет ли,  доработка Вашей прошивки с включением ранее обещанных сценариев? По скольку на основании этого планировал строить всю систему. Или переключаться на альтернативные решения?

tretyakovsa, 2017-11-18:
Сценарии обязательно будут сделаны. Как только смогу сидеть у компьютера.

alexandersciv, 2017-12-10:
**@tretyakovsa** , **@renat2985** , понимаю, что есть объективные причины остановки развития Вашего замечательного проекта, и что Вы ни кому ни чем не обязаны из за этого, но хотелось бы определённости в сроках развития проекта. Может пока без кардинального изменения прошивки, которая и так в процессе тестирования зарекомендовала себя хорошо, просто добавить сценарии? Для доработки интерфейса на разных устройствах могу выложить скриншоты и предложения по изменению, если Вы будите их рассматривать как рабочие варианты.

tretyakovsa, 2017-12-12:
До 25 декабря будет новая версия,

alexandersciv, 2017-12-12:
**@tretyakovsa** Хорошая новость по новой версии, значит и здоровье поправляется!
**@renat2985** На счёт предложений по интерфейсу пока подождём?

renat2985, 2017-12-12:
@alexandersciv показывайте скриншоты, их можно приложить прямо сюда. А лучше создайте новую тему, например "Новые дизайны".