tretyakovsa/Sonoff_WiFi_switch #64: Цветовая палитра RGB проекта

renat2985, 2017-11-14:
После того как появилось вот это видео: https://www.youtube.com/watch?v=7z_xcVtL0aE&feature=youtu.be&list=PL6NJTNxbvy-IPTDQk8XjTV41oRrFafrRi&t=420
В котором я рассказываю как можно сменить стандартные цветовые картинки на свои. Мне стали писать мол закачал картинку и перестала работать вся система. Проблема в том что сейчас у нас свободно около 100кб. И если вы закачаете картинку более 100Кб у вас все прекратит работать.

В этой теме я предлагаю выкладывать красивые цветовые палитры не более 20Кб.
[image]
[image]

renat2985, 2017-11-14:
Еще парочка:
[image]
[image]

renat2985, 2017-11-14:
И еще такой тематики:
[image]
[image]
[image]

renat2985, 2017-11-14:
[image]
[image]
[image]

renat2985, 2017-11-14:
[image]
[image]

alexandersciv, 2017-11-14:
**@renat2985**  По мне, так строгое классическое решения с насыщенным цветом и чёткими границами, самое лучшее.

renat2985, 2017-11-14:
[image]
[image]
[image]
[image]
[image]